espressif/esp-idf #4318: PppoS(Sim800L): Cannot Connnect with MQTT

MrCurio, 2019-11-11:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: DevKitC: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-369
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0

- Operating System:     Linux
- Power Supply:      external 5V

## Problem Description

I have recently bought a sim800L module and i am not able to connect to the broker of the example: PppoS of esp-idf. The module is working correctly since i get an IP address and all the info, such as: IMSI, battery state... and i can see that i am connected by GPRS. 

The problem comes when i initialize the MQTT client. Every time i try to connect to the broker i get the message: "MQTT client: Error Transport connect" and right after, i get the event "ERROR" followed by "DISCONNECTED".

The Url of the broker is the same as in the example so, i assume, thats not the problem. i wonder if my data plan company is stopping my connection. I am from Spain and my company is Movistar.es

And another doubt: really often it resets due to timeout even if the module is in stably mode.
Does it work better with the other module of the example: BG96

Thanks in advance!!

### Expected Behavior
Connect to broker, subscribe to any topic and then publish

### Actual Behavior
Doesnt connect to the broker

### Steps to repropduce

1. Connect pins of the module: 
5V --> External 5V battery with enough output current
GND -> GND
Sim_TX -> Default Example(26) 
Sim_RX-> Default Example(25)
Reset -> pin number 4 of esp32 and high (otherwise resets constantly)

2. Make flash monitor example

### Code to reproduce this issue

The code is the same as in the example

## Debug Logs

ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 44
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 98 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 44
E (124780) MQTT_CLIENT: Error transport connect
I (124780) pppos_example: MQTT_EVENT_ERROR
I (124780) pppos_example: MQTT_EVENT_DISCONNECTED
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_input[1]: got 50 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_input[1]: got 50 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
I (139780) pppos_example: MQTT other event id: 7
pppos_netif_output[1]: proto=0x21, len = 44
pppos_netif_output[1]: proto=0x21, len = 44
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 44
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 44
pppos_netif_output[1]: proto=0x21, len = 44
E (158280) MQTT_CLIENT: Error transport connect
I (158280) pppos_example: MQTT_EVENT_ERROR
I (158280) pppos_example: MQTT_EVENT_DISCONNECTED

----------------------------Reset issue------------------------------------
I (451) spi_flash: detected chip: generic
I (452) spi_flash: flash io: dio
I (452) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (460) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (470) uart: queue free spaces: 30
E (970) esp-modem: esp_modem_dte_send_cmd(215): process command timeout
E (970) dce_service: esp_modem_dce_sync(48): send command failed
E (970) sim800: sim800_init(460): sync failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d460c  PS      : 0x00060130  A0      : 0x800d26a9  A1      : 0x3ffb6200  
0x400d460c: app_main at /home/juan-platero/esp/esp-idf/examples/protocols/pppos_client/main/pppos_client_main.c:285

A2      : 0x00000000  A3      : 0x3ffb2ea8  A4      : 0x3ffb97bc  A5      : 0x00000000  
A6      : 0x00060021  A7      : 0x00060023  A8      : 0x800d460a  A9      : 0x3ffb61d0  
A10     : 0x00000000  A11     : 0x3ffbb1d4  A12     : 0x400d5920  A13     : 0x3ffb6240  
0x400d5920: sim800_deinit at /home/juan-platero/esp/esp-idf/examples/protocols/pppos_client/components/modem/src/sim800.c:428

A14     : 0x00000000  A15     : 0x0001c200  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000007c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 708a5ad985c93136b3d03532c5b62401be8ab83d9eb67ff9d1c25e850e4aa394

Backtrace: 0x400d4609:0x3ffb6200 0x400d26a6:0x3ffb62e0 0x40088a05:0x3ffb6300
0x400d4609: app_main at /home/juan-platero/esp/esp-idf/examples/protocols/pppos_client/main/pppos_client_main.c:279

0x400d26a6: main_task at /home/juan-platero/esp/esp-idf/components/esp32/cpu_start.c:569

0x40088a05: vPortTaskWrapper at /home/juan-platero/esp/esp-idf/components/freertos/port.c:143

----------------------------------------------------------

david-cermak, 2019-11-12:
Hi @MrCurio 

The first issue is very unlikely the esp-modem problem if you are correctly getting the IP address. An easy way to check is to try to access some page with the SIM card and then update the pppos-client example to use e.g. `esp_http_client` to get to the same page.

As for the other one (reset issue), could you please try to increase [this timeout](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/include/esp_modem_dce.h#L53) value and test again?

MrCurio, 2019-11-12:
I have found what the problem was. Due to current power to my module, the synchronization between the esp32 and the module was not as good as it should be. Thats why sometimes i got the ip and suddenly i couldnt connect!

Anyway, thanks for your reply @david-cermak !

I will close the issue! Bye!

closed 2019-11-12 by MrCurio

PIYUSHGIRI2021, 2021-09-23:
Hii, @david-cermak ,in my module also i get same issue,first i got ip but after that ,below error get and MQTT DISCONNECT.
I (78097) esp-netif_lwip-ppp: Connection lost
I (78097) pppos_example: Modem Disconnect from PPP Server

E (158427) TRANS_TCP: [sock=54] connect() error: Host is unreachable
E (158427) MQTT_CLIENT: Error transport connect

E (351247) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned 0
E (351247) MQTT_CLIENT: MQTT connect failed.

I continue publish data using MQTT but after some time Connection lost and disconnect from PPP ...please let me know how can i resolve this issue,

i used counter at MQTT DISCONNECT event and if 3 times get MQTT DISCONNECT so i called pppos_stop() and pppos_start() API,i try to reconnect but some times its work some time not work,
please let me know in between my pppos disconnect,what is a reason behind this issue.
thanks in advance!

david-cermak, 2021-09-23:
Hi @PIYUSHGIRI2021 

I'm afraid we cannot help if it's a traditional connection lost kind of error. I'd suggest switching back to command mode and checking the signal strength if you cannot connect.

PIYUSHGIRI2021, 2021-09-24:
Hi @david-cermak 
my APN is CMNET ,main issue are PPPOS is not continuously connect it will lost Connection after sometime, so how to handle this issue,if mqtt through continue(using while loop) i publish data and sim remove or so network not come or any other setuation ip lost so how can i resonnect..
right now i used this API for reconnect if 4 time get mqtt disconnect or PPPOs connection lost
void reconnect_fun(void)
{
    //esp_mqtt_client_destroy(mqtt_client);
    esp_modem_stop_ppp(dte);
    esp_modem_start_ppp(dte);
    /* attach the modem to the network interface */
    esp_netif_attach(esp_netif, modem_netif_adapter);
}

please let me know how can i handle when at starting publish data properly after sometime (in-between)not publish, this condition happen due to IP lost or any pppos connection lost.it is not reconnect automatically..so how can i resolve this issue.
thanxs.

chegewara, 2021-09-24:
There is `RESET` pin on `SIM800L`. Worth to use it (pull down for 1-2s).

I am currently working with this module and it is running when no wifi connection, so maybe not the same use case as yours, but i have reliable mechanism, i think.
```
void connectLTEmodem()
{
    if(dte) return;

    dte = esp_modem_dte_init(&config);
    /* Register event handler */
    ESP_ERROR_CHECK(esp_modem_set_event_handler(dte, modem_event_handler, ESP_EVENT_ANY_ID, NULL));

    // Init netif object
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_PPP();
    esp_netif = esp_netif_new(&cfg);
    assert(esp_netif);

    modem_netif_adapter = esp_modem_netif_setup(dte);
    esp_modem_netif_set_default_handlers(modem_netif_adapter, esp_netif);

    digitalWrite(MODEM_RESET_PIN, 0);
    delay(2000);
    digitalWrite(MODEM_RESET_PIN, 1);

    while (!dce)
    {
        /* create dce object */
#if CONFIG_EXAMPLE_MODEM_DEVICE_SIM800
        dce = sim800_init(dte);
#elif CONFIG_EXAMPLE_MODEM_DEVICE_BG96
        dce = bg96_init(dte);
#elif CONFIG_EXAMPLE_MODEM_DEVICE_SIM7600
        dce = sim7600_init(dte);
#else
#error "Unsupported DCE"
#endif
    }

    assert(dce != NULL);
    ESP_ERROR_CHECK(dce->set_flow_ctrl(dce, MODEM_FLOW_CONTROL_NONE));
    ESP_ERROR_CHECK(dce->store_profile(dce));
    while (!dce->oper || strlen(dce->oper) < 2)
    {
        delay(1000);
        if(dce && !dce->oper < 2) esp_modem_dce_get_operator_name(dce);
        else ESP_LOGW(TAG, "Operator name: %s", dce?dce->oper:"");
    }

    /* Print Module ID, Operator, IMEI, IMSI */
    ESP_LOGW(TAG, "Module: %s", dce->name);
    ESP_LOGW(TAG, "Operator: %s", dce->oper);
    ESP_LOGW(TAG, "IMEI: %s", dce->imei);
    ESP_LOGW(TAG, "IMSI: %s", dce->imsi);
    /* Get signal quality */
    uint32_t rssi = 0, ber = 0;
    ESP_ERROR_CHECK(dce->get_signal_quality(dce, &rssi, &ber));
    ESP_LOGI(TAG, "rssi: %d, ber: %d", rssi, ber);
    /* Get battery voltage */
    uint32_t voltage = 0, bcs = 0, bcl = 0;
    ESP_ERROR_CHECK(dce->get_battery_status(dce, &bcs, &bcl, &voltage));
    ESP_LOGI(TAG, "Battery voltage: %d mV", voltage);
    /* setup PPPoS network parameters */
#if !defined(CONFIG_EXAMPLE_MODEM_PPP_AUTH_NONE) && (defined(CONFIG_LWIP_PPP_PAP_SUPPORT) || defined(CONFIG_LWIP_PPP_CHAP_SUPPORT))
    esp_netif_ppp_set_auth(esp_netif, auth_type, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);
#endif
    /* attach the modem to the network interface */
    esp_netif_attach(esp_netif, modem_netif_adapter);
    /* Wait for IP address */
    xEventGroupWaitBits(event_group, CONNECT_BIT, pdTRUE, pdTRUE, portMAX_DELAY);

    LTE_CONNECTED = true;
}

// --------------------------------- > DEINIT PPPOS <------------------------------- //
void disconnectLTEmodem()
{
    if(!dte) return;

    // /* Exit PPP mode */
    ESP_ERROR_CHECK(esp_modem_stop_ppp(dte));

    // /* Power down module */
    ESP_ERROR_CHECK(dce->power_down(dce));
    ESP_LOGI(TAG, "Power down");
    ESP_ERROR_CHECK(dce->deinit(dce));

    /* Unregister events, destroy the netif adapter and destroy its esp-netif instance */
    esp_modem_netif_clear_default_handlers(modem_netif_adapter);
    esp_modem_netif_teardown(modem_netif_adapter);
    esp_netif_destroy(esp_netif);

    ESP_ERROR_CHECK(dte->deinit(dte));
    dte = NULL;
    dce = NULL;
    ESP_LOGW(TAG, "LTE modem and connection destroyed");
    LTE_CONNECTED = false;
}
```
Its basically copy/paste espressif example with few small changes.
It is not fully tested yet, and i dont know how it will behave when `SIM800L` lost signal and disconnect.

PIYUSHGIRI2021, 2021-11-01:
Hii @david-cermak & @MrCurio 
   Is there possible to work PPPOS and GPS both are work together?if yes than pls let me know which way we can do it.

Thanks

david-cermak, 2021-11-02:
Hi @PIYUSHGIRI2021 

Yes, it is possible. Assuming you mean sending and receiving `AT+CGPSINFO`-like commands and at the same time maintaining the network connection in PPP mode.
There are basically two options:
1) pausing and resuming PPP mode using `+++` and `ATO` commands (I have some WIP commits when experimented with two-phase OTA -- paused the network while updating FW -- resumed for downloading the image. I can share it here if you like)
2) using CMUX mode with one virtual terminal designated to PPP mode (example here https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/simple_cmux_client)