espressif/esp-idf #4319: uart_read_bytes trow double char

caveman1996, 2019-11-11:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-815-ga45e99853-dirty
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

We were using the version 3.3 before we got a problem with our system. I've than made a Ubuntu virtual machine to compile the project because the solution Espressif provide for Windows didn't worked for us. 
By making the new virtual machine, I've downloaded the new IDF version (I'm not so sure how to pick the version I want). Everything is compilling (after some tweaks because of the new compiller), thought the project doesn't run as before. 
After some tests, I saw that uart_read_bytes() always give one more character at the beginning of the received char array. I've try to debug the problem myself and I didn't found anything.

### Expected Behavior
Received data on UART 0: 41 04 02 01
Decoded data by uart_read_bytes: 41 04 02 01

### Actual Behavior
(example 1)
Received data on UART 0: 41 04 02 01
Decoded data by uart_read_bytes: 441 04 02 01

(example 2)
Received data on UART 0: A6
Decoded data by uart_read_bytes: AA6

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main(){
    uart_flush(uart_num);
    while (bufferedSize == 0) {
	uart_get_buffered_data_len(uart_num, &bufferedSize);
    }

    if (uart_read_bytes(uart_num, &answerBuf[memIndexBuf], (uint32_t)bufferedSize, (100 / portTICK_PERIOD_MS)) == 0) {
	return ESP_FAIL;
    }

    ESP_LOGD(TAG_FUNC, "String in buffer: %s", answerBuf);
}

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3832259/sdkconfig.txt)

koobest, 2019-11-11:
```
bzero(&answerBuf[memIndexBuf,bufferedSize+1);

if (uart_read_bytes(uart_num, &answerBuf[memIndexBuf], (uint32_t)bufferedSize, (100 / portTICK_PERIOD_MS)) == 0) {
	return ESP_FAIL;
    }
```

caveman1996, 2019-11-11:
> bzero

Thanks for the idea, but I already done it at the initialization (array[x] = {0};) and by a memset().

negativekelvin, 2019-11-12:
What is memIndexBuf

koobest, 2019-11-18:
Hi, @caveman1996 
Did you solve this issue?

Alvin1Zhang, 2020-01-08:
@caveman1996 Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-05-27:
@caveman1996 Thanks for reporting, will close the ticket now due to short of feedback, feel free to reopen with more details/updates. Thanks.

closed 2020-05-27 by Alvin1Zhang