espressif/esp-idf #4322: DMA ADC Read Failed On Connection to AP when in WiFi STA (station) mode

wammaw1314, 2019-11-12:
My problem is basically the same as #3714, but there are some supplements.

## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-815-ga45e998-dirty
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

When acquiring ADC data through DMA transfers and the device is in station mode (WIFI_MODE_STA). As the device connects to the AP, the ADC DMA buffer ceases to show valid data and continues to read (a probably random) fixed value.
I experimented with #3714 and added this line of code during WiFi initialization: `esp_wifi_set_ps(WIFI_PS_NONE);`

Get the following results:
1. The DMA ADC can read the normal value without restarting the device.
2. After executing `esp_restart();`, the device DMA ADC is read only as a zero value.
3. If the device is reset by resetting the IO via ESP32, the result is normal.

### Code to reproduce this issue

The full code
Copy your ESP_IDF example structure:
examples\peripherals\i2s_adc_dac\main\app_main.c
and replace everything in app_main.c with this:
[app_main.zip](https://github.com/espressif/esp-idf/files/3834153/app_main.zip)

## Debug Logs

Program running debugging log is as follows:
[I2S-DMA-ADC-WiFi-Reboot-log.txt](https://github.com/espressif/esp-idf/files/3834193/I2S-DMA-ADC-WiFi-Reboot-log.txt)

#### I hope to get your help, thank you !

negativekelvin, 2019-11-12:
Did you see the comment https://github.com/espressif/esp-idf/issues/3973#issuecomment-549648060 ?

wammaw1314, 2019-11-12:
@negativekelvin Thanks, I saw it. I am using the latest master branch, so now I have to wait for the fix to be ready, right?

negativekelvin, 2019-11-12:
Right

koobest, 2019-11-18:
@wammaw1314 
The patch to solve this issue has been added to the master branch, please help test it :)
(https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_random.c)

thanks !!

negativekelvin, 2019-11-18:
https://github.com/espressif/esp-idf/commit/ba72de2099f17c2244e4e088327dfd42e55aab1f

original-standard, 2020-07-29:
I'm in the same situation with the bluetooth connection.
Can the WIFI_PS_NONE mode help with my problem?

tom-borcin, 2021-05-07:
Hi @wammaw1314,

we think the issue is fixed now and therefore we are closing it to keep our backlog manageable. If you don't agree with this decision, please let me know.

Thank you,

Tomas

closed 2021-05-07 by tom-borcin