espressif/esp-idf #4323: Not able to read the binary from amazon s3 bucket

ppvastkar, 2019-11-12:
----------------------------- Delete below -----------------------------
When I am using a simple OTA example code, I am not able to read binary from the server. below is my log.
Does anyone have an idea who to resolve this issue, and whats going on..?

Device Log:
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=x-amz-id-2, value=1mWt/DtJ55j2t0TBqY9UMwZmWDVYC8YT28kd2GEHr8usz2EU7KuQkvYKR27c61TLuv/wlL3jsmk=
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=x-amz-request-id, value=16394AAE8B6572ED
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Date, value=Mon, 11 Nov 2019 14:35:45 GMT
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Last-Modified, value=Mon, 11 Nov 2019 10:11:09 GMT
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=ETag, value="1a6a35824a5050cdd8cc97be24419c97"
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Accept-Ranges, value=bytes
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/octet-stream
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Content-Length, value=1306624
<DBG><../main/ota_handler.c><85>< http_ota_envet_handler>HTTP_EVENT_ON_HEADER, key=Server, value=AmazonS3
<DBG><../main/ota_handler.c><148><cta_https_ota>content length: 1306624
<INFO><../main/ota_handler.c><153><cta_https_ota>Starting OTA...
<INFO><../main/ota_handler.c><164><cta_https_ota>Writing to partition subtype 16 at offset 0x310000
<INFO><../main/ota_handler.c><173><cta_https_ota>esp_ota_begin succeeded
<INFO><../main/ota_handler.c><175><cta_https_ota>Please Wait. This may take time
I (50092) mbedtls: ssl_tls.c:8207 => read

I (50092) mbedtls: ssl_tls.c:4305 => read record

I (50102) mbedtls: ssl_tls.c:2532 => fetch input

I (50102) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (50112) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (50112) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (50122) mbedtls: ssl_tls.c:2738 <= fetch input

W (50132) mbedtls: ssl_tls.c:4088 bad message length

W (50132) mbedtls: ssl_tls.c:4338 ssl_get_next_record() returned -29184 (-0x7200)

W (50142) mbedtls: ssl_tls.c:8272 mbedtls_ssl_read_record() returned -29184 (-0x7200)

E (50152) esp-tls: read error :-29184:
E (50162) TRANS_SSL: esp_tls_conn_read error, errno=Success
<INFO><../main/ota_handler.c><190><cta_https_ota>Connection closed,all data received
<DBG><../main/ota_handler.c><94>< http_ota_envet_handler>HTTP_EVENT_DISCONNECTED
<DBG><../main/ota_handler.c><94>< http_ota_envet_handler>HTTP_EVENT_DISCONNECTED
<DBG><../main/ota_handler.c><211><cta_https_ota>Total binary data length writen: 0
<ERR><../main/ota_handler.c><219><cta_https_ota>Error: esp_ota_end failed! err=0x258. Image is invalid
<ERR><../main/ota_handler.c><301><OTA_Task>Firmware Upgrades Failed

My Code:
void OTA_Task( void* pvParameters )
{
	uint32_t NotificationVal = 0;

	while( TRUE) {

		NotificationVal = ulTaskNotifyTake( pdTRUE, ( TickType_t ) 1000 );

		if( NotificationVal > 0 ) {

			DBG_PRINT("Starting OTA...");

			DBG_PRINT("Binary URL: %s",g_ota_url);

			DBG_PRINT("Free Heap Size: %d bytes",xPortGetFreeHeapSize());

			/* Wait for the callback to set the CONNECTED_BIT in the event group. */
			xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);

			DBG_PRINT("Connect to Wifi ! Start to Connect to Server....");

			esp_http_client_config_t config = {

				.url = g_ota_url, //https://otabinfile.s3.ap-south-1.amazonaws.com/xxxx.bin

		        .event_handler = cta_http_ota_event_handler

		    };

			esp_err_t ret = cta_https_ota(&config);

			if (ret == ESP_OK) {

				esp_restart();

			} else {

				ERR_PRINT("Firmware Upgrades Failed");

				DBG_PRINT("Free Heap Size: %d bytes",xPortGetFreeHeapSize());

			}

		}//NotificationVal

	}//While

ppvastkar, 2019-11-12:
By change in menuconfig this issue goes away.

closed 2019-11-12 by ppvastkar

lakshmisivareddy, 2020-04-24:
HI 
also facing the same issue 
can you please let us know what is the change in menuconfig

ppvastkar, 2020-04-24:
In your Amazon S3 make sure you have assigned the public access to the file.
Also, make sure you have 2 different partitions in menuconfig. OTA_0 & OTA_1