espressif/esp-idf #4325: TCP Server fails to receive/response after a while

msmorkan, 2019-11-12:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v3.3-71-g46b12a5
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

TCP Socket stops working without crashing.
- ESP32 is in softAP mode
- Client is written in android (just sends same string repeatedly: "{'cmd':0}" then puts this into array which has a size of 1408 bytes)
- No error
- Client continues to send while ESP32 does not show any sign of receive
- If ESP32 was not hanging it should break (it does in normal conditions) but I **think** it hangs on **recv(...)** line

### Expected Behavior

Respond/receive commands coming periodically (every 250 ms) from TCP client 

### Actual Behavior

Stops receiving/sending after a minute or two, while client still thinks that its connected.

### Steps to repropduce

1. Removed redundant code that is not relevant to socket, problem did not resolved

```cpp
#include "esp_log.h"

void app_main()
{
    serviceSetupTcpInit();
}

```
[Please click here to see full code](https://gist.github.com/msmorkan/6a90d14c1f99fa7bca331c0d50bbfa12)

## Other items if possible

-  [See sdkconfig file here](https://gist.github.com/msmorkan/12e3d41610b6ee576ce8785acd46e10f)

Thanks a lot

negativekelvin, 2019-11-12:
Cjson free bugs are very popular

https://github.com/DaveGamble/cJSON#parsing-json

msmorkan, 2019-11-12:
I am already using CJSON

negativekelvin, 2019-11-12:
> Once it returns, you are fully responsible for deallocating it after use with cJSON_Delete.

msmorkan, 2019-11-12:
> > Once it returns, you are fully responsible for deallocating it after use with cJSON_Delete.

Thanks. I am trying now. Do you think that causes a problem like this?

liuzfesp, 2020-03-25:
Hi @msmorkan if your program has memory leak, the WiFi/TCPIP stack will finally has problem to receive any packet. Moreover, https://gist.github.com/msmorkan/6a90d14c1f99fa7bca331c0d50bbfa12#file-service_tcp_setup-h-L39 indicates the recv() in your application is a blocking call without any timeout, that's not recommended way to use TCP recv() for a robust application. What my suggestion is:

- Fix memory leak first
- Give receive() a timeout value via setsockopt(socket_id, SOL_SOCKET, SO_RCVTIMEO, ...)

Then do a test.

liuzfesp, 2020-03-31:
Hi @msmorkan, any update about this issue?

msmorkan, 2020-03-31:
> Hi @msmorkan, any update about this issue?

Hello @liuzfesp , 
Handling cJSON worked very well. I think problem is solved after fixing the memory leak issue. 
Thanks a lot

closed 2020-03-31 by msmorkan