espressif/esp-idf #4327: The function(`WRITE_RTC_FIELD`、`WRITE_RTC_REG`) called in the assembler is useless in ULP.

GitOfLance, 2019-11-12:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version :  v4.0-dev-1443-g39f090a4f 2nd stage bootloader
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description:
I use `ULP` to control `GPIO25`, so that the square wave whose output duty ratio is `1/3` is used to control the infrared emission. However, the pin level change is not captured. The code is as follows:
```
/**
 * FunctionName	: ulp_program_init
 * Description	: ulp协议处理程序初始化，
 * Input		: void
 * Output		: 启动ULP程序
 * Return		: void
 * Others		: 需要提前配置menuconfig
 */
void ulp_program_init()
{
    int ret = -1;
    gpio_num_t gpio_num;
    static int i = 0;
    
    ESP_LOGI(TAG, "ready exec ulp program before into deep sleep");

    //加载汇编二进制文件到RTC内存
    ret = ulp_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
    if (0 != ret)
    {
        ESP_LOGE(TAG, "failed to load ulp bin program");
        exit(-1);
    }
    ESP_LOGI(TAG, "test---1");

    //断言使用的gpio是不是RTC IO
    assert(rtc_gpio_desc[PIR_IE_GPIO_ULP].reg && "GPIO used for IE must be an RTC IO");
    assert(rtc_gpio_desc[PIR_IR_GPIO_ULP].reg && "GPIO used for IE must be an RTC IO");

    ESP_LOGI(TAG, "test---2");

    //IE引脚初始化为输出
    rtc_gpio_init(PIR_IE_GPIO_ULP);
    rtc_gpio_set_direction(PIR_IE_GPIO_ULP, RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_pulldown_dis(PIR_IE_GPIO_ULP);
    rtc_gpio_pullup_dis(PIR_IE_GPIO_ULP);
    rtc_gpio_hold_en(PIR_IE_GPIO_ULP);
    
    ESP_LOGI(TAG, "test---3");

    //IR引脚初始化为输入
    rtc_gpio_init(PIR_IR_GPIO_ULP);
    rtc_gpio_set_direction(PIR_IR_GPIO_ULP, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_dis(PIR_IR_GPIO_ULP);
    rtc_gpio_pullup_dis(PIR_IR_GPIO_ULP);
    rtc_gpio_hold_en(PIR_IR_GPIO_ULP);

    ESP_LOGI(TAG, "test---4");
    
    //断开GPIO12和15的连接，通过上拉或者下拉来降低功耗
    rtc_gpio_isolate(GPIO_NUM_12);
    rtc_gpio_isolate(GPIO_NUM_15);

    

    //设置唤醒ULP的时间，也就是ULP工作的周期
    ulp_set_wakeup_period(0, 10 * 1000 * 1000);       //10 * 1000ms唤醒一次
    
    ESP_LOGI(TAG, "test---5");
    
    //校验ULP内部时钟
    // calibrate 8M/256 clock against XTAL, get 8M/256 clock period
    uint32_t rtc_8md256_period = rtc_clk_cal(RTC_CAL_8MD256, 100);
    uint32_t rtc_fast_freq_hz = 1000000ULL * (1 << RTC_CLK_CAL_FRACT) * 256 / rtc_8md256_period;
    ESP_LOGW(TAG, "RTC FAST Freq Hz = %d kHz", (rtc_fast_freq_hz / 1000));
    

    //开始执行汇编程序
    ret = ulp_run(&ulp_entry - RTC_SLOW_MEM);       //程序存储在RTC SLOW 内存中
    if (ESP_OK != ret)
    {
        ESP_LOGE(TAG, "failed to run ulp bin program");
    }
    else
    {
        ESP_LOGW(TAG, "success to run ulp bin program");
    }
}

```

Assembler code:
```
	
	//初始化变量
	MOVE R1, 0				//R1存放计算到的IR引脚为目标电平的次数
	MOVE R2, 0				//外层循环变量i
LOOP_FOR_i:
	ADD R2, R2, 1			//i++, 因为JUMP指令太多，所以放在前面
	SUB R0, R2, 50		//外层循环决定了载波周期的个数
	JUMP LOOP_END, EQ		//if (j == 50)，循环退出

	/* 内层循环 */
	//初始化循环变量
	MOVE R3, 0				//j循环
LOOP_FOR_j:	
	ADD R3, R3, 1		//j++，因为JUMP指令太多，所以放在前面
	SUB R0, R3, 4			//R0 = R3 - 4, if (j < 4)，初始值为1
	JUMP LOOP_FOR_i, EQ		//if (j == 4)
	
	SUB R0, R3, 1			//R0 = R3 - 1; temp = j - 1
	JUMP IE_OUT_HIGH, EQ		//if (j == 1)
IE_OUT_LOW:
	WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 0)		//向RTC_GPIO6（GPIO25）写入0						//设置GPIO25输出高电平
	WAIT 1000				//延时10个周期，注意这个流程执行的是先输出高电平或者低电平，然后JUMP到内层循环开始处

#if 0
	
	//判定GPIO33是否是高电平
	READ_RTC_REG(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT_S + 8, 1)		//读取GPIO33--RTC_GPIO8的引脚电平到R0
	JUMPR IR_IN_HIGH, 1, GE		//if (R0 >= 1)
	JUMP LOOP_FOR_j
#else

	JUMP LOOP_FOR_j

#endif

IE_OUT_HIGH:
	WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 1)		//向RTC_GPIO6（GPIO25）写入1						//设置GPIO25输出高电平
	WAIT 1000

#if 0	
	//判定GPIO33是否是高电平
	READ_RTC_REG(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT_S + 8, 1)		//读取GPIO33--RTC_GPIO8的引脚电平到R0
	JUMPR IR_IN_HIGH, 1, GE		//if (R0 >= 1) -- 高电平
	JUMP LOOP_FOR_j
#else

	JUMP LOOP_FOR_j			//返回内层循环判定

#endif
	
IR_IN_HIGH:				//如果是高电平，R1++
	ADD R1, R1, 1
	JUMP LOOP_FOR_j

LOOP_END:		//R1存放的是IR引脚计数值,导出R1，R2，R3 -- 测试用

#if 0

	MOVE R0, num_i
	LD R2, R0, 0
	MOVE R0, num_j
	LD R3, R0, 0
	MOVE R0, num_ir
	LD R1, R0, 0

	MOVE R0, R1		//R0=R1=记录的IR电平次数值
	//JUMPR WAKE_UP_CPU, 50, GE		//if (R0 > 50)则唤醒CPU，纯理论上R1≈(50 -1 - 10) *3 = 117
	
#else	

	WAKE
	HALT		//停止ULP，ULP唤醒计时器开始计时
	
#endif

WAKE_UP_CPU:
	WAKE
	HALT		//启动CPU的时候停止ULP程序

```

In addition, I wrote a simulation program to control the ULP to execute only once. Wake up the CPU first, then stop the ULP, and no longer let the ULP wake-up timer work, as follows:

```
	MOVE R1, 10
LOOP_TEST:
	WAIT 200
	SUB R1, R1, 1
	SUB R0, R1, 1
	JUMP LOOP_TEST_END, EQ
	JUMP LOOP_TEST
	
LOOP_TEST_END:
	WAKE
	//REG_WR RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0	
	//WRITE_RTC_REG(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 1, 0);		//停止ULP
	WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0);		//停止ULP

	HALT		
	

```

However, using these two commands can not stop the ULP, the CPU will be periodically awakened by the ULP:
```
REG_WR RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0	
WRITE_RTC_REG(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 1, 0);		//停止ULP
```

With this command, the ULP cannot wake up the CPU after execution:
```
WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0);		//停止ULP
```

I want to ask a few points:
1. The information prompts that the ULP can be suspended [link](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-guides/ulp-legacy.html#ulp-program-flow). How to operate, which API should be used to implement this function in the main program (C language)?
[image]

2. As mentioned above, you can use the `SLEEP` instruction to switch the ULP timer. Is there a corresponding C language function?
[image]

3. Does this `macro`(**RTC_CAL_8MD256**) say that the working clock of ULP is 8MHz/256=31.25KHz, but why is my calibration getting 8MHz?
[image]
```
    uint32_t rtc_8md256_period = rtc_clk_cal(RTC_CAL_8MD256, 100);
    uint32_t rtc_fast_freq_hz = 1000000ULL * (1 << RTC_CLK_CAL_FRACT) * 256 / rtc_8md256_period;
    ESP_LOGW(TAG, "RTC FAST Freq Hz = %d kHz", (rtc_fast_freq_hz / 1000));  
```

4. If the working clock of the ULP is **31.25KHz**, can it be improved?
5. Is there a problem with this cross-compilation chain prompt?
[image]

6. If the ULP is executing the assembler A.S, and the main program wants the ULP to execute B.S, what should I do? Because I repeated the assembly load ulp_load and ulp_run will get an error:
[image]

negativekelvin, 2019-11-12:
You are not using REG_WR and WRITE_RTC_REG properly.  These are equivalent:

```
WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)
WRITE_RTC_REG(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 1, 0)
REG_WR (((RTC_CNTL_STATE0_REG) - DR_REG_RTCCNTL_BASE) / 4), RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0
```

GitOfLance, 2019-11-13:
Hi, @negativekelvin 
Thank you for your answer. After I modified the code according to your suggestion, I can achieve the effect of **stopping ULP**. But in the code, I first call `WAKE`, then the operation register `disables the ULP wake-up register`, and finally calls `HALT` to exit the program. The ULP wake-up function is also enabled in the main program, but the actual effect is not awakened. As shown below:
**WAKE-DISABLE-HALT**:
```
	MOVE R1, 10
LOOP_TEST:
	WAIT 200
	SUB R1, R1, 1
	SUB R0, R1, 1
	JUMP LOOP_TEST_END, EQ
	JUMP LOOP_TEST
	
LOOP_TEST_END:
	WAKE
	//REG_WR RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0	//错误写法
	//REG_WR (((RTC_CNTL_STATE0_REG) - DR_REG_RTCCNTL_BASE) / 4), RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0
	//WRITE_RTC_REG(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 1, 0);		//停止ULP，参数不对，传入的应该是24，而不是BIT24
	WRITE_RTC_REG(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 1, 0);		//停止ULP
	//WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0);		//停止ULP

	HALT	
```
[image]

**main.c**:
[image]

**Code effect**：
[image]

negativekelvin, 2019-11-13:
https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/examples/system/ulp/main/ulp/pulse_cnt.S#L150-L153

GitOfLance, 2019-11-13:
Oh, I understand, I ignore this passage:
[Note that before using WAKE instruction, ULP program may needs to wait until RTC controller is ready to wake up the main CPU. This is indicated using RTC_CNTL_RDY_FOR_WAKEUP bit of RTC_CNTL_LOW_POWER_ST_REG register. If WAKE instruction is executed while RTC_CNTL_RDY_FOR_WAKEUP is zero, it has no effect (wake up does not occur).](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-guides/ulp_instruction_set.html#wake-wake-up-the-chip)
Thanks！

Alvin1Zhang, 2019-11-13:
@negativekelvin Thanks for the notes. @GitOfLance Thanks for reporting, has your issue been resolved? Thanks.

GitOfLance, 2019-11-13:
Yes, I have other questions. How do I manipulate RTC GPIO with assembly? I used the following codes to let `RTC_GPIO6` output a square wave, but no level change is detected on the pin:
```
	//WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 0)		//向RTC_GPIO6（GPIO25）写入0	//设置GPIO25输出高电平
	WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 0)		//向RTC_GPIO6（GPIO25）写入0	//设置GPIO25输出高电平

...
	//WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 1)		//向RTC_GPIO6（GPIO25）写入1	//设置GPIO25输出高电平
	WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG, RTC_GPIO_OUT_DATA_S + 6, 1, 1)		//向RTC_GPIO6（GPIO25）写入0	//设置GPIO25输出高电平
```

What should I do?

negativekelvin, 2019-11-13:
https://github.com/espressif/esp-iot-solution/blob/40cec135eace36cf11ab72988e27546a2c0d025b/examples/ulp_examples/ulp_watering_device/main/main.c#L98-L100

https://github.com/espressif/esp-iot-solution/blob/40cec135eace36cf11ab72988e27546a2c0d025b/examples/ulp_examples/ulp_watering_device/main/ulp/adc.S#L107-L109

GitOfLance, 2019-11-13:
I understand, I did not comment out the function `rtc_gpio_hold_en`！
Does this `macro`(**RTC_CAL_8MD256**) say that the working clock of ULP is 8MHz/256=31.25KHz, but why is my calibration getting 8MHz?
   [image]

```
    uint32_t rtc_8md256_period = rtc_clk_cal(RTC_CAL_8MD256, 100);
    uint32_t rtc_fast_freq_hz = 1000000ULL * (1 << RTC_CLK_CAL_FRACT) * 256 / rtc_8md256_period;
    ESP_LOGW(TAG, "RTC FAST Freq Hz = %d kHz", (rtc_fast_freq_hz / 1000));  
```
What is the working clock of ULP, is it 8MHz or 31.25KHz? If the working clock of the ULP is **31.25KHz**, can it be improved?

negativekelvin, 2019-11-13:
8mhz is correct

GitOfLance, 2019-11-13:
OK, I got it. Thanks!

closed 2021-07-20 by igrr