espressif/esp-idf #4328: BLE stack crash when using ble_gattc_notify

nyronium, 2019-11-12:
## Environment
- Development Kit:      none
- IDF version: v4.1-dev-815-ga45e99853
- Build System:         CMake
- Operating System:     Linux

## Problem Description

Followup to issue #4012.

When using the NimBLE bluetooth stack, the system will crash when using `ble_gattc_notify()` a lot. [I have provided an example project to isolate the issue](https://github.com/nyronium/esp_notify_error_poc).

Due to a logger in the BLE stack in `components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c` (line 490 on current esp-idf master), the bug is not triggered very often. The logging takes so long that the system can often keep up with emptying the queue. But the crash will occur after 5-10 seconds.

To make debugging easier, comment that line out to disable the logger completely. Then the crash is guaranteed.

On that note: Why is there a logger anyway? And it cannot really be disabled (or have i missed it?). This introduces a lot of unnecessary CPU time and increases BLE latency.

### Expected Behavior

The system should never crash, just return an error in notify. Even better would be if
you could query the amount of bytes the queue can still accepts. This would allow applications to stop notifying while the queue cannot hold another packet without having any code run through error handling.

### Actual Behavior

Crash due to core panic: StoreProhibited.

### Steps to repropduce

I have created a contrived example with no dependencies to help you debugging the issue. [You can find it here](https://github.com/nyronium/esp_notify_error_poc).

For completeness here a quick description of what to do:
Call `ble_gattc_notify();` in quick succession (2-3 times) on a suitable characteristic. The characteristic must be subscribed by at least one device to force a tx transaction.

Example:
1. create 16 bytes characteristic
2. connect a device which subscribes to it
3. call `ble_gattc_notify()` for the connection and characteristic in a loop: `while (true) { notify(); }`.

prasad-alatkar, 2019-11-13:
Hi @nyronium 

Thank you for sharing your test code, I was able to reproduce the issue very easily with your code, not to mention not having worked on .cpp codes it got a bit tricky there :sweat_smile: .

 

> On that note: Why is there a logger anyway? And it cannot really be disabled (or have i missed it?). This introduces a lot of unnecessary CPU time and increases BLE latency.

We have ported NimBLE and we have tried to keep `esp_nimble` inline with upstream NimBLE as much as possible. However you can always customize logging from menuconfig option `component config --> Log output --> Default Log verbosity`, in your case it must be set to default `INFO`, please try to change it as per your need.

Now coming to the issue, I see you are sending notifications (`ble_gattc_notify`) in a very tight while loop. I think synchronization can be achieved from application end here. Please try below small patch with your code, I hope it might help you overcome the crash issue. 
[Notify_continuous_while_loop.txt](https://github.com/espressif/esp-idf/files/3843134/Notify_continuous_while_loop.txt)

nyronium, 2019-11-14:
Hi @prasad-alatkar,

Thanks for the hint with the logging configuration, I didn't think it would be tied to the same loglevel as the `ESP_LOG*` macros.

Also thanks for the patch, I see you are solving this on the application side by introducing a semaphore to block further calls to notify until the previous one is sent. In my original application I am heavily relying on low latency BLE (many small MIDI packets of 5 bytes), and the semaphore really increases the average time needed to send a packet, which is not optimal for me. Are there some best practices for low-latency NimBLE?

Also, it isn't quite clear to me if you consider the reported crash a bug in the BLE stack or not.

prasad-alatkar, 2019-11-18:
Hi @nyronium , one solution can be after sending notification, check return code of `ble_gattc_notify`, if the error code is not `0` add a delay as per your application. You can also increase `ACL buffer count` from menuconfig (`Component config → Bluetooth → NimBLE Options → ACL Buffer count`). 

To answer your question regarding crash, yes this crash should not happen, I am working to resolve the crash.

prasad-alatkar, 2019-12-03:
Hi @nyronium Can you please pull latest master and let us know if the crash issue is resolved ?

prasad-alatkar, 2019-12-16:
Hi @nyronium the fix is available on github master, commit id : 656005ef

nyronium, 2019-12-16:
Hi @prasad-alatkar. Thanks for addressing the issue, in my first tests I haven't encountered any further problems. I will report back, if I notice anything problematic in the future. Thanks.

closed 2019-12-16 by nyronium