espressif/esp-idf #4329: Windows WSL esp-idf install.sh error

LeongHouWang, 2019-11-12:
I am following this guide
[https://docs.espressif.com/projects/esp-idf/en/latest/get-started/#get-started-set-up-tools](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/#get-started-set-up-tools)
 to install esp-idf on my windows wsl until Step3 setting up the tools by launching ./install.sh . About the error code I cannot find solutions on the internet yet.

My Windows WSL is Ubuntu 18.04.3 LTS and it is a fresh install solely for esp32.

Below is the error message I copied from the terminal.

```
Installing ESP-IDF tools                                                                                        
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, openocd-esp32                             Installing xtensa-esp32-elf@esp32-2019r1-8.2.0                                                                  Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz to /home/leong/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz.tmp                                                              Traceback (most recent call last):                                                                                
File "/home/leong/esp/esp-idf/tools/idf_tools.py", line 1346, in <module>                                         main(sys.argv[1:])                                                                                            
File "/home/leong/esp/esp-idf/tools/idf_tools.py", line 1342, in main                                             action_func(args)                                                                                             
File "/home/leong/esp/esp-idf/tools/idf_tools.py", line 1136, in action_install                                   tool_obj.download(tool_version)                                                                               
File "/home/leong/esp/esp-idf/tools/idf_tools.py", line 570, in download                                          urlretrieve(url, local_temp_path, report_progress if not global_non_interactive else None)                    File "/usr/lib/python2.7/urllib.py", line 98, in urlretrieve                                                      
return opener.retrieve(url, filename, reporthook, data)                                                       
File "/usr/lib/python2.7/urllib.py", line 248, in retrieve                                                        
fp = self.open(url, data)                                                                                     
File "/usr/lib/python2.7/urllib.py", line 216, in open                                                            
return getattr(self, name)(url)                                                                               
File "/usr/lib/python2.7/urllib.py", line 446, in open_https                                                      h.endheaders(data)                                                                                            
File "/usr/lib/python2.7/httplib.py", line 1054, in endheaders                                                    self._send_output(message_body)                                                                               
File "/usr/lib/python2.7/httplib.py", line 892, in _send_output                                                   self.send(msg)                                                                                                
File "/usr/lib/python2.7/httplib.py", line 854, in send                                                           self.connect()                                                                                                
File "/usr/lib/python2.7/httplib.py", line 1279, in connect                                                       server_hostname=server_hostname)                                                                              
File "/usr/lib/python2.7/ssl.py", line 369, in wrap_socket                                                        _context=self)                                                                                                
File "/usr/lib/python2.7/ssl.py", line 617, in __init__                                                           self.do_handshake()                                                                                           
File "/usr/lib/python2.7/ssl.py", line 846, in do_handshake                                                       self._sslobj.do_handshake()                                                                                 
IOError: [Errno socket error] [Errno 0] Error 
```

Thank you in advance.

Alvin1Zhang, 2019-11-12:
@LeongHouWang Thanks for reporting.

dobairoland, 2019-11-13:
Hi @LeongHouWang. Could you please try it again? The download server was down around the time you've reported this issue.

LeongHouWang, 2019-11-15:
Hi, yes, yesterday I have finished the installation. Thanks for the support

closed 2019-11-18 by igrr