espressif/esp-idf #4330: [BLE Mesh] How should I test the friendship and low power node functions?

zhanlej, 2019-11-13:
I simply read your SDK code and think it should support the `friendship` and `low power` functions. But for these two functions I have the following problems:

- Because I don't know much about these two functions, so when I use `menuconfig` to configure these two functions, I don't understand many parameter settings.
- How should the friendship node be designed? Is it only necessary to enable the friendship function?
- How should the low power node be designed? Does the device need to be designed with low power or sleep mode? Does your ESP32 support the above model?

Can you provide new examples for the above mentioned issues and explain in detail?

In which branch is the latest version of the ble mesh code? Is it in the `master` branch or the `ble_mesh_release/esp-ble-mesh-v0.6.1` branch?

Campou, 2019-12-02:
Hi @zhanlej 

Sorry for the late reply.

Currently for Friend node, it only needs to enable the Friend functionality, and when the Low Power node enters the network, it will try to find proper Friend node and establish a friendship.

For Low Power node, after the friendship is established, the BLE Scan functionality will be disabled currenlty, and it will be re-enabled after a certain delay. The Low Power node is useful when the device is a sensor or some other things which need low power consumption.

We will provide corresponding examples to illustrate how to use the friendship functionality later.

Thanks & Regards

Campou, 2020-05-27:
Hi @zhanlej 

Sorry for the late reply.

Any updates about the testing progress?

Thanks.

Alvin1Zhang, 2020-07-27:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-07-27 by Alvin1Zhang

Thecave3, 2021-04-20:
Hi @Campou , 

> Currently for Friend node, it only needs to enable the Friend functionality, and when the Low Power node enters the network, it will try to find proper Friend node and establish a friendship.
>
> For [...]

Where this behaviour is described?
Is this behavior compliant with the standard? 
 
> We will provide corresponding examples to illustrate how to use the friendship functionality later.

Are these examples available to everyone? 
Is it possible to have them? 

Thanks & Regards

Shane555, 2022-08-01:
Hi, are there any ways/examples to test friendship node and LPN? How would i know in a mesh if they are working as expected according to the bluetooth mesh specifications

ZacDaMan, 2022-11-04:
Are there examples for this yet? I'm similarly stuck.

Paramesh19, 2023-03-30:
@xiongweichao Hi Is there anyone who worked on LPN and friend node  Kindly provide the feedback on how to use those features...
Thank you in advance....