espressif/esp-idf #4331: Flashing error with latest idf under OsX

Francois-Belanger, 2019-11-13:
- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-815-ga45e99853
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [external Battery]

Unable to program after updating all the ESP tools.

esptool.py -p /dev/cu.SLAB_USBtoUART -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x1000 bootloader/bootloader.bin 0x10000 calimotion.bin
esptool.py v2.9-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........___
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:47:54:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 136...

A fatal error occurred: Timed out waiting for packet content
esptool.py failed with exit code 2
--- Build failed
--- Press Ctrl+] to exit monitor.
--- Press Ctrl+F to build & flash project.
--- Press Ctrl+A to build & flash app.
--- Press any other key to resume monitor (resets target).

Trying to program same board than before fail with the latest build.

projectgus, 2019-11-13:
Hi @Francois-Belanger,

The failure indicates most likely a hardware issue (could be power supply or serial signal integrity.)

If you set the baud rate back to something lower (the old default was 115200bps), does it start working?

Angus

Francois-Belanger, 2019-11-13:
I will redo some tests.

Francois-Belanger, 2019-11-13:
Power supply is very stable at 3.0V
Uarts lines all clean.
Could it be an issue with the build ?  Bad file in my build directory ?  Bad config ?
I was using the WROVER fine on the same boards, just change it to WROBER-B and it doesn't work anymore...

Francois-Belanger, 2019-11-13:
ok, found the issue.  It's line MTDI that was at logic 1 instead of 0.
Thanks for the help.

closed 2019-11-13 by Francois-Belanger