espressif/esp-idf #4333: Allow sharing SPI signals between SD card and other SPI devices

satyanraj, 2019-11-13:
On our PCB/hardware, the SPI bus is shared between SD card and other devices because of the limited number of I/Os. This is currently not possible with esp-idf and either SD card works only OR the other device. Is this because of the SD card driver?

koobest, 2019-11-13:
Yes, the SD card will monopolize one SPI.  Another solution is to use HSPI and VSPI, sharing CLK, MOSI and MISO,

satyanraj, 2019-11-13:
> Another solution is to use HSPI and VSPI, sharing CLK, MOSI and MISO

Is it possible to use both HSPI and VSPI with same pins for CLK, MIOS & MOSI?
Or 
Do we need a different set of ESP32 pins for HSPI and VSPI?

iosix, 2019-11-13:
No - if you initialize the SD first you can then add peripherals to the bus and all works seamlessly. You just can’t initialize the bus by itself first. 

Robert Vogt IV
CEO
IOSiX, LLC
1300 Tefft Ct #1
Saline, MI  48176
P: +1-855-OBD-1939
C: +1-734-730-9690
robert@iosix.com

koobest, 2019-11-18:
Hi, @satyanraj 

The following is a API to exchange SPI signals.
```
// host = HSPI_HOST or VSPI_HOST
void spi_pin_attach(int host)
{
    // change this pins according to your board
    const int spi_mosi = 19;
    const int spi_miso = 22;
    const int spi_clk = 21;

    // MOSI
    gpio_set_direction((gpio_num_t)spi_mosi, GPIO_MODE_INPUT_OUTPUT);
    gpio_matrix_out(spi_mosi, spi_periph_signal[host].spid_out, false, false);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[spi_mosi], 2);

    // MISO
    gpio_set_direction((gpio_num_t)spi_miso, GPIO_MODE_INPUT);
    gpio_matrix_in(spi_miso, spi_periph_signal[host].spiq_in, false);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[spi_miso], 2);

    // CLK
    gpio_set_direction((gpio_num_t)spi_clk, GPIO_MODE_INPUT_OUTPUT);
    gpio_matrix_out(spi_clk, spi_periph_signal[host].spiclk_out, false, false);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[spi_clk], 2);
}
```
For example, if the SD card uses VSPI and the other device uses HSPI, when accessing the SD card, spi_pin_attach (VSPI) should be called first, and then read or write SD card.

[image]

koobest, 2019-12-08:
Hi, @satyanraj 
Has your issue been resolved ?
thanks !!

Alvin1Zhang, 2020-01-08:
@satyanraj Thanks for reporting, would you please help share if any updates for this issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-05-27:
@satyanraj Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates. Thanks.

closed 2020-05-27 by Alvin1Zhang