espressif/esp-idf #4335: How to free the RAM memory?

Kirankumar-xenlabs, 2019-11-13:
I want to free the  20kb RAM, how can do it? 

Thanks and Regards 
Kiran

negativekelvin, 2019-11-13:
Be more specific

Kirankumar-xenlabs, 2019-11-13:
Thank you for response, I am using esp32, in my project I am using wifi and ble. I am capturing the image and uploading it on google cloud,but It will not happening, for this it needs more RAM, for this task I want to free the some RAM.

negativekelvin, 2019-11-13:
Please move this discussion to [esp32.com](http://esp32.com) forum

Kirankumar-xenlabs, 2019-11-13:
I am getting following error,
E (271678) esp-tls: mbedtls_ssl_handshake returned -0x4d80
E (271678) esp-tls: Failed to open new connection
E (271678) TRANS_SSL: Failed to open a new connection
E (271678) HTTP_CLIENT: Connection failed, sock < 0

Alvin1Zhang, 2019-11-14:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. For "how" questions and third party issues, please help submit a post on our forum at https://esp32.com. Thanks.

closed 2019-11-15 by Kirankumar-xenlabs