espressif/esp-idf #4336: sdio_slave_recv does not work with byte mode sending

Martvvliet, 2019-11-13:
## Environment

- Module or chip used: [ESP32-PICO-D4]
- IDF version v4.0
- Build System:         [CMake]
- Compiler version:  [xtensa-esp32-elf-gcc 8.2.0]
- Operating System:     [Windows]
- Power Supply:         -

## Problem Description

The ESP is configured as a SDIO slave device. When I try to use the sdio_slave_recv() function only works if I use block mode or bytemode where the number of bytes is the same as the blocksize. If I use the byte mode with a number of bytes, lower the the blocksize, the functions does not get interrupted, released by the SDIO interrupt, so no data is received. I read in the examples that this is possible, but how. Should this be enabled in some way? I did check the flags in the sdio_intr() ISR and I can see that a flag from SLC.slc0_int_st.val is not set when receiving the wrong number of bytes.

### Expected Behavior
sdio_slave_recv() reads data over the SDIO bus in byte mode. The number of bytes can be less than the blocksize

### Actual Behavior
sdio_slave_recv() only receives data when in blockmode or in byte mode when the number of bytes is equal or more than the block size. If reduce the blocksize, the symptoms are the same

### Steps to repropduce

### Code to reproduce this issue

```cpp
sdio_slave_config_t config = {
        .sending_mode     = SDIO_SLAVE_SEND_PACKET,
        .send_queue_size  = SDIO_SLAVE_QUEUE_SIZE,
        .recv_buffer_size = BUFFER_SIZE,
        .event_cb         = sdio_interrupt,
        //.flags              = SDIO_SLAVE_FLAG_INTERNAL_PULLUP,
    };

    ret = sdio_slave_initialize(&config);
    set_error_info(ret, "sdio_init, sdio_slave_initialize", TAG_SDIO);

    slave_write_reg(0, JOB_IDLE);

    sdio_slave_buf_handle_t handle;
    for(int i = 0; i < BUFFER_NUM; i++)
    {
        handle = sdio_slave_recv_register_buf(buffer[i]);
        assert(handle != NULL);

        ret = sdio_slave_recv_load_buf(handle);
        set_error_info(ret, "sdio_init, sdio_slave_recv_load_buf", TAG_SDIO);
    }

    sdio_slave_set_host_intena(SDIO_SLAVE_HOSTINT_BIT0 | SDIO_SLAVE_HOSTINT_BIT1 |
                               SDIO_SLAVE_HOSTINT_BIT2 | SDIO_SLAVE_HOSTINT_BIT3 |
                               SDIO_SLAVE_HOSTINT_BIT4 | SDIO_SLAVE_HOSTINT_BIT5 |
                               SDIO_SLAVE_HOSTINT_BIT6 | SDIO_SLAVE_HOSTINT_BIT7);
    sdio_slave_start();
    sdioInitialised = true;
```

ginkgm, 2019-11-14:
thanks for your report, we will check this

ginkgm, 2019-11-14:
What's the host/master you are using? Could you please provide the code/process of master side for us to reproduce the issue?

Especially for the deatils of each CMD53 command you send from master. (function, op, address, length)

Martvvliet, 2019-11-14:
Thank you for your repsonse, I use a NXP i.MX ARM processor as host. Initialisation is done with the following code:
`_card.host.base              = BOARD_ESP_PERIPHERAL;
    _card.host.sourceClock_Hz    = BOARD_ESP_HOST_CLK_FREQ;
    _card.host.config.endianMode = kUSDHC_EndianModeLittle;

    static status_t stat = SDIO_HostInit(&_card);
    if(stat != kStatus_Success)
    {
        LOGE("Could not initialize the SDIO interface with the ESP: %ld", stat);
        return -EIO;
    }

    // stat = SDIO_CardReset(&_card);
    // if(stat != kStatus_Success)
    // {
    //     LOGE("Could not reset the SDIO interface with the ESP: %ld", stat);
    //     return -EIO;
    // }

    static sdmmchost_card_int_t card_int;
    card_int.cardInterrupt = ESP_SDIO_interrupt_callback;
    _card.usrParam.cardInt = &card_int;

    stat = SDIO_CardInit(&_card);
    if(stat != kStatus_Success)
    {
        LOGE("Could not initialize the SDIO interface with the ESP: %ld", stat);
        return -EIO;
    }

    stat = SDIO_SetDataBusWidth(&_card, kSDIO_DataBus4Bit);
    if(stat != kStatus_Success)
    {
        LOGE("Could not set the data buswidth: %ld", stat);
        return -EIO;
    }

    stat = SDIO_EnableIO(&_card, kSDIO_FunctionNum1, true);
    if(stat != kStatus_Success)
    {
        LOGE("Could not enable the ESP function 1: %ld", stat);
        return -EIO;
    }

    SDIO_SetIOIRQHandler(&_card, kSDIO_FunctionNum1, SDIO_IRQ_handler);
    stat = SDIO_EnableIOInterrupt(&_card, kSDIO_FunctionNum1, true);
    if(stat != kStatus_Success)
    {
        LOGE("Could not enable the IO interrupt: %ld", stat);
        return -EIO;
    }

    stat = SDIO_SetBlockSize(&_card, kSDIO_FunctionNum1, 512);
    if(stat != kStatus_Success)
    {
        LOGE("Could not set the SDIO function 1 blocksize to 512: %ld", stat);
        return -EIO;
    }

    // Read function capabilities
    for(int i = 1; i < 8; i++)
    {
        stat = SDIO_GetCardCapability(&_card, static_cast<sdio_func_num_t>(i));
        if(kStatus_Success != stat)
        {
            LOGW("Could not read card capability for function %d: %ld", i, stat);
        }
    }`

Sending is done with the following lines

`status_t stat = SDIO_IO_Write_Extended(&_card, kSDIO_FunctionNum1, 0x1f800 - count, buffer,
                                           count, flags);`
When settings _flags_ to 0, it works in byte mode. while debugging the NXP processor. I use FuncNum1 because I designed my own protocol. 
The write extended function uses the following structure to store the arguments before actually sending it over SDIO. 
`typedef struct _usdhc_command
{
    uint32_t index;                          /*!< Command index */
    uint32_t argument;                       /*!< Command argument */
    usdhc_card_command_type_t type;          /*!< Command type */
    usdhc_card_response_type_t responseType; /*!< Command response type */
    uint32_t response[4U];                   /*!< Response for this command */
    uint32_t responseErrorFlags;             /*!< response error flag, the flag which need to check
                                                 the command reponse*/
    uint32_t flags;                          /*!< Cmd flags */
} usdhc_command_t;`

the argument and index are filled with the following code:
`command.index = kSDIO_RWIOExtended;
    command.argument =
        (func << SDIO_CMD_ARGUMENT_FUNC_NUM_POS) |
        ((regAddr & SDIO_CMD_ARGUMENT_REG_ADDR_MASK) << SDIO_CMD_ARGUMENT_REG_ADDR_POS) |
        (1U << SDIO_CMD_ARGUMENT_RW_POS) | (count & SDIO_EXTEND_CMD_COUNT_MASK) |
        ((blockMode ? 1 : 0) << SDIO_EXTEND_CMD_ARGUMENT_BLOCK_MODE_POS |
         ((opCode ? 1 : 0) << SDIO_EXTEND_CMD_ARGUMENT_OP_CODE_POS));
    command.responseType = kCARD_ResponseTypeR5;
    command.responseErrorFlags =
        (kSDIO_StatusCmdCRCError | kSDIO_StatusIllegalCmd | kSDIO_StatusError |
         kSDIO_StatusFunctionNumError | kSDIO_StatusOutofRange);`

 I compared the _command.argument_ values when sending the working and the not working data transfers and count did correspond to the count argument in both cases.

Martvvliet, 2019-11-20:
Hmm, it works when I turn on the opCode. Is this something ESP specific or did I miss something in the SDIO protocol?

ginkgm, 2019-11-22:
Hi @Martvvliet ,
This is ESP specific and written in the TRM (see section 8.3.2 figure 24, not very clear though). When you use the DMA part of address on Function 1, you also need to specify the OP=1.

Martvvliet, 2019-11-26:
Hi @ginkgm, I see. It might be helpful for the future to explain it a bit on the SDIO slave driver page. This page contains good information and is very useful as reference while programming but didn’t explain this information. Anyway thank you for helping 😄

closed 2019-11-26 by Martvvliet