espressif/esp-idf #4337: Bluetooth classic role switch fails for reconnecting HID devices

nickpas, 2019-11-13:
Hello,

In my project I want to connect HID devices to ESP32 via bluetooth classic and have them automatically reconnect after powering them off & on again. (according to standards for HID devices)

I'm coming here after previous discussion held here: https://groups.google.com/forum/#!topic/btstack-dev/eWqUROqFBV0

## Environment

- Development Kit: LOLIN32
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v3.2.3-37-gd8e0e4835
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

Bluetooth classic role switch fails with several HID devices when reconnecting to ESP32.
(I'm not sure it is specific to HID devices, could be in general for bluetooth peripherals connecting back. My target is HID devices and I have so far only tested with such devices.)

### Expected Behavior

Role switch succeeds after accepting the incoming connection request (where the accept includes the request to become master).

### Actual Behavior

For following devices this fails:
- Dell mouse with broadcom chip
- Apple mouse with Apple chip
- Wiimote with broadcom chip

It works for following device:
- VRtech with csr chip

The packet logs show that in all cases all commands given are equal but the result is not.

### Steps to reproduce

After discovering a HID device, I succeed in connecting to it, bond with it and read-out several reports.
When turning off a HID device and turning it back on, the HID device takes the initiative to reconnect with the ESP32. The ESP32 sees in the incoming request and the command is given to accept it with the request to become master. Then the HCI reports back that the role switch failed or succeeded.

1. scan for devices
2. connect to discovered device
3. ask for authentication & handle related steps
4. setup HID control
5. setup HID interrupt
6. push some buttons on the HID device to see incoming HID reports
7. turn off the HID device
8. turn on the HID device
9. accept the incoming connection with the request to become master

## Debug Logs

Please check out this post from the previous discussion for detailed packet logs which can be opened with wireshark: https://groups.google.com/d/msg/btstack-dev/eWqUROqFBV0/CK3N68vKAAAJ

Pointers are given to the relevant packets.

### Comments

In this case I used BTstack as bluetooth stack, but the issue seems to be located in ESP32. I ran the same code on a linux machine with a bluetooth dongle (with broadcom chip) and there everything worked. See this post for comparison: https://groups.google.com/d/msg/btstack-dev/eWqUROqFBV0/eNLiTqEFCwAJ

It is also strange that the same code on ESP32 works for only 1 device and the others not. (The failing HID devices disconnect if the role switch does not succeed. So not much workaround possible at this moment.)

Any help to further pinpoint the issue is highly appreciated!

bye,
Nick

Alvin1Zhang, 2020-01-08:
@nickpas Thanks for reporting, and sorry for getting back to you so late. We have one fix internally under reviewing, will inform once the fix is available on GitHub. Thanks.

mringwal, 2020-02-06:
@nickpas Interestingly, when ESP32 is initiating a new connection and the remote triggers a role change, the connection fails at least for my iPhone SE. see https://github.com/espressif/esp-idf/issues/4695

Maybe related... :)

nickpas, 2020-03-21:
@Alvin1Zhang Thanks for picking this one up. Do you have any updates on the progress? Is the fix in the latest IDF? (didn't try it yet as my project is on hold due to this blocker)

@mringwal I didn't try this yet, but it seems indeed a very similar issue. Let's  hope both are solved with the above planned fix. :)

mringwal, 2020-03-21:
@nickpas The role switch on ESP32 from master to slave works with current master branch. Please retry your code and let us know...

nickpas, 2020-03-22:
I did a quick & dirty upgrade to master and latest BTstack, and it seems to be working!
Thx!

Alvin1Zhang, 2020-03-23:
@nickpas Thanks for feedback and glad your issue gets resolved. Thanks.

closed 2020-03-23 by Alvin1Zhang