espressif/esp-idf #4339: 蓝牙客户端注册通知问题？

Shaopus, 2019-11-14:
你好

我先碰到一个问题，在`ble gattc_demo`上修改，连接了一个蓝牙设备，这个蓝牙设备特征值有很多通知的权限，然后我执行`esp_ble_gattc_register_for_notify`，注册了7个特征值后，就出现以下错误。
`BT_APPL: Max Notification Reached, registration failed.`

请问这是什么问题呢？期待尽快回答，谢谢！

Shaopus, 2019-11-15:
当我注册通知数超过7个后，就出现了这个`E (3934) BT_APPL: Max Notification Reached, registration failed.`，请问这个是蓝牙协议栈限制的吗？

GYC-Espressif, 2019-12-04:
这个限制只是为了节省内存，如果你需要增加注册通知的数量，可以在
`components/bt/host/bluedroid/bta/gatt/include/bta_gattc_int.h` 
文件中 找到 `BTA_GATTC_NOTIF_REG_MAX ` 的定义进行修改 增加注册通知的最大数量

Shaopus, 2019-12-04:
> 这个限制只是为了节省内存，如果你需要增加注册通知的数量，可以在
> `components/bt/host/bluedroid/bta/gatt/include/bta_gattc_int.h`
> 文件中 找到 `BTA_GATTC_NOTIF_REG_MAX ` 的定义进行修改 增加注册通知的最大数量

收到，感谢！

closed 2019-12-04 by Shaopus