espressif/esp-idf #4341: esp-idf-tools-setup-2.1 Fail to Install - Windows 10

mikeoto, 2019-11-14:
## Environment

- Development Kit:      [ESP32-Wroom]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
Here is my install paths:

**Using Python 3.6:
      C:\Users\Michael Bot\AppData\Local\Programs\Python\Python36-32\python.exe

Using Git 2.21.0.windows.1:
      C:\Program Files\Git\cmd\git.exe

Using existing ESP-IDF copy: 
      C:\esp\esp-idf

IDF tools directory (IDF_TOOLS_PATH):
      C:\esp32

Running esp-idf-tools-setup-2.1.exe, I get errors as follows:

Running command: C:\Users\Michael Bot\AppData\Local\Programs\Python\Python36-32\python.exe C:\esp\esp-idf\tools\idf_tools.py install-python-env
                          Using base prefix 'C:\\Users\\Michael Bot\\AppData\\Local\\Programs\\Python\\Python36-32'
                          New python executable in C:\esptools\.espressif\python_env\idf4.1_py3.6_env\Scripts\python.exe
                          Installing setuptools, pip, wheel...
                          done.
                          Processing \\${idf_path}\tools\kconfig_new\esp-windows-curses
                          ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '\\\\${IDF_PATH}\\tools\\kconfig_new\\esp-windows-curses'
                          
                          Creating a new Python environment in C:\esptools\.espressif\python_env\idf4.1_py3.6_env
                          Installing Python packages from C:\esp\esp-idf\requirements.txt
                          Traceback (most recent call last):
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1349, in <module>
                              main(sys.argv[1:])
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1345, in main
                              action_func(args)
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1163, in action_install_python_env
                              subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
                            File "C:\Users\Michael Bot\AppData\Local\Programs\Python\Python36-32\lib\subprocess.py", line 291, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['C:\\esptools\\.espressif\\python_env\\idf4.1_py3.6_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\esp\\esp-idf\\requirements.

Let me know if there is something I'm missing.

joeynoodlesfang, 2019-11-15:
Second that, same issue.

dobairoland, 2019-11-15:
Are you sure that you have IDF v3.2? Maybe `IDF_PATH` points to the wrong directory?

The failure regarding `esp-windows-curses` indicates that the installer is using the `requirements.txt` file included only in IDF v4.1.

mikeoto, 2019-11-15:
If I was using the IDF Tools Installer, how could I have an incorrect IDF? I am able to get this system running by downloading and installing the parts individually but if you don't do that, the installer doesn't seem to be working correctly.

flufsor, 2019-11-15:
I fixed it with a manually downloaded esp-idf from github, and setting up the IDF_PATH variable pointing to it.
Its a temp fix but works atleast.

mikeoto, 2019-11-15:
my manual install isnt working right. I was able to get it to compile and flash one time but then the path was lost after exiting cmd. When I tried to install again, the install.bat cant find some components and starts to download everything from scratch. Very frustrating. Is there an easier way to do this? Should I stick to using the msys32 instead of the Cmake system?

I think I was using the V4.1 version instead of V3.2, its very hard to find out if Im having problems with installation.

keelung-yang, 2019-12-05:
> I fixed it with a manually downloaded esp-idf from github, and setting up the IDF_PATH variable pointing to it.
> Its a temp fix but works atleast.

@flufsor I tried your solution on Win10 64 bit and Python 3.8 64 bit, but still not works.

@igrr  Maybe it's caused by **esp-2019r2-8.2.0**, esp-idf-tools-setup-2.1.exe works on **r1** version as I remembered.

igrr, 2019-12-05:
No, this is caused by the fact that the installer didn't set IDF_PATH before invoking the Python packages installation process. Since requirements.txt now uses IDF_PATH variable, it leads to this error.

The fix has been merged on our end, but due to some infrastructure issues it hasn't synced to GitHub yet. This issue will be updated when this happens. Until then, please use release/v4.0 branch as a workaround.

closed 2019-12-16 by espressif-bot

804enl, 2020-04-13:
I have been trying to install ESP-IDF with
 https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe
but it fails because windows 10 does have xcopy.
I tried 
https://dl.espressif.com/dl/esp-idf-tools-setup-2.2.exe
But it fails near the finish, because it is missing a file,
Can you please help I have been trying for more then a couple of days.
I was useing Arduino but could not get the esp32 to run bluetooth or WiFi.
I am using a( EASY Kit esp32-C1 ) board.
Can you please help
Kind regards James Webster Email: james.webster2@bigpond.com

igrr, 2020-04-13:
@804enl could you please attach the setup error log you got with esp-idf-tools-setup-2.3.exe?

804enl, 2020-04-13:
OK Thank youAs soon as I get back to my computer Regards James Sent from my Samsung Mobile on the Telstra Mobile Network

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/espressif/esp-idf/issues/4341#issuecomment-613056711",
"url": "https://github.com/espressif/esp-idf/issues/4341#issuecomment-613056711",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

804enl, 2020-04-14:
Hi Ivan
Here is the log 
hope this helps
Regards James Webster

Sent: Tuesday, 14 Apr, 2020 At 5:31 AM

2020-04-14 10:08:02.235   Log opened. (Time zone: UTC+10:00)
2020-04-14 10:08:02.235   Setup version: Inno Setup version 6.0.2 (u)
2020-04-14 10:08:02.235   Original Setup EXE: D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.3.exe
2020-04-14 10:08:02.235   Setup command line: /SL5="$708F2,350580993,721408,D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.3.exe" 
2020-04-14 10:08:02.235   Windows version: 10.0.16299  (NT platform: Yes)
2020-04-14 10:08:02.236   64-bit Windows: Yes
2020-04-14 10:08:02.236   Processor architecture: x64
2020-04-14 10:08:02.247   User privileges: None
2020-04-14 10:08:02.257   Administrative install mode: No
2020-04-14 10:08:02.257   Install mode root key: HKEY_CURRENT_USER
2020-04-14 10:08:02.257   64-bit install mode: Yes
2020-04-14 10:08:02.297   Created temporary directory: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp
2020-04-14 10:08:02.652   -- DLL function import --
2020-04-14 10:08:02.652   Function name: idpAddFile
2020-04-14 10:08:02.652   DLL name: files:idp.dll
2020-04-14 10:08:02.652   Extracting temporary file: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.741   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.741   Importing the DLL function.
2020-04-14 10:08:02.763   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.763   -- DLL function import --
2020-04-14 10:08:02.763   Function name: idpAddFileComp
2020-04-14 10:08:02.763   DLL name: files:idp.dll
2020-04-14 10:08:02.764   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.764   Importing the DLL function.
2020-04-14 10:08:02.764   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.764   -- DLL function import --
2020-04-14 10:08:02.764   Function name: idpAddMirror
2020-04-14 10:08:02.764   DLL name: files:idp.dll
2020-04-14 10:08:02.764   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.764   Importing the DLL function.
2020-04-14 10:08:02.764   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.764   -- DLL function import --
2020-04-14 10:08:02.764   Function name: idpAddFtpDir
2020-04-14 10:08:02.764   DLL name: files:idp.dll
2020-04-14 10:08:02.765   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.765   Importing the DLL function.
2020-04-14 10:08:02.765   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.765   -- DLL function import --
2020-04-14 10:08:02.765   Function name: idpAddFtpDirComp
2020-04-14 10:08:02.765   DLL name: files:idp.dll
2020-04-14 10:08:02.765   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.765   Importing the DLL function.
2020-04-14 10:08:02.765   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.765   -- DLL function import --
2020-04-14 10:08:02.765   Function name: idpClearFiles
2020-04-14 10:08:02.765   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.766   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.766   -- DLL function import --
2020-04-14 10:08:02.766   Function name: idpFilesCount
2020-04-14 10:08:02.766   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.766   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.766   -- DLL function import --
2020-04-14 10:08:02.766   Function name: idpFtpDirsCount
2020-04-14 10:08:02.766   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.767   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.767   -- DLL function import --
2020-04-14 10:08:02.767   Function name: idpFileDownloaded
[... 431 lines cut ...]
                          
                          Folders: 2381
                          Files: 11484
                          Size:       614298370
                          Compressed: 415318076
                          
2020-04-14 10:13:12.300   --------
2020-04-14 10:13:12.307   Copying C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\esp-idf-v4.0 to C:\Users\James\Desktop\esp-idf-7
2020-04-14 10:13:12.550   Done, exit code=1
2020-04-14 10:13:12.551   --------
2020-04-14 10:13:12.551   Running command: cmd.exe /c ""xcopy" /s /e /i /h /q "C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\esp-idf-v4.0" "C:\Users\James\Desktop\esp-idf-7""
                          '"xcopy"' is not recognized as an internal or external command,
                          operable program or batch file.
                          
2020-04-14 10:13:12.551   --------
2020-04-14 10:13:12.551   Message box (OK):
                          Installation has failed with exit code 1
2020-04-14 10:14:29.241   User chose OK.
2020-04-14 10:14:29.256   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-04-14 10:14:30.888   User chose Yes.
2020-04-14 10:14:33.198   CurStepChanged raised an exception.
2020-04-14 10:14:33.456   Need to restart Windows? No
2020-04-14 10:14:33.467   Deinitializing Setup.

804enl, 2020-04-14:
Goog day Ivan 
This is the log from:  esp-idf-tools-setup-2.2.exe
I though this might help
Regards James Webster

2020-04-14 10:30:28.913   Log opened. (Time zone: UTC+10:00)
2020-04-14 10:30:28.914   Setup version: Inno Setup version 6.0.2 (u)
2020-04-14 10:30:28.914   Original Setup EXE: D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.2.exe
2020-04-14 10:30:28.914   Setup command line: /SL5="$F0996,408206375,721408,D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.2.exe" 
2020-04-14 10:30:28.914   Windows version: 10.0.16299  (NT platform: Yes)
2020-04-14 10:30:28.914   64-bit Windows: Yes
2020-04-14 10:30:28.914   Processor architecture: x64
2020-04-14 10:30:28.915   User privileges: None
2020-04-14 10:30:28.916   Administrative install mode: No
2020-04-14 10:30:28.916   Install mode root key: HKEY_CURRENT_USER
2020-04-14 10:30:28.916   64-bit install mode: Yes
2020-04-14 10:30:28.928   Created temporary directory: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp
2020-04-14 10:30:28.940   -- DLL function import --
2020-04-14 10:30:28.940   Function name: idpAddFile
2020-04-14 10:30:28.940   DLL name: files:idp.dll
2020-04-14 10:30:28.941   Extracting temporary file: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.956   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.956   Importing the DLL function.
2020-04-14 10:30:28.974   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.974   -- DLL function import --
2020-04-14 10:30:28.974   Function name: idpAddFileComp
2020-04-14 10:30:28.974   DLL name: files:idp.dll
2020-04-14 10:30:28.975   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.975   Importing the DLL function.
2020-04-14 10:30:28.975   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.975   -- DLL function import --
2020-04-14 10:30:28.975   Function name: idpAddMirror
2020-04-14 10:30:28.975   DLL name: files:idp.dll
2020-04-14 10:30:28.975   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.975   Importing the DLL function.
2020-04-14 10:30:28.975   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.975   -- DLL function import --
2020-04-14 10:30:28.975   Function name: idpAddFtpDir
2020-04-14 10:30:28.975   DLL name: files:idp.dll
2020-04-14 10:30:28.975   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.975   Importing the DLL function.
2020-04-14 10:30:28.975   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.975   -- DLL function import --
2020-04-14 10:30:28.976   Function name: idpAddFtpDirComp
2020-04-14 10:30:28.976   DLL name: files:idp.dll
2020-04-14 10:30:28.976   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.976   Importing the DLL function.
2020-04-14 10:30:28.976   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.976   -- DLL function import --
2020-04-14 10:30:28.976   Function name: idpClearFiles
2020-04-14 10:30:28.976   DLL name: files:idp.dll
2020-04-14 10:30:28.976   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.976   Importing the DLL function.
2020-04-14 10:30:28.976   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.976   -- DLL function import --
2020-04-14 10:30:28.976   Function name: idpFilesCount
2020-04-14 10:30:28.976   DLL name: files:idp.dll
2020-04-14 10:30:28.976   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.976   Importing the DLL function.
2020-04-14 10:30:28.976   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.976   -- DLL function import --
2020-04-14 10:30:28.976   Function name: idpFtpDirsCount
2020-04-14 10:30:28.977   DLL name: files:idp.dll
2020-04-14 10:30:28.977   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
2020-04-14 10:30:28.977   Importing the DLL function.
2020-04-14 10:30:28.977   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:30:28.977   -- DLL function import --
2020-04-14 10:30:28.977   Function name: idpFileDownloaded
2020-04-14 10:30:28.977   DLL name: files:idp.dll
2020-04-14 10:30:28.977   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-EVHM5.tmp\idp.dll
[... 505 lines cut ...]
2020-04-14 10:35:12.543   --------
2020-04-14 10:35:12.554   Installing Python environment:C:\Users\James\AppData\Local\Programs\Python\Python37\python.exe C:\Users\James\Desktop\esp-idf-7\tools\idf_tools.py install-python-env
2020-04-14 10:35:17.184   Done, exit code=1
2020-04-14 10:35:17.184   --------
2020-04-14 10:35:17.184   Running command: C:\Users\James\AppData\Local\Programs\Python\Python37\python.exe C:\Users\James\Desktop\esp-idf-7\tools\idf_tools.py install-python-env
                          usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                                            [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [-- [... cut ...]
                          subprocess.CalledProcessError: Command '['C:\\Users\\James\\AppData\\Local\\Programs\\Python\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\James\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 2.
                          
2020-04-14 10:35:17.185   --------
2020-04-14 10:35:17.185   Message box (OK):
2020-04-14 10:38:45.802   User chose OK.
2020-04-14 10:38:45.816   Message box (Yes/No):
2020-04-14 10:38:47.030   User chose Yes.
2020-04-14 10:38:47.140   CurStepChanged raised an exception.

804enl, 2020-04-14:
Hi Ivan
Did not realise I should have posted the Log file for  "esp-idf-tools-setup-2.3.exe"
hear.
Ishould have said that windows 10 does NOT have xcopy.
Hope this helps
This is the file, Thank you.
2020-04-14 10:08:02.235   Log opened. (Time zone: UTC+10:00)
2020-04-14 10:08:02.235   Setup version: Inno Setup version 6.0.2 (u)
2020-04-14 10:08:02.235   Original Setup EXE: D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.3.exe
2020-04-14 10:08:02.235   Setup command line: /SL5="$708F2,350580993,721408,D:\A A ELECTRONICS and  ATMEL\ESP-IDF\ESP IDF TOOLS\esp-idf-tools-setup-2.3.exe" 
2020-04-14 10:08:02.235   Windows version: 10.0.16299  (NT platform: Yes)
2020-04-14 10:08:02.236   64-bit Windows: Yes
2020-04-14 10:08:02.236   Processor architecture: x64
2020-04-14 10:08:02.247   User privileges: None
2020-04-14 10:08:02.257   Administrative install mode: No
2020-04-14 10:08:02.257   Install mode root key: HKEY_CURRENT_USER
2020-04-14 10:08:02.257   64-bit install mode: Yes
2020-04-14 10:08:02.297   Created temporary directory: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp
2020-04-14 10:08:02.652   -- DLL function import --
2020-04-14 10:08:02.652   Function name: idpAddFile
2020-04-14 10:08:02.652   DLL name: files:idp.dll
2020-04-14 10:08:02.652   Extracting temporary file: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.741   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.741   Importing the DLL function.
2020-04-14 10:08:02.763   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.763   -- DLL function import --
2020-04-14 10:08:02.763   Function name: idpAddFileComp
2020-04-14 10:08:02.763   DLL name: files:idp.dll
2020-04-14 10:08:02.764   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.764   Importing the DLL function.
2020-04-14 10:08:02.764   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.764   -- DLL function import --
2020-04-14 10:08:02.764   Function name: idpAddMirror
2020-04-14 10:08:02.764   DLL name: files:idp.dll
2020-04-14 10:08:02.764   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.764   Importing the DLL function.
2020-04-14 10:08:02.764   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.764   -- DLL function import --
2020-04-14 10:08:02.764   Function name: idpAddFtpDir
2020-04-14 10:08:02.764   DLL name: files:idp.dll
2020-04-14 10:08:02.765   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.765   Importing the DLL function.
2020-04-14 10:08:02.765   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.765   -- DLL function import --
2020-04-14 10:08:02.765   Function name: idpAddFtpDirComp
2020-04-14 10:08:02.765   DLL name: files:idp.dll
2020-04-14 10:08:02.765   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.765   Importing the DLL function.
2020-04-14 10:08:02.765   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.765   -- DLL function import --
2020-04-14 10:08:02.765   Function name: idpClearFiles
2020-04-14 10:08:02.765   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.766   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.766   -- DLL function import --
2020-04-14 10:08:02.766   Function name: idpFilesCount
2020-04-14 10:08:02.766   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.766   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.766   -- DLL function import --
2020-04-14 10:08:02.766   Function name: idpFtpDirsCount
2020-04-14 10:08:02.766   DLL name: files:idp.dll
2020-04-14 10:08:02.766   Dest DLL name: C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\idp.dll
2020-04-14 10:08:02.766   Importing the DLL function.
2020-04-14 10:08:02.767   Successfully imported the DLL function. Delay loaded? No
2020-04-14 10:08:02.767   -- DLL function import --
2020-04-14 10:08:02.767   Function name: idpFileDownloaded
2020-04-14 10:08:02.767   DLL name: files:idp.dll
[... 433 lines cut ...]
                          
                          Folders: 2381
                          Files: 11484
                          Size:       614298370
                          Compressed: 415318076
                          
2020-04-14 10:13:12.300   --------
2020-04-14 10:13:12.307   Copying C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\esp-idf-v4.0 to C:\Users\James\Desktop\esp-idf-7
2020-04-14 10:13:12.550   Done, exit code=1
2020-04-14 10:13:12.551   --------
2020-04-14 10:13:12.551   Running command: cmd.exe /c ""xcopy" /s /e /i /h /q "C:\Users\James\AppData\Local\Temp\is-T11P2.tmp\esp-idf-v4.0" "C:\Users\James\Desktop\esp-idf-7""
                          '"xcopy"' is not recognized as an internal or external command,
                          operable program or batch file.
                          
2020-04-14 10:13:12.551   --------
2020-04-14 10:13:12.551   Message box (OK):
                          Installation has failed with exit code 1
2020-04-14 10:14:29.241   User chose OK.
2020-04-14 10:14:29.256   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-04-14 10:14:30.888   User chose Yes.
2020-04-14 10:14:33.198   CurStepChanged raised an exception.
2020-04-14 10:14:33.456   Need to restart Windows? No
2020-04-14 10:14:33.467   Deinitializing Setup.

igrr, 2020-04-15:
@804enl it seems there is a problem with your system/user PATH environment variable:

> Setting PATH for this process: C:\Users\James\AppData\Local\Programs\Python\Python37;C:\Program Files\Git\cmd;C:\Program Files [...]

The path doesn't include standard Windows paths, like `%SystemRoot%\system32`, `%SystemRoot%`, etc. This is not normal, as such paths are included in PATH by default on Windows 10.

Please add `C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\` to PATH environment variable ([link](https://superuser.com/a/949577)) and try again.

amulya-incorrigible, 2020-10-30:
can someone please help me? I am recieving an installation error for idf-esp32 on windows 10. I tried different version releases too, the error is same
here is my log:
2020-10-30 10:45:36.736   --------
2020-10-30 10:45:36.745   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\asio\asio\objects\info\alternates
2020-10-30 10:45:36.747   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bootloader\subproject\components\micro-ecc\micro-ecc\objects\info\alternates
2020-10-30 10:45:36.748   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bt\controller\lib\objects\info\alternates
2020-10-30 10:45:36.750   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bt\host\nimble\nimble\objects\info\alternates
2020-10-30 10:45:36.752   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\coap\libcoap\modules\ext\tinydtls\objects\info\alternates
2020-10-30 10:45:36.753   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\coap\libcoap\objects\info\alternates
2020-10-30 10:45:36.755   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\esptool_py\esptool\objects\info\alternates
2020-10-30 10:45:36.756   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\esp_wifi\lib_esp32\objects\info\alternates
2020-10-30 10:45:36.758   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\expat\expat\objects\info\alternates
2020-10-30 10:45:36.759   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\json\cJSON\objects\info\alternates
2020-10-30 10:45:36.761   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\libsodium\libsodium\objects\info\alternates
2020-10-30 10:45:36.762   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\lwip\lwip\objects\info\alternates
2020-10-30 10:45:36.764   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\mbedtls\mbedtls\objects\info\alternates
2020-10-30 10:45:36.765   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\mqtt\esp-mqtt\objects\info\alternates
2020-10-30 10:45:36.768   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\modules\third-party\mruby\objects\info\alternates
2020-10-30 10:45:36.771   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\modules\third-party\neverbleed\objects\info\alternates
2020-10-30 10:45:36.772   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\objects\info\alternates
2020-10-30 10:45:36.773   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\protobuf-c\protobuf-c\objects\info\alternates
2020-10-30 10:45:36.775   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\spiffs\spiffs\objects\info\alternates
2020-10-30 10:45:36.776   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\unity\unity\objects\info\alternates
2020-10-30 10:45:36.778   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\examples\build_system\cmake\import_lib\main\lib\tinyxml2\objects\info\alternates
2020-10-30 10:45:36.779   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\objects\info\alternates
2020-10-30 10:45:36.780   idf_tools.py exists in IDF directory
2020-10-30 10:45:36.780   idf_tools.py command: C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2
2020-10-30 10:45:36.780   Installing tools:C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2 install
2020-10-30 10:45:37.102   Done, exit code=1
2020-10-30 10:45:37.102   --------
2020-10-30 10:45:37.102   Running command: C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2 install
                          Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@esp-2020r3-8.4.0
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
[... cut ...]
2020-10-30 10:45:37.102   --------
2020-10-30 10:45:37.102   Message box (OK):
                          Installation has failed with exit code 1
2020-10-30 10:46:25.858   User chose OK.
2020-10-30 10:46:25.865   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-10-30 10:46:27.373   User chose Yes.
2020-10-30 10:46:27.404   CurStepChanged raised an exception.

igrr, 2020-10-30:
@amulya-incorrigible it looks like your copy of Python doesn't have HTTPS support installed (`WARNING: Download failure <urlopen error unknown url type: https>`). I would recommend installing Python 3.7 or 3.8 from Python.org, and choosing it in the IDF Tools installer.

imranrajjad, 2020-12-11:
I am getting following error on Windows 10

Processing d:\esp-idf\esp-idf\tools\kconfig_new\esp-windows-curses
Exception:
Traceback (most recent call last):
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2610, in _dep_map
    return self.__dep_map
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2685, in __getattr__
    raise AttributeError(attr)
AttributeError: _Distribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\basecommand.py", line 209, in main
    status = self.run(options, args)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\commands\install.py", line 310, in run
    wb.build(autobuilding=True)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\wheel.py", line 748, in build
    self.requirement_set.prepare_files(self.finder)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\req\req_set.py", line 360, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\req\req_set.py", line 647, in _prepare_file
    set(req_to_install.extras) - set(dist.extras)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2810, in extras
    return [dep for dep in self._dep_map if dep]
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2624, in _dep_map
    dm.setdefault(extra,[]).extend(parse_requirements(reqs))
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2980, in parse_requirements
    "version spec")
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2945, in scan_list
    raise RequirementParseError(msg, line, "at", line[p:])
pip._vendor.pkg_resources.RequirementParseError: Expected version spec in windows-curses; sys_platform == "win32" at ; sys_platform == "win32"
You are using pip version 8.1.1, however version 20.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
Command failed: "D:\esp-idf\python_env\idf4.1_py3.5_env\Scripts\python.exe" -m pip install -r "D:\esp-idf\esp-idf\requirements.txt"
Exception:
Traceback (most recent call last):
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2610, in _dep_map
    return self.__dep_map
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2685, in __getattr__
    raise AttributeError(attr)
AttributeError: _Distribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\basecommand.py", line 209, in main
    status = self.run(options, args)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\commands\install.py", line 310, in run
    wb.build(autobuilding=True)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\wheel.py", line 748, in build
    self.requirement_set.prepare_files(self.finder)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\req\req_set.py", line 360, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\req\req_set.py", line 647, in _prepare_file
    set(req_to_install.extras) - set(dist.extras)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2810, in extras
    return [dep for dep in self._dep_map if dep]
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2624, in _dep_map
    dm.setdefault(extra,[]).extend(parse_requirements(reqs))
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2980, in parse_requirements
    "version spec")
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2945, in scan_list
    raise RequirementParseError(msg, line, "at", line[p:])
pip._vendor.pkg_resources.RequirementParseError: Expected version spec in windows-curses; sys_platform == "win32" at ; sys_platform == "win32"
You are using pip version 8.1.1, however version 20.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
Command failed: "D:\esp-idf\python_env\idf4.1_py3.5_env\Scripts\python.exe" -m pip install -r "D:\esp-idf\esp-idf\requirements.txt"
Exception:
[... 6 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\basecommand.py", line 209, in main
    status = self.run(options, args)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\commands\install.py", line 310, in run
    wb.build(autobuilding=True)
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\wheel.py", line 748, in build
    self.requirement_set.prepare_files(self.finder)
[... cut ...]
    "version spec")
  File "D:\esp-idf\python_env\idf4.1_py3.5_env\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 2945, in scan_list
    raise RequirementParseError(msg, line, "at", line[p:])
pip._vendor.pkg_resources.RequirementParseError: Expected version spec in windows-curses; sys_platform == "win32" at ; sys_platform == "win32"
You are using pip version 8.1.1, however version 20.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

igrr, 2020-12-11:
@imranrajjad It seems that you are using Python 3.5 together with a very outdated pip version (8.1.1). Please try installing an up-to-date Python 3.7 or 3.8 from python.org. One way is to choose "Install Python 3.7" option in the IDF Tools Installer. It will download and install a more recent version of Python for you.