espressif/esp-idf #4343: 'sprintf' writing a terminating nul past the end of the destination

Kirankumar-xenlabs, 2019-11-15:
Hello everyone,
I updated my ESP-IDF to v4.0beta2, I try to build my project, I am getting following error, but my project builds on v3.2.2.

error: 'sprintf' writing a terminating nul past the end of the destination [-Werror=format-overflow=]
   sprintf(post_data, "scope=https://www.googleapis.com/auth/devstorage.read_write&client_id=%s", CONFIG_GCP_CLIENT_ID);
                                                                                               ^
../components/google_api/gcp_auth_get_device_code.c:16:3: note: 'sprintf' output 144 bytes into a destination of size 143
   sprintf(post_data, "scope=https://www.googleapis.com/auth/devstorage.read_write&client_id=%s", CONFIG_GCP_CLIENT_ID);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

I need to fix this how can I do this, Please help me.

Regards,
Kiran

negativekelvin, 2019-11-15:
Reduce size of CONFIG_GCP_CLIENT_ID or increase size of post_data.  This question belongs on esp32.com forum.  Only use GitHub issues for bugs in esp-idf code.

Kirankumar-xenlabs, 2019-11-16:
Thank you so much @negativekelvin  I made changes according you said, its working fine now

closed 2019-11-16 by Kirankumar-xenlabs