espressif/esp-idf #4344: esp_timer_start_once/esp_timer_start_periodic does not work if timeout_us/period_us arguments set to 2hr+

AxelLin, 2019-11-15:
I'm testing on IDF-4.0-beta2.
I'm running the esp_timer example with below changes:
Change the timeout_us/period_us arguments to 2hr.
I would expect the timeout happen after 2hr, but the timeout happens shortly:

diff --git a/examples/system/esp_timer/main/esp_timer_example_main.c b/examples/system/esp_timer/main/esp_timer_example_main.c
index ab6a697c2bfd..ba772a67ea08 100644
--- a/examples/system/esp_timer/main/esp_timer_example_main.c
+++ b/examples/system/esp_timer/main/esp_timer_example_main.c
@@ -48,8 +48,10 @@ void app_main()
     ESP_ERROR_CHECK(esp_timer_create(&oneshot_timer_args, &oneshot_timer));

     /* Start the timers */
-    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 500000));
-    ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer, 5000000));
+    //ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 500000));
+    //ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer, 5000000));
+    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 7200000000));
+    ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer, 7200000000));//2hr
     ESP_LOGI(TAG, "Started timers, time since boot: %lld us", esp_timer_get_time());

     /* Print debugging information about timers to console every 2 seconds */

Below is what I got:

I (315) example: Started timers, time since boot: 22917 us
periodic        7200000000    7200022906          1          0             0
one-shot                 0    7200022912          1          0             0
periodic        7200000000    7200022906          1          0             0
one-shot                 0    7200022912          1          0             0
periodic        7200000000    7200022906          1          0             0
one-shot                 0    7200022912          1          0             0
periodic        7200000000    7200022906          1          0             0
one-shot                 0    7200022912          1          0             0
periodic        7200000000    7200022906          1          0             0
one-shot                 0    7200022912          1          0             0
I (10335) example: Entering light sleep for 0.5s, time since boot: 10037654 us
I (10335) example: Woke up from light sleep, time since boot: 10538173 us

igrr, 2019-11-15:
Have you tried using `7200000000ULL` as an argument?

AxelLin, 2019-11-15:
The same result with 7200000000ULL.

negativekelvin, 2019-11-15:
Where does the timeout happen?  I don't see it in the logs.

AxelLin, 2019-11-15:
I recheck the example code and found it restart before reaching timeout.
So now I add while(1) usleep(2000000); to avoid restart before timeout.
And I re-run the test again, it works if using 7200000000ULL.

closed 2019-11-15 by AxelLin