espressif/esp-idf #4346: BLE - before the bonding with the aimed device, the notify packets and characteristics can be reached by any device for 30 seconds

Antdigapony, 2019-11-15:
Hello! 
I have the next problem:
I connect ESP32-WROVER-KIT with the Android app. During the test at the Android's side I use the nRF connect - program. At the ESP32's side I use the *nkolban*'s wrapper over the BLE-stack of ESP-32. 
I created the custom BLE service and the custom BLE characteristic with properties to read, write, notify, indicate and added the descriptor to it. The permissions of both - characteristic and descriptor are READ_ENCRYPTED | WRITE_ENCRYPTED. The BLE Device and Server on ESP32 are set to the bonding and passkey mode.
Also in the loop-function I change the characteristic manually and send the notify via the call of the notify() function of *nkolban*. And here is a problem:

Expected behavior:
 - if not bonded device connects to the ESP32 - the characteristic value is not accessible for the device - both read request, both without it.

Actual behavior:
 - not bonded device cannot read the characteristic value with the read request, but the notifying packets from the ESP32's loop function bring the characteristic values to the device during 30 sec, which (30 seconds) are provided by the passkey awaiting process. 

So, is there any way to encrypt and protect the notifying packets? Because, this seems to be a gap for the hacking of the BLE connection.

WCCWCC, 2020-06-29:
Hi， @Antdigapony 

* Can you provide the code and recurrence steps?
* I will also build a case locally to see if the problem actually exists?
* Note: I have not used the function nkolban

[image]

chegewara, 2020-06-29:
> * not bonded device cannot read the characteristic value with the read request, but the notifying packets from the ESP32's loop function bring the characteristic values to the device during 30 sec, which (30 seconds) are provided by the passkey awaiting process.

This is most likely bug in Kolban's library. Its because library is not keeping CCC settings for each connected client and is sending notifications to every connected device, so its enough that only 1 device is bonded and enable notifications. Then any connected device can get notifications until esp32 get restarted or notifications will be disabled by bonded client.

Alvin1Zhang, 2020-08-19:
Feel free to reopen.

closed 2020-08-19 by Alvin1Zhang