espressif/esp-idf #4347: Unable to connect to CISCO wpa2 enterprise network

persan666, 2019-11-15:
## Environment

- Development Kit:      [CUSTOM]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: 
    // v3.3-127-g0a0f2caa1
- Build System:         [Make]
- Compiler version:
    // 1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:         [Battery]

## Problem Description

A customer of mine has trouble connecting to their factory network that uses CISCO WPA2 enterprise security.
He uses PEAP-MSCHAPV2 as a security method without any certificates being set (client or CA) with username and password assigned.
Logging is set to DEBUG but does not generate much info.

The customer receives the following error message from the CISCO network:
'** 12321 PEAP failed SSL / TLS handshake because client rejected ISE local certificate **'

I interpret it as ESP32 is rejecting the server certificate for some reason.

I have managed to connect to my own router (ASUS RT-AC86U) with freeRADIUS with the same ESP32 hardware/software version and settings.

One thing I noticed in the log output is the line:
**W (197204) wifi: add mismatch**
I have not seen this before. I have search for it but it seems to be embedded in the sealed espressif wifi library somewhere. 

### Expected Behavior

### Actual Behavior

### Steps to repropduce
I am unable to reproduce this due to lack of customer hardware

### Code to reproduce this issue

```cpp

```
## Debug Logs

I (193481) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (195159) wifi: new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (195164) wifi: state: run -> auth (b0)
I (196165) wifi: state: run -> init (200)
I (196170) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (196171) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (196176) wifi: state: init -> auth (b0)
I (197179) wifi: state: auth -> init (200)
I (197180) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (197185) wifi: new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (197189) wifi: state: init -> auth (b0)
I (197194) wifi: state: auth -> assoc (0)
I (197199) wifi: state: assoc -> run (10)
W (197204) wifi: add mismatch
I (197205) wpa: wpa2_task prio:2, stack:6656

I (197244) wpa: >>>>>wpa2 FAILED
```

negativekelvin, 2019-11-15:
Possibly https://github.com/espressif/esp-idf/issues/248#issuecomment-360338336

Can you get a log with the wpa debug output like https://github.com/espressif/esp-idf/issues/2381

persan666, 2019-11-18:
Thanks for replying @negativekelvin.
Unfortunately the log presented is with debug message activated.
When I connect to my own system I get the first **debug** messages after the following message 
**I (197205) wpa: wpa2_task prio:2, stack:6656**

As the comment #https://github.com/espressif/esp-idf/issues/248#issuecomment-360338336
says, it will not work if I don't supply a CA certificate. Is that correct or has there been any changes for this since jan 2018 when the comment was written.

negativekelvin, 2019-11-18:
Not sure, but it is unusual that there is no debug prints and it fails so quickly (40ms)

sagb2015, 2019-11-18:
Hi @persan666,  can you cherrypick and try with 
https://github.com/espressif/esp-idf/commit/7e3676f30732f0824205e578bbf56e0c4ae9e7c8
You can try disabling EXAMPLE_VALIDATE_SERVER_CERT option for the example (do not provide and ca cert). Also, please provide the sample code with which you see the issue.  We have also seen that depending on default authentication set at the Radius Server side, the behaviour seen changes. This is because on response to NAK from ESP32 ( do not support default method for authentication), the authentication does not proceed with next available method since it is implementation specific. Anyways, we have started looking at the old enterprise issues including  https://github.com/espressif/esp-idf/issues/248 and will try to address them. Any help in collecting information with interop issues. will be appreciated.

persan666, 2019-11-18:
Hi @sagb2015, If I understand the code from the example correctly. If the EXAMPLE_VALIDATE_SERVER_CERT compiler directive is defined, the CA cert and key is set using the esp_wifi_sta_wpa2_ent_set_ca_cert. My project does not use that set function either, so I should get the same behavior. Or do the EXAMPLE_VALIDATE_SERVER_CERT compiler directive affect something else some where?

sagb2015, 2019-11-18:
Hi @persan666 - Yes, that's correct. If esp_wifi_sta_wpa2_ent_set_ca_cert is not called, then the failure should not be there. Is it possible for you to provide your code or try with the example code to reproduce the issue? Also enable debug log for both IDF (make menuconfig ->log output->) and (make menuconfig -> WiFi -> Enable WiFi debug log).

persan666, 2019-11-18:
Hi @sagb2015, unfortunately I am unable to try with the Example code due to that the device is too far away on another continent. And it can only handle my software over OTA updates.

Debug is already defined in menuconfig and before the connection attempt is made the debug level is set with 
```c++
esp_log_level_set("*", ESP_LOG_DEBUG);`
```
In principle this is the code used for the connection:

```c++
wifi_config_t wificonf;

memset(&wificonf, 0, sizeof(wifi_config_t));

strncpy((char *)wificonf.sta.ssid, (char *)settings->ssid, strlen((char *)settings->ssid));

wificonf.sta.ssid[strlen((char *)wificonf.sta.ssid)] = '\0';

strncpy((char*)wificonf.sta.password, (char*)settings->password, strlen((char *)settings->password));

wificonf.sta.password[strlen((char *)wificonf.sta.password)] = '\0';

memcpy(wificonf.sta.bssid, settings->bssid, sizeof(settings->bssid)/sizeof(settings->bssid[0]));

if(settings->bssid_used == 1){
    if(settings->channel != 0){
        wificonf.sta.channel = settings->channel;
    }
    wificonf.sta.bssid_set = true;
}else{
    wificonf.sta.bssid_set = false;
}

wificonf.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;

if(esp_wifi_set_config(ESP_IF_WIFI_STA, &wificonf) != ESP_OK){
    ESP_LOGE(TAG, "Failed to apply WIFI settings.");
}

esp_err_t esp_ret;

esp_wifi_sta_wpa2_ent_disable();

if(eap_settings->security_mode == SPM_WIFI_AUTH_WPA2_ENTERPRISE){
    if(strlen((char *)eap_settings->eap_id) > 0){
        if((esp_ret = esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)eap_settings->eap_id, strlen((char *)eap_settings->eap_id))) != ESP_OK){
            return esp_ret;
        }
    }
    if (eap_settings->eap_method == SPM_WIFI_EAP_PEAP_MSCHAP_V2 || eap_settings->eap_method == SPM_WIFI_EAP_TTLS_MSCHAP_V2) {
        if((esp_ret = esp_wifi_sta_wpa2_ent_set_username((uint8_t *)eap_settings->eap_username, strlen((char *)eap_settings->eap_username))) != ESP_OK){                      
            return esp_ret;
        }
        if((esp_ret = esp_wifi_sta_wpa2_ent_set_password((uint8_t *)eap_settings->eap_password, strlen((char *)eap_settings->eap_password))) != ESP_OK){
            return esp_ret;
        }
    }
    esp_wpa2_config_t wpa2_config = WPA2_CONFIG_INIT_DEFAULT();
    if((esp_ret = esp_wifi_sta_wpa2_ent_enable(&wpa2_config)) != ESP_OK){
        return esp_ret;
    }
}

if(p_config->com.ip_settings.use_dhcp == false){
    set_static_ip(&p_config->com);
}else{
    tcpip_adapter_dhcpc_start(TCPIP_ADAPTER_IF_STA);
}

esp_wifi_connect();
```

sagb2015, 2019-11-19:
This may actually be problem with ISE certificates and ESP32 not being able to skip the verification for same ([ref](https://community.cisco.com/t5/identity-services-engine-ise/12321-peap-failed-ssl-tls-handshake-because-the-client-rejected/td-p/3567818)). We will try to create this problem locally first.

sagb2015, 2019-11-19:
Also, is it possible to get full EAP logs on radius server side? Without  any logs, it will be tough to debug. BTW, wifi has different log settings. You will have to enable that explicitly to get device side log.

persan666, 2019-11-19:
Sorry, I didn't look carefully, this is how I enables the logging:
Is this correct, or can I enable something more?

```c++
esp_log_level_set("*", ESP_LOG_DEBUG);
esp_log_level_set("wifi", ESP_LOG_DEBUG);
esp_log_level_set("dhcpc", ESP_LOG_DEBUG);
esp_log_level_set("mbedtls", ESP_LOG_DEBUG);
```

persan666, 2019-11-19:
I read your comment again and saw that I over looked the following text:
**and (make menuconfig -> WiFi -> Enable WiFi debug log).**
I will enable that and ask the customer to try again and see if I can get some more logging info.

persan666, 2019-11-22:
Here are the more detaild log from the customer.  Some info has been masked out and replaced with (?????).

I (87727) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (87742) wifi: Start wifi connect
I (87745) wifi: connect status 0 -> 0
I (87748) wifi: connect chan=0
I (87751) wifi: first chan=1
I (87755) wifi: connect status 0 -> 1
I (87757) wifi: filter: set rx policy=3
I (87760) wifi: clear scan ap list
I (87763) wifi: start scan: type=0x50f, priority=2, cb=0x401a76ec, arg=0x0, ss_state=0x1, time=83055590, index=0
I (87773) wifi: perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
I (87782) wifi: rsn valid: gcipher=3 ucipher=3 akm=4

I (87784) wifi: profile match: ss_state=0x7
I (87791) wifi: rsn valid: gcipher=3 ucipher=3 akm=4

I (87793) wifi: profile match: ss_state=0x7
I (87900) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (87904) wifi: perform scan: ss_state=0xd, chan<2,0>, dur<0,120>
I (88026) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88031) wifi: perform scan: ss_state=0xd, chan<3,0>, dur<0,120>
I (88153) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88157) wifi: perform scan: ss_state=0xd, chan<4,0>, dur<0,120>
I (88278) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88283) wifi: perform scan: ss_state=0xd, chan<5,0>, dur<0,120>
I (88404) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88408) wifi: perform scan: ss_state=0xd, chan<6,0>, dur<0,120>
I (88414) wifi: rsn valid: gcipher=3 ucipher=3 akm=4

I (88420) wifi: profile match: ss_state=0x7
I (88530) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88530) wifi: perform scan: ss_state=0xd, chan<7,0>, dur<0,120>
I (88655) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88656) wifi: perform scan: ss_state=0xd, chan<8,0>, dur<0,120>
I (88782) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88782) wifi: perform scan: ss_state=0xd, chan<9,0>, dur<0,120>
I (88907) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (88908) wifi: perform scan: ss_state=0xd, chan<10,0>, dur<0,120>
I (89034) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (89034) wifi: perform scan: ss_state=0xd, chan<11,0>, dur<0,120>
I (89041) wifi: rsn valid: gcipher=3 ucipher=3 akm=4

I (89043) wifi: profile match: ss_state=0x7
I (89049) wifi: rsn valid: gcipher=3 ucipher=3 akm=4

I (89049) wifi: remove 5c:a4:8a:d7:8e:e3 from rc list
I (89053) wifi: profile match: ss_state=0x7
I (89159) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (89160) wifi: perform scan: ss_state=0xd, chan<12,0>, dur<360,360>
I (89525) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (89526) wifi: perform scan: ss_state=0xd, chan<13,0>, dur<360,360>
I (89889) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (89890) wifi: filter: set rx policy=4
I (89894) wifi: first chan=1
I (89895) wifi: handoff_cb: status=0
I (89895) wifi: ap found, mac=?????
I (89897) wifi: new_bss=?????, cur_bss=0x0, new_chan=<1,0>, cur_chan=1
I (89904) wifi: filter: set rx policy=5
I (89908) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (89914) wifi: connect_op: status=0, auth=4, cipher=3 
I (89919) wifi: auth mode is not none
I (90619) wifi: connect_bss: auth=1, reconnect=0
I (90619) wifi: state: init -> auth (b0)
I (90620) wifi: start 1s AUTH timer
I (90625) wifi: clear scan ap list
I (90627) wifi: recv auth: seq=2, status=0
I (90628) wifi: state: auth -> assoc (0)
I (90632) wifi: restart connect 1s timer for assoc
[... 91 lines cut ...]
D (91522) wpa: X509: Extension: extnID=????? critical=0 [0m
D (91522) wpa: TLSv1: Certificate 2 (len 1377) [0m
D (91527) wpa: X509: Version X.509v3 [0m
D (91537) wpa: X509: serialNumber 2862990861 [0m
D (91537) wpa: X509: issuer CN=????? [0m
D (91558) wpa: X509: Validity: notBefore: 0 notAfter: 0 [0m
D (91559) wpa: X509: subject CN=????? [0m
D (91562) wpa: X509: Extension: extnID=????? critical=0 [0m
D (91579) wpa: X509: KeyUsage 0x61 [0m
D (91579) wpa: ASN.1: Extended tag data: 0x04 [0m
D (91603) wpa: X509: Extension: extnID=????? critical=255 [0m
D (91603) wpa: X509: BasicConstraints - cA=255 [0m
D (91621) wpa: X509: Extension: extnID=????? critical=0 [0m
D (91621) wpa: X509: Extension: extnID=????? critical=0 [0m
D (91641) wpa: X509: Extension: extnID=????? critical=0 [0m
D (91642) wpa: X509: Validate certificate chain [0m
D (91657) wpa: X509: 0: O=BT, CN=????? [0m
D (91829) wpa: X509: Certificate Digest matches with calculated tbsCertificate hash [0m
D (91849) wpa: X509: 1: DC=?????, DC=?????, DC=?????, CN=????? [0m
D (92313) wpa: X509: Unrecognized digestAlgorithm [0m
D (92329) wpa: X509: Invalid certificate signature within chain [0m
D (92329) wpa: TLSv1: Server certificate chain validation failed (reason=1) [0m
D (92349) wpa: TLSv1: Send Alert(2:42) [0m
D (92349) wpa: SSL: 7 bytes left to be sent out (of total 7 bytes) [0m
I (92385) wpa: >>>>>wpa2 FAILED**

negativekelvin, 2019-11-22:
The second certificate causes 
`D (92313) wpa: X509: Unrecognized digestAlgorithm [0m`

What is the signature used on the cert?

sagb2015, 2019-11-22:
There are two things which are wrong here - 
1) The certificate chain validity has failed because of digest not getting recognised. 
2) x509_certificate_chain_validate is called even if CA certificate is not supplied. 

For the cases when CA certificate is not provided but digest is proper, the function x509_certificate_chain_validate returns 0 (SUCCESS) and hence we were not able to reproduce the issue. We will fix this.

negativekelvin, 2019-11-22:
@sagb2015 don't you think there is some value in validating the certificate chain even if there is no local trust root?

negativekelvin, 2019-11-24:
Also @sagb2015 what about using mbedtls x509 functions?

sagb2015, 2019-11-25:
@negativekelvin - I think as far as parsing is okay, there is no need to validate the certificate. However, in this case even the parsed data seems to be random bytes.

@persan666 Is it possible to get the public certificate from your customer?

negativekelvin, 2019-11-25:
@sagb2015 but using mbedtls will support more signature algorithms like ecdsa and will also reduce code size not having two x509 parsers.  I don't know if it is random bytes or redacted.

sagb2015, 2019-11-26:
@negativekelvin - We will add the support for mbedtls x509 parsing, but keeping in-line with standard supplicant that will be a compile-time option. So we will still have to fix this. As far as validation is concerned typically it is done by radius server at the start and I think attacks are always possible when you disable authentication. But I understand your concern and we will keep the code flow as is. Waiting for @persan666 to provide the public certificate from customer...

persan666, 2019-11-26:
Hi, sorry for the delay. We are still waiting for response from customer to get the public certificate.
Do you know of any temporary workaround to disable server certificate check for the case with no CA certificate.

sagb2015, 2019-11-26:
Workaround would be disable the call to x509_certificate_chain_validate when "server authentication" is disabled. It however depends on whether rest of the parsing is okay.

persan666, 2019-11-27:
Hi,
I have now received the CA Certificate from the customer. The customer is using sha512rsa and for what I understand, ESP-IDF does not support that? I can send the public certificate in a PM to the one who can make use of it.

sagb2015, 2019-11-27:
@persan666 - I expected different error message for sha512rsa ("X509: Unsupported certificate signature").  You can send the certificate to me. We have also put using mbedtls for enterprise on our TBD list. I haven't checked but this should be supported by mbedtls. We can give you a workaround until then.

persan666, 2019-11-27:
Thanks for the fast reply!
How can I send the certificate to you? Let me know when you have a workaround ready.

sagb2015, 2019-11-27:
You can check the commit log for email.

persan666, 2019-11-28:
The certificate has been sent to you @sagb2015 now.

sagb2015, 2019-11-28:
@persan666 Can you try the following workaround - Replace [this](https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/src/tls/tlsv1_client_read.c#L367 )  line with  
_**"if (conn->cred && g_wpa_ca_cert &&"**_

persan666, 2019-11-28:
Great!
I will make the change and ask the customer to try it.

sagb2015, 2019-12-31:
@persan666: Any update on this?

Alvin1Zhang, 2020-01-08:
@persan666 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

persan666, 2020-01-08:
Hi, sorry for late respond.
The patch seems to work! I have not yet got any detailed information from the customer.
I will update here when I get more information.

Alvin1Zhang, 2020-01-08:
@persan666 Thanks for the updates.

persan666, 2020-01-22:
Hi again,
I have been in contact with the customer and they now experience this problem.
here is the log from the connection attempt. (Some information has been masked out) 

I (192028) wifi: clear blacklist
I (192031) wpa: WPA2 ENTERPRISE VERSION: [v2.0] disable
D (192036) wpa: TLSv1: Selected cipher suite: 0x0000
D (192041) wpa: TLSv1: Record Layer - New write cipher suite 0x0000
D (192047) wpa: TLSv1: Record Layer - New read cipher suite 0x0000 
I (192054) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable
D (192059) tcpip_adapter: dhcp client re init
I (192069) wifi: Start wifi connect
I (192072) wifi: connect status 2 -> 0
I (192076) wifi: connect chan=0
I (192080) wifi: first chan=1
I (192081) wifi: connect status 0 -> 1
I (192085) wifi: filter: set rx policy=3
I (192088) wifi: clear scan ap list
I (192091) wifi: start scan: type=0x50f, priority=2, cb=0x401a848c, arg=0x0, ss_state=0x1, time=186656842, index=0
I (192102) wifi: perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
I (192110) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (192115) wifi: profile match: ss_state=0x7
I (192117) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (192121) wifi: set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
I (192129) wifi: sig_b=1, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
I (192135) wifi: update trc
I (192138) wifi: profile match: ss_state=0x7
I (192142) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (192148) wifi: profile match: ss_state=0x7
I (192228) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (192259) wifi: perform scan: ss_state=0xd, chan<2,0>, dur<0,120>
I (192384) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (192416) wifi: perform scan: ss_state=0xd, chan<3,0>, dur<0,120>
I (192542) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (192573) wifi: perform scan: ss_state=0xd, chan<4,0>, dur<0,120>
I (192698) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (192729) wifi: perform scan: ss_state=0xd, chan<5,0>, dur<0,120>
I (192855) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (192886) wifi: perform scan: ss_state=0xd, chan<6,0>, dur<0,120>
I (192894) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (192896) wifi: profile match: ss_state=0x7
I (192901) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (192902) wifi: remove 5c:a4:8a:d7:90:53 from rc list
I (192904) wifi: profile match: ss_state=0x7
I (193012) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193043) wifi: perform scan: ss_state=0xd, chan<7,0>, dur<0,120>
I (193170) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193201) wifi: perform scan: ss_state=0xd, chan<8,0>, dur<0,120>
I (193326) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193357) wifi: perform scan: ss_state=0xd, chan<9,0>, dur<0,120>
I (193484) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193515) wifi: perform scan: ss_state=0xd, chan<10,0>, dur<0,120>
I (193640) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193671) wifi: perform scan: ss_state=0xd, chan<11,0>, dur<0,120>
I (193680) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (193680) wifi: rsn valid: gcipher=3 ucipher=3 akm=4
I (193684) wifi: remove 5c:a4:8a:d7:8e:e3 from rc list
I (193687) wifi: profile match: ss_state=0x7
I (193798) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (193833) wifi: filter: set rx policy=4
I (193834) wifi: first chan=1
I (193834) wifi: handoff_cb: status=0
I (193838) wifi: ap found, mac=00:1e:f6:81:42:e3
I (193838) wifi: new_bss=0x3ffca1b0, cur_bss=0x3ffca480, new_chan=<1,0>, cur_chan=1
I (193844) wifi: filter: set rx policy=5
I (193847) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (193854) wifi: connect_op: status=0, auth=4, cipher=3 
I (193859) wifi: auth mode is not none
I (193862) wifi: connect_bss: auth=1, reconnect=0
I (193866) wifi: state: run -> auth (b0)
[... 315 lines cut ...]
D (226544) wpa: EAP-PEAP: using label 'client EAP encryption' in key derivation
D (226566) wpa: SSL: Building ACK (type=25 id=159 ver=1) 
D (226602) wpa: SSL: Received packet(len=43) - Flags 0x01
D (226602) wpa: EAP-PEAP: received 37 bytes encrypted data for Phase 2
D (226622) wpa: TLSv1: Received content type 23 version 3.1 length 32
D (226624) wpa: EAP-PEAP: received Phase 2: code=1 identifier=160 length=5
D (226638) wpa: EAP-PEAP: Phase 2 Request: type=1
D (226639) wpa: SSL: 53 bytes left to be sent out (of total 53 bytes)
D (226678) wpa: SSL: Received packet(len=75) - Flags 0x01
D (226678) wpa: EAP-PEAP: received 69 bytes encrypted data for Phase 2
D (226691) wpa: TLSv1: Received content type 23 version 3.1 length 64
D (226714) wpa: EAP-PEAP: received Phase 2: code=1 identifier=161 length=40
D (226716) wpa: EAP-PEAP: Phase 2 Request: type=26
D (226732) wpa: EAP-PEAP: Selected Phase 2 EAP vendor 0 method 26
D (226732) wpa: EAP-MSCHAPV2: RX identifier 161 mschapv2_id 161
D (226751) wpa: EAP-MSCHAPV2: Generate Challenge Response
D (226754) wpa: SSL: 101 bytes left to be sent out (of total 101 bytes)
D (226797) wpa: SSL: Received packet(len=91) - Flags 0x01
D (226797) wpa: EAP-PEAP: received 85 bytes encrypted data for Phase 2
D (226813) wpa: TLSv1: Received content type 23 version 3.1 length 80
D (226816) wpa: EAP-PEAP: received Phase 2: code=1 identifier=162 length=57
D (226830) wpa: EAP-PEAP: Phase 2 Request: type=26
D (226851) wpa: EAP-MSCHAPV2: RX identifier 162 mschapv2_id 161
E (226851) wpa: phase 2 response failure
E (226855) wpa: Response build fail, return.

AxelLin, 2020-06-18:
> @persan666 Can you try the following workaround - Replace [this](https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/src/tls/tlsv1_client_read.c#L367) line with
> _**"if (conn->cred && g_wpa_ca_cert &&"**_

@sagb2015 
How is the status?
Is this issue fixed properly or is this workaround still needed?

persan666, 2020-06-18:
Hi!
Sorry for not responding!
The latest error was caused by entered the wrong password too many times.
I do not know if the workaround is still needed, but the latest versions of the esp-idf look the same for what I can see.

sagb2015, 2020-06-22:
@AxelLin We are in process of porting supplicant TLS layer over mbedtls. Once the porting is done, more cert types should be supported. We will keep the issue open till then.

AxelLin, 2020-06-22:
@sagb2015 
Erh, I don't quite understand your reply.
It seems you mix bug fix with porting TLS things.

You provided the workaround on 28 Nov 2019 and persan666 confirmed it works.
Isn't it important to fix things first rather than rework for porting supplicant TLS layer?
If your porting work can be done soon (e.g. within a week), it's probably fine to ask people wait.
Otherwise, I do think you should provide fix bug first.
(especially considering you might need to backport the fix to older idf versinos).

sagb2015, 2020-06-22:
Hi @AxelLin, 
The current code does not support the digest algorithm required by the certificate, neither does it claim to do so. The workaround provided was to skip certificate chain validation completely if local trust root is not programmed. Please refer to the discussion above with @negativekelvin. The code is supposed calculate the digest even if local root is not provided and we prefer to keep that behaviour. The only way to fix this properly is to add support for new certificate types. Hope this helps!

sagb2015, 2020-08-04:
@persan666 We have added mbedls-based porting of supplicant TLS. Can you help to check this issue with master? No workaround should be needed in this case.

sagb2015, 2020-08-18:
@persan666  We are closing this issue. Feel free to reopen in case you still face it.

closed 2020-08-18 by sagb2015