espressif/esp-idf #4349: mqtt_client: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

rrroonn, 2019-11-15:
- Development Kit: TTGO 
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz

- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c

```
$ git describe --tags
fatal: not a git repository (or any of the parent directories): .git
```

- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):

xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     Windows
- Power Supply:        USB

=========================================
mqtt client consistently panics  when publishing to an unavailable server.

Excerpt from mqtt_client.c (with some added debug code)
```
int esp_mqtt_client_publish(esp_mqtt_client_handle_t client, const char *topic, const char *data, int len, int qos, int retain)
{
    uint16_t pending_msg_id = 0;

    /* Acceptable publish messages:
        data == NULL, len == 0: publish null message
        data valid,   len == 0: publish all data, payload len is determined from string length
        data valid,   len >  0: publish data with defined length
     */
    if (len <= 0 && data != NULL) {
        len = strlen(data);
    }

    ESP_LOGE("RWB_MQTT", "client=%08Xh, topic=%08Xh, data=%08Xh",(unsigned)client,(unsigned)topic,(unsigned)data);

    MQTT_API_LOCK_FROM_OTHER_TASK(client);
    mqtt_message_t *publish_msg = mqtt_msg_publish(&client->mqtt_state.mqtt_connection,
                                  topic, data, len,
                                  qos, retain,
                                  &pending_msg_id);

    if (publish_msg == NULL) {
        ESP_LOGE(TAG, "Publish message cannot be created");
        MQTT_API_UNLOCK_FROM_OTHER_TASK(client);
        return 0;
    }
// debug messages to highlight problem
    ESP_LOGE("RWB_MQTT", "publish_message=%08Xh",(unsigned)publish_msg);
    ESP_LOGE("RWB_MQTT", "publish_message->data=%08Xh",(unsigned)publish_msg->data);

    /* We have to set as pending all the qos>0 messages */
    if (qos > 0) {
        client->mqtt_state.outbound_message = publish_msg;
// ======= PANIC ON NEXT LINE !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
        client->mqtt_state.pending_msg_type = mqtt_get_type(client->mqtt_state.outbound_message->data);
        client->mqtt_state.pending_msg_id = pending_msg_id;
        client->mqtt_state.pending_publish_qos = qos;
        client->mqtt_state.pending_msg_count ++;
        // by default store as QUEUED (not transmitted yet) only for messages which would fit outbound buffer
        if (client->mqtt_state.mqtt_connection.message.fragmented_msg_total_length == 0) {
            mqtt_enqueue(client);
        }
```

[... 172 lines cut ...]

ELF file SHA256: 73e7f9e5c9503a45d0c84fed268ce4ea101f410b727dd51162e73d7fc782f734

Backtrace: 0x400d969f:0x3ffbba40 0x400d6b12:0x3ffbba90 0x400d4adf:0x3ffbbab0 0x400d4f26:0x3ffbbad0 0x400d4837:0x3ffbbaf0 0x400d4864:0x3ffbbb20 0x400
d1b6a:0x3ffbbb50 0x400919c5:0x3ffbbb70
0x400d969f: esp_mqtt_client_publish at C:/msys32/home/ron/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1280 (discriminator 2)

0x400d6b12: mqtt_publish at C:/msys32/home/ron/esp/Tammy_Gateway/main/mqtt.c:57

0x400d4adf: mqttPublish at C:/msys32/home/ron/esp/Tammy_Gateway/main/alarm.c:141

0x400d4f26: alarmInit at C:/msys32/home/ron/esp/Tammy_Gateway/main/alarm.c:386

0x400d4837: Setup at C:/msys32/home/ron/esp/Tammy_Gateway/main/main.c:977

0x400d4864: app_main at C:/msys32/home/ron/esp/Tammy_Gateway/main/main.c:986

0x400d1b6a: main_task at C:/msys32/home/ron/esp/esp-idf/components/esp32/cpu_start.c:569

0x400919c5: vPortTaskWrapper at C:/msys32/home/ron/esp/esp-idf/components/freertos/port.c:403

Rebooting...
```

The problem appears to be in the way messages with QOS>0 are handled before a successful connection to the server.

david-cermak, 2019-11-16:
Hi @rrroonn 

Are you really on `v3.2-dev-1148-g96cd3b75c`? This is a very old version and looking at the fraction of code from mqtt_client you've posted it looks like a more recent one. Could you please share exact version you use? Is it 4.1 as indicated in the log?

As for QoS>0 messages, this is true, they are handled (queued) even if there's no connection to the server and one typical issue for continuous publish without connection is that the outbox can no longer hold these messages. This doesn't seem to be your problem, though.

rrroonn, 2019-11-16:
Hi David,

> Are you really on v3.2-dev-1148-g96cd3b75c? 
No, that was from the issue template - if I run 'git describe --tags'
```
$ git describe --tags
fatal: not a git repository (or any of the parent directories): .git
```
The version was recently updated from espressif website - how can I provide the detail you are after?

I was having problems in MQTT and downloaded the latest mqtt code from github (but problem remained).

Yes, I understand that messages are queued, but there is something wrong in the queuing process to return  'publish_message->data=00000000h', as the very next line dereferences this null pointer and hence the panic.

I am not a git guru, nor a linux guru.  I just install the environments as required and concentrate on coding the application.  Tell me how to provide any more information that you seek and I will endeavour to provide it.

thanks,

Ron

rrroonn, 2019-11-16:
I worked out how to get the git version:

v4.1-dev-369-g4dac7c7df

ESP-Marius, 2019-11-18:
@rrroonn  @david-cermak 
Seems like there are two issues here:

1. The message buffer is initialized in `esp_mqtt_connect` (`mqtt_msg_init`), and since your connection fails during `esp_transport_connect` this initialization is never done

2. `mqtt_msg_publish` is able to catch that the message cannot be created and returns a "fail message", but it seems like we doesn't properly detect this return value in `esp_mqtt_client_publish`, which leads the SW to continue with an invalid message.

rrroonn, 2019-11-18:
Yes, that sounds very plausible to me.  Thank you for looking into this.

closed 2020-01-10 by espressif-bot

enricop, 2020-01-11:
Hello @david-cermak , would it be possible to update the mqtt submodule also for esp-if `release/v4.0` branch?
We had a crash in this part of component code while testing on that branch.

david-cermak, 2020-01-12:
Hi @enricop , yes, this commit will be backported to all major releases.