espressif/esp-idf #4350: Unable to read characteristic descriptor upon ESP_GATTC_REG_FOR_NOTIFY_EVT event

joehui, 2019-11-16:
I am trying to use ESP32 as a BLE client. I had some problem trying to get BLE "notification" working.

Using  the official Gatt client example as a reference, I called esp_ble_gattc_register_for_notify() upon ESP_GATTC_SEARCH_CMPL_EVT event.  The issue is that I was not able to read the descriptors of the characteristic when ESP_GATTC_REG_FOR_NOTIFY_EVT was received.  Calling esp_ble_gattc_get_attr_count()  returns a count of zero.

I noticed that the ESP_GAP_BLE_AUTH_CMPL_EVT event (indicating pairing success) came quite a bit later after the ESP_GATTC_REG_FOR_NOTIFY_EVT event.

After spending quite a bit of time on this problem, I started to wonder if the descriptor info isn’t available yet when I made the register call at ESP_GATTC_SEARCH_CMPL_EVT.  With this idea in mind, I  moved the esp_ble_gattc_register_for_notify() code to the ESP_GAP_BLE_AUTH_CMPL_EVT event handler.  After that change, things started to work.

Does my change  make sense?   Is it guaranteed that char descriptor info will be available by the time ESP_GAP_BLE_AUTH_CMPL_EVT comes?

Thanks,
Joseph

Alvin1Zhang, 2020-01-08:
@joehui Thanks for reporting, and sorry for getting back to you so late. We already had a fix under internal review, will inform you once the fix is available on GitHub. Thanks.

Alvin1Zhang, 2020-05-27:
@joehui The issue is same with https://github.com/espressif/esp-idf/issues/4392 and the fix has been available at https://github.com/espressif/esp-idf/commit/6655fa8a933442ed67bb6498b6e496ef9941030e, thanks for reporting. Will close the ticket now, feel free to reopen if the issue still happens.

closed 2020-05-27 by Alvin1Zhang