espressif/esp-idf #4351: Please increase task stack size to at least 2048

liudr, 2019-11-17:
https://github.com/espressif/esp-idf/blob/d57890cdff10648371bc13956230635d2944d254/examples/peripherals/uart/uart_echo/main/uart_echo_example_main.c#L62

Off the bat this sample code fails if I add anything as much as a printf with arguments to the echo task. Please increase the stack allocation to at least 2048 so anyone trying to make some sense out of this sample code (and all other sample codes) won't end up getting just more confused. Cost me a few hours to finally pin-point it as stack issue. If it isn't broken don't fix it? What if it breaks under the slightest stress?

koobest, 2019-11-18:
We will fix this soon, thansk !!

negativekelvin, 2019-11-18:
I don't necessarily agree that stack size should be bumped arbitrarily to save users from themselves.  User needs to learn printf uses lots of stack but otherwise large stack may not be required.  Maybe a conditional ENABLE_DEBUG_PRINTF which would select for stack size.

liudr, 2019-11-18:
> 
> 
> I don't necessarily agree that stack size should be bumped arbitrarily to save users from themselves. User needs to learn printf uses lots of stack but otherwise large stack may not be required. Maybe a conditional ENABLE_DEBUG_PRINTF which would select for stack size.

I understand your intent to hold developers themselves responsible for shooting their feet since it is C. If you're a C programmer, you'd be gravitated towards printf. A sample code should anticipate a few common issues such as developers using printf and causing stack overflow. Not addressing a common issue negatively impacts the effectiveness of the sample code.

I'd like to be greeted by a friendly comment saying "Since printf needs a large stack size, we are setting stack to 2048 to prevent overflow in case you use printf." where the task is created. Another conditional or def is just going to complicate things unnecessarily besides requiring the same comments. What about LOGI() and LOGE()? Are they macros of printf? Are they thread safe versions of printf? So much to learn.

negativekelvin, 2019-11-18:
Yes at the very least a comment about the generous stack size.  Maybe there should be a whole set of 
[fatal errors](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html ) examples

`What about LOGI() and LOGE()? Are they macros of printf? Are they thread safe versions of printf? So much to learn.`

Yes.  And there is also nano printf. 
https://github.com/espressif/esp-idf/blob/81ca1c01395f604972fbf141cfbe49764a746023/components/newlib/Kconfig#L51-L70
https://github.com/espressif/esp-idf/blob/d57890cdff10648371bc13956230635d2944d254/components/wpa_supplicant/src/esp_supplicant/esp_wifi_driver.h#L21-L25

jhubbardsf, 2019-11-18:
> I understand your intent to hold developers themselves responsible for shooting their feet since it is C. If you're a C programmer, you'd be gravitated towards printf. A sample code should [...]

Just wanted to agree with this. I also painfully had to learn this stuff since I haven't always been a C developer. It would do a lot to increase ease of use and thus build the community more.

Alvin1Zhang, 2020-01-08:
@liudr Thanks for reporting. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-08 by Alvin1Zhang