espressif/esp-idf #4354: Installation on Win10 Fails if no Defender module is installed

atomicleads, 2019-11-17:
Installation fails and hungs if no Defender module is found on windows 10.
If there is no defender module, please just skip defender exceptions.

Running command: powershell -ExecutionPolicy ByPass -File "C:\Users\User\.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\User\.espressif\*.exe"
Import-Module : The specified module 'Defender' was not loaded because no valid module file was found in any module 
directory.
At C:\Users\User\.espressif\dist\tools_WD_excl.ps1:61 char:1
+ Import-Module Defender
+ ~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ResourceUnavailable: (Defender:String) [Import-Module], FileNotFoundException
    + FullyQualifiedErrorId : Modules_ModuleNotFound,Microsoft.PowerShell.Commands.ImportModuleCommand
 
C:\Users\User\.espressif\dist\tools_WD_excl.ps1 : Windows Defender cmdlets not available
    + CategoryInfo          : NotSpecified: (:) [Write-Error], NotSupportedException
    + FullyQualifiedErrorId : System.NotSupportedException,tools_WD_excl.ps1

igrr, 2019-11-18:
This seems to be the same issue as reported in #4225.

atomicleads, 2019-11-26:
yes, it's a dupe, but different OS.

closed 2019-11-26 by atomicleads

dxzl, 2020-02-05:
This issue is still happening with esp-idf-tools-setup-2.2.exe on Windows 10:

Running command: powershell -ExecutionPolicy ByPass -File "C:\Users\Me\.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\Me\.espressif\*.exe"                       
--------
Message box (OK):
Installation has failed with exit code 1
User chose OK.

pacucha42, 2020-02-24:
Hi,
this is weird behavior as the installer checks the WinDefender module existence first and disables the option checkbox when not found (the check is done as lookup for **%PSMODULEPATH%\Defender***). Furthermore, it's quite complicated to get rid of the Defender in Windows 10, unless you run Server version... would you make one more test, please? Attached you find updated PS1 scripts (another check for Defender cmdlet before actual import). Plus listing of your PSMODULEPATH would be helpful too - you can use wdmodule_listing.cmd frm the attachment. Well, and your OS version is interesting too :-) (run **systeminfo | findstr /B /C:"OS Name" /C:"OS Version"** in your cmdline)

Thank you very much
Martin
[tools_WD_update.zip](https://github.com/espressif/esp-idf/files/4244353/tools_WD_update.zip)

dxzl, 2020-02-24:
Hi Martin,
The error I experienced occurred with the "Register the ESP-IDF Tools Executables as Windows Defender exclusions" box checked. So I reinstalled with the box unchecked and got through the installation. It has been a while since I was working on this, but I seem to recall also having to install OpenSSL (but that might have been related to the PlatformIO ESP32 installation in VSCode - I just don't recall).
Here is my Windows 10 PSModulePath:
PSModulePath=C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\;C:\Program Files (x86)\Microsoft SQL Server\110\Tools\PowerShell\Modules\;C:\Program Files\Microsoft Message Analyzer\PowerShell\;C:\Program Files\WindowsPowerShell\Modules\;C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ResourceManager\AzureResourceManager\;C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\;C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\Storage\
Many thanks!

pacucha42, 2020-02-24:
Aaaaa, that's a big PSMODULEPATH field :-) Let me check whether we have troubles with the string parsing or such... it's quite unlikely your Windows 10 doesn't contain the Defender module so I guess the issue is elsewhere.

Thank you
Martin

zhang00956, 2020-07-08:
在win10上，我解决了这个问题。首先使用esp-idf-tools-setup.exe，在安装的时候不要勾选 regeister the ESP-IDF tools as windows definder。然后打开cmd窗口手动执行命令powershell -ExecutionPolicy ByPass -File "D:\.espressif\dist\tools_WD_excl.ps1" -AddExclPath "D:\.espressif\*.exe"，D:\.espressif需要改成你自己的路径。

vadaliya, 2020-11-07:
> On win10, I solved this problem. First use esp-idf-tools-setup.exe, do not check regeister the ESP-IDF tools as windows definder during installation. Then open the cmd window and manually execute the command powershell -ExecutionPolicy ByPass -File "D:.espressif\dist\tools_WD_excl.ps1" -AddExclPath "D:.espressif*.exe", D:.espressif needs to be changed to your own path.

Thanks... My Problem is Solved.. Using this method...