espressif/esp-idf #4356: Combining projects?

MusabAybek, 2019-11-17:
Hi. I'm new on Esp32 world as well as the coding world. So, maybe my question is so easy for you guys. I've already posted at esp32.com but couldn't get answer until now.
I'm working on 2 different projects and wanna combine them and build 1 project.
For example; I have example codes for printing some info on ILI9341 screen and I have example codes for using DS1307 module. I need to combine them and print the time info which is stored in DS1307, on ILI9341 screen. Both projects have different Makefiles, Components, etc. How to do that? Thanks in advance guys...

Alvin1Zhang, 2019-11-18:
@MusabAybek Thanks for reporting, would you please help share the post link on esp32.com? Thanks.

Alvin1Zhang, 2019-11-18:
@MusabAybek Thanks for reporting, checked that your post is here https://esp32.com/viewtopic.php?f=2&t=13164&p=51951#p51951. For how questions and third party issues, suggest to post on our forum at esp32.com. Will close the ticket and let us keep tracking in the post https://esp32.com/viewtopic.php?f=2&t=13164&p=51951#p51951. Thanks.

closed 2019-11-18 by Alvin1Zhang