espressif/esp-idf #4357: esp-tls: mbedtls_ssl_handshake returned -0x10

Kirankumar-xenlabs, 2019-11-18:
I am trying to upload the captured images on google cloud, it uploading up to 12 images after that it returning the following error,
E (600424) esp-tls: mbedtls_ssl_handshake returned -0x4d80
E (600434) esp-tls: Failed to open new connection
E (600434) TRANS_SSL: Failed to open a new connection
E (600434) HTTP_CLIENT: Connection failed, sock < 0

Regards,
Kiran

Kirankumar-xenlabs, 2019-11-20:
@igrr please help to fix it

mahavirj, 2019-11-20:
@Kirankumar-xenlabs 

Information you have provided is not per IDF github issue template, it is very difficult to help in such scenarios.

First of all, error you mentioned above corresponds to `#define MBEDTLS_ERR_MPI_ALLOC_FAILED                      -0x0010  /**< Memory allocation failed. */`, hence you might want to monitor free heap size in your application. APIs `esp_get_free_heap_size`, `esp_get_minimum_free_heap_size` and `heap_caps_get_largest_free_block` can help you for this purpose.

For memory leak analysis you may refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks

Kirankumar-xenlabs, 2019-11-20:
Thank you @mahavirj , I monitoring the free heap size, I think mbedtls is not releasing memory, please refer following log,
Photo N0. 3
I (1124844) example_google_storage: Taking picture...
I (1124934) example_google_storage: Setting Indian Timezone...
I (1124934) example_google_storage: Picture 240AC4BB3340/2019_11_20_12_30_03/2019_11_20_12_30_13.jpg taken.
I (1124934) example_google_storage: Uploading 240AC4BB3340/2019_11_20_12_30_03/2019_11_20_12_30_13.jpg...
Before Upload To GCS DRAM 4191004 IRAM 4198312
I (1124954) example_google_storage: Insert object [240AC4BB3340/2019_11_20_12_30_03/2019_11_20_12_30_13.jpg] in bucket [esp-32-cam.appspot.com]...
E (1129994) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (1129994) esp-tls: Failed to open new connection
E (1129994) TRANS_SSL: Failed to open a new connection
E (1129994) HTTP_CLIENT: Connection failed, sock < 0
After Upload To GCS DRAM 4190552 IRAM 4197860

mahavirj, 2019-11-20:
Sorry but I am unable to understand above log.

Please provide information in following order:

- ESP-IDF version
- Application `sdkconfig` file
- Application code (to understand if you are using same TLS connection for each upload or different one)
- Entire console log
- Any chance you are using cloud framework from https://github.com/espressif/esp-google-iot?

Kirankumar-xenlabs, 2019-11-20:
Thank you, @mahavirj, I am using ESP-IDF v3.3, I attached sdkconfig file, application code, and console log, please find the attachment and I am uploading the images on google cloud for that I am using the API https://github.com/raphaelbs/esp32-cam-ai-thinker 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3867839/sdkconfig.txt)
[code.txt](https://github.com/espressif/esp-idf/files/3867849/code.txt)
[esp_log.txt](https://github.com/espressif/esp-idf/files/3867858/esp_log.txt)

mahavirj, 2019-11-20:
Can you please try your application by changing to following configuration:
`make menuconfig -> Component config -> mbedTLS -> Memory allocation strategy -> External SPIRAM`

Kirankumar-xenlabs, 2019-11-20:
Ok, but I am not using external SPIRAM

negativekelvin, 2019-11-20:
Where are you getting these numbers? They can't be valid. DRAM 4190552 IRAM 4197860

Kirankumar-xenlabs, 2019-11-20:
printf("DRAM %u IRAM %u\n", heap_caps_get_free_size(MALLOC_CAP_8BIT), heap_caps_get_free_size(MALLOC_CAP_32BIT));

negativekelvin, 2019-11-20:
If you have 4mb free then you are using spiram

Kirankumar-xenlabs, 2019-11-20:
@negativekelvin I am not using spiram, in my project, I think mbedtls taking memory on every new connection.

mahavirj, 2019-11-26:
@Kirankumar-xenlabs Your `sdkconfig` has `SPIRAM` enabled, so you can still potentially try out suggestion from https://github.com/espressif/esp-idf/issues/4357#issuecomment-555885443?

Kirankumar-xenlabs, 2019-11-27:
Thank you so much @mahavirj  and @negativekelvin  I fixed it.

closed 2019-11-27 by mahavirj

vparuthi, 2020-03-03:
Hello, 

I was wondering if you could help me with an issue I am having regarding making https requests. My code seems to crash whenever I make an https request with esp_http_client but works fine for http requests. Below is my task that runs once a wifi connection is established:

```
void send_http_request(void *arg){
    ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
    esp_http_client_config_t http_config = {
            .url = "https://www.howsmyssl.com",
            .event_handler = http_event_handle,
            .method = HTTP_METHOD_GET,
            .cert_pem = "-----BEGIN CERTIFICATE-----\n ..."
    };

    esp_http_client_handle_t client = esp_http_client_init(&http_config);

    ESP_LOGI(WIFI_TASK_NAME, "Watermark: %i, free heap size: %i, %i", uxTaskGetStackHighWaterMark(NULL),
            esp_get_free_heap_size(),
             esp_get_minimum_free_heap_size());
    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        ESP_LOGI(WIFI_TASK_NAME, "Status = %d, content_length = %d",
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));
    }else{
        ESP_LOGI(WIFI_TASK_NAME, "send_http_request(): %i, %s", err, esp_err_to_name(err));
    }

    esp_http_client_cleanup(client);
    for(;;){}
}
```

The code seems to crash on the following line: 
```
    esp_err_t err = esp_http_client_perform(client);
```

with the following console output:

```
I (918) WIFI_TASK: Event = 0004
I (918) WIFI_TASK: SYSTEM_EVENT_STA_CONNECTED
I (3328) event: sta ip: 192.168.7.104, mask: 255.255.255.0, gw: 192.168.7.1
I (3328) WIFI_TASK: Event = 0007
I (3328) WIFI_TASK: SYSTEM_EVENT_STA_GOT_IP:192.168.7.104
I (3338) WIFI_TASK: Watermark: 2520, free heap size: 229772, 219344
E (3448) esp-tls: mbedtls_ssl_handshake returned -0x1c
I (3448) esp-tls: Certificate verified.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40127124  PS      : 0x00060a30  A0      : 0x80127a8d  A1      : 0x3ffb95a0  
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127124: [Errno 2] No such file or directory
A2      : 0xc6351bff  A3      : 0x00000190  A4      : 0x3ffb9510  A5      : 0x0000000c  
A6      : 0x3ffb9260  A7      : 0x00000003  A8      : 0x80123660  A9      : 0x3ffb9590  
A10     : 0xc6351bff  A11     : 0x00000000  A12     : 0x00000190  A13     : 0x3ffba36c  
A14     : 0x3ffb91d0  A15     : 0x00000019  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xc6351c07  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4000c46c: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4000c477: [Errno 2] No such file or directory

ELF file SHA256: e1b2cfd4353c7e3aaee1938c143dae2d2a224974d3f4b14c207552fa68671d98

Backtrace: 0x40127124:0x3ffb95a0 0x40127a8a:0x3ffb95c0 0x4015473c:0x3ffb95e0 0x40154f7b:0x3ffb9600 0x40154fbf:0x3ffb9620 0x401542bb:0x3ffb9640 0x40157489:0x3ffb9660 0x4014c274:0x3ffb9680 0x4014c5df:0x3ffb96a0 0x400d6f75:0x3ffb96c0
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127124: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40127a8a: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x4015473c: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40154f7b: [Errno 2] No such file or directory
xtensa-esp32-elf-addr2line -pfiaC -e /Users/vp/esp/Pebble_ESP32_Firmware/build/esp32_app 0x40154fbf: [Errno 2] No such file or directory
[... 98 lines cut ...]

#
# Certificates
#
CONFIG_MBEDTLS_PEM_PARSE_C=y
CONFIG_MBEDTLS_PEM_WRITE_C=y
CONFIG_MBEDTLS_X509_CRL_PARSE_C=y
CONFIG_MBEDTLS_X509_CSR_PARSE_C=y
CONFIG_MBEDTLS_ECP_C=y
CONFIG_MBEDTLS_ECDH_C=y
CONFIG_MBEDTLS_ECDSA_C=y
CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP224K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
CONFIG_MBEDTLS_ECP_NIST_OPTIM=y
```

Alvin1Zhang, 2020-03-04:
@vparuthi Hi, thanks for the updates. If your issue is not the same one as mentioned in this ticket, would you please help file another Github Issue? This would help us for better tracking of issues. Thanks for understanding.

vparuthi, 2020-03-04:
> @vparuthi Hi, thanks for the updates. If your issue is not the same one as mentioned in this ticket, would you please help file another Github Issue? This would help us for better tracking of issues. Thanks for understanding.

It isn't the exact same but the issue is similar. I'll make another issue :)

MarkEvens, 2021-08-06:
@mahavirj  How much minimum heap is required ? Because my free heap size is Free "48428" and still I am getting below error.
```
E (61836) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x10
I (61846) esp-tls-mbedtls: Certificate verified.
E (61846) esp-tls: Failed to open new connection
E (61846) TRANS_SSL: Failed to open a new connection
E (61856) HTTP_CLIENT: Connection failed, sock < 0
E (61856) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (61866) esp_https_ota: Failed to establish HTTP connection
```