espressif/esp-idf #4358: Olimex ESP32-PRO

davideferrario, 2019-11-18:
I am working on a project based on Olimex ESP32-PRO hardware.

I am able to compile/flash the hardwarebut when program starts I got this error message:

I (316) spi_flash: detected chip: generic
E (316) spi_flash: failed to get chip size
assertion "flash_ret == ESP_OK" failed: file "/opt/esp/esp-idf/components/esp32/cpu_start.c", line 431, function: start_cpu0_default
abort() was called at PC 0x400de89f on core 0
0x400de89f: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

negativekelvin, 2019-11-18:
https://esp32.com/viewtopic.php?t=12842

projectgus, 2020-10-11:
Sorry noone got back to you on this at the time. Closing as a duplicate of #4248.

closed 2020-10-11 by projectgus