espressif/esp-idf #4359: esp_wifi_80211_tx() not allowing action frames but documentation says they are supported.

georgcampana, 2019-11-18:
## Environment

- Development Kit:      Any
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it):  3.30300.190916 (3.3.0)
- Build System:  Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  2.50200.80 (5.2.0)
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
The `esp_wifi_80211_tx()` is documented to allow transmission of several packet types including the Action (type=0 management and subtype = 0xd action) frame.
By attempting to send an action frame the esp_wifi_80211_tx() function returns an error: 
` wifi: unsupport frame type: c0`

### Expected Behavior
The `esp_wifi_80211_tx()` should transmit the action frame
### Actual Behavior
The frame is not transmitted and the error ` wifi: unsupport frame type: c0` is returned

### Steps to repropduce

1. Compile the main.c source code at [https://gist.github.com/georgcampana/0d4098e87f26ff478480727dc2737c20](https://gist.github.com/georgcampana/0d4098e87f26ff478480727dc2737c20)
2. Run and check the debug output. The error is returned
3. Now decomment the `//#define PKT2SEND beacon_packet` and comment  `#define PKT2SEND action_packet`
4. Compile and run again
5. The beacon packet is correctly transmitted

### Code to reproduce this issue

```cpp
// see here on GIST a main.c to reproduce the issue
```
[https://gist.github.com/georgcampana/0d4098e87f26ff478480727dc2737c20](https://gist.github.com/georgcampana/0d4098e87f26ff478480727dc2737c20)

## Debug Logs

```
I (168) wifi: wifi driver task: 3ffb77d4, prio:23, stack:3584, core=0
␛[0;32mI (168) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (168) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
I (198) wifi: wifi firmware version: aeed694
I (198) wifi: config NVS flash: enabled
I (198) wifi: config nano formating: disabled
I (198) wifi: Init dynamic tx buffer num: 32
I (208) wifi: Init data frame dynamic rx buffer num: 32
I (208) wifi: Init management frame dynamic rx buffer num: 32
I (218) wifi: Init management short buffer num: 32
I (218) wifi: Init static rx buffer size: 1600
I (228) wifi: Init static rx buffer num: 10
I (228) wifi: Init dynamic rx buffer num: 32
␛[0;32mI (1178) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0␛[0m
I (1178) wifi: mode : softAP (d8:a0:1d:49:33:c1)
I (1178) wifi: Total power save buffer number: 16
I (1178) wifi: Init max length of beacon: 752/752
I (1188) wifi: Init max length of beacon: 752/752
I (1188) wifi: Set ps type: 0

E (1398) wifi: unsupport frame type: c0
E (1598) wifi: unsupport frame type: c0
E (1798) wifi: unsupport frame type: c0
. . . .
```

sagb2015, 2019-11-18:
Hi @georgcampana, FC[0] frame should be 0xd0 and not 0x0d. Please correct the same and you should see action frames over the air. Also, looks like you are trying to WiFiAware (NAN) functionality. FYI we are working on implementing the same and official NAN support will be present in a quarter or so.

sagb2015, 2019-11-18:
Feel free to reopen if case of doubts or issues.

closed 2019-11-18 by sagb2015

georgcampana, 2019-11-18:
Thx Sagar. You're right about the wrong Frame Control encoding. Don't know how I missed that one.