espressif/esp-idf #4361: [sugggestion] Add alarm API (absolute timestamp one shot timer) to esp_timer component

AloyseTech, 2019-11-18:
### Actual Behavior

esp_timer components offer a function to start a one shot timer : `esp_timer_start_once(timer, relative_time_from_now)`. This function allows to execute a callback in X microseconds from the current time. 

### Suggested Behavior

It could be very useful to have a similar function which would use an absolute timestamp : `esp_timer_start_alarm(timer, absolute_time)`. The implementation would be easy, based on the one shot timer (in esp_timer.c) : 
```
esp_err_t IRAM_ATTR esp_timer_start_alarm(esp_timer_handle_t timer, uint64_t timestamp_us)
{
    if (timer == NULL) {
        return ESP_ERR_INVALID_ARG;
    }
    if (!is_initialized() || timer_armed(timer)) {
        return ESP_ERR_INVALID_STATE;
    }
    timer_list_lock();
    timer->alarm = timestamp_us; // start_once use esp_timer_get_time() + timeout_us
    timer->period = 0;
#if WITH_PROFILING
    timer->times_armed++;
#endif
    esp_err_t err = timer_insert(timer);
    timer_list_unlock();
    return err;
}
```

(This suggestion is related to this forum post : https://www.esp32.com/viewtopic.php?f=2&t=13150)

0xjakob, 2019-11-20:
Hi @AloyseTech. Thanks for the suggestion! At the moment, however, we can't really see the use cases for this feature, since you can easily calculate the alarm time using get_current_time(). Is there a major drawback for your use case when not having this particular function?

AloyseTech, 2019-11-20:
I’m currently trying to implement a very tightly time constrained system were I timestamp an incoming event (GPIO edge interrupt) and I need to execute some action relative to this timestamp with a +/-20us precision. It’s the first time I have to deal with such timing constraints but I think that any call to esp_timer_get_time to calculate the alarm would add some error.

0xjakob, 2019-11-20:
OK, I understand. But still not sure whether esp_timer is the right component for that. It's backed by a FreeRTOS task which adds some overhead already. Another option would be using the other hardware timer of the ESP: https://gitlab.espressif.cn:6688/espressif/esp-idf/blob/master/components/driver/include/driver/timer.h.
I haven't looked into details there, but it's exclusively for users. So it's latency isn't dependent on other timers set by the system.

AloyseTech, 2019-11-20:
I tried this driver once but had some strange results (variable latency mainly). I will retry the timer_group driver and report back, thanks for the tip.

AloyseTech, 2019-11-20:
@0xjakob where can I find the crystal characteristics (initial accuracy, temperature drift...) of the esp32-wroom and the pico-d4?

igrr, 2019-11-20:
> I tried this driver once but had some strange results (variable latency mainly)

Some tips for getting more predictable interrupt latency (without resorting to [high-level interrupts](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/hlinterrupts.html)):

* pass [ESP_INTR_FLAG_IRAM](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html#c.ESP_INTR_FLAG_IRAM) and [ESP_INTR_FLAG_LEVEL3](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html#c.ESP_INTR_FLAG_LEVEL3) flags to intr_alloc_flags when calling `timer_isr_register`, and place your ISR code into IRAM. More on this [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers).
* Connect the timer interrupt to a less crowded core, (usually CPU 1). You can do this by creating a task, with affinity set to CPU1, and calling `timer_isr_register` there.

> where can I find the crystal characteristics (initial accuracy, temperature drift...) of the esp32-wroom and the pico-d4?

Regarding the hardware specs, I would suggest asking in https://esp32.com/viewtopic.php?f=5&t=905, since this is not an ESP-IDF related question.

AloyseTech, 2019-11-20:
Thanks @igrr for the tips.
Since the action I need to tightly execute is an SPI transmission, is there a better way than to have a task waiting (overhead caused by the xQueueReceived) for the interrupt and doing the transmission? Would it be better to to the SPI transmission inside the ISR directly and protect the SPI peripheral with a mutex?

igrr, 2019-11-20:
Having the (relatively high priority) task wait for a semaphore and unlocking the semaphore from the ISR seems to be the best option. SPI driver is not useable from the ISR since it relies on some blocking functions, internally.

Another option would be to use SPI HAL headers directly, bypassing the driver. However since we don't consider the HAL to be part of a public API, it may have non backwards compatible changes in future versions of IDF. Also if you run into an issue while using the HAL we will be less likely to support you in debugging it, than if you use the driver.

Regarding the external oscillator, I think it is possible, but again you will need to check the datasheet or the hardware design guidelines for the details.

AloyseTech, 2019-11-21:
Thanks @igrr for the help. I'm currently refactoring my code to avoid overhead and latency. For low latency and deterministic GPIO interrupt I expect it is better to use the gpio_isr_register API instead of gpio_install_isr_service/gpio_isr_handler_add, right?

AloyseTech, 2019-11-22:
I have been able to use the timer_group to generate accurate timing. It is in fact more reliable in term of latency than the esp_timer.

closed 2019-11-22 by AloyseTech