espressif/esp-idf #4362: LEDC set_duty_and_update function result in a deadlock

jilchab, 2019-11-18:
## Environment

Development Kit: none
Module or chip used: ESP32
IDF version: v4.0 branch master SHA:d57890cdff10648371bc13956230635d2944d254
Build System: Make
Compiler version: 1.22.0-80-g6c4433a
Operating System: macOS
Power Supply: external 3.3V

## Problem Description

I recently pull master to update ESP-IDF and I have issues with LEDC module
When I call `ledc_set_duty_and_update()` my program stop. After some tests I found out it might be a deadlock in ledc.c at the line 749: 
```
if (fade_mode == LEDC_FADE_WAIT_DONE) {
    xSemaphoreTake(s_ledc_fade_rec[speed_mode][channel]->ledc_fade_sem, portMAX_DELAY);
}
```

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
void app_main(void)
{
    ledc_channel_config_t ledc_channel;
    ledc_timer_config_t ledc_timer = {
        .duty_resolution    = LEDC_TIMER_13_BIT,
        .freq_hz            = 5000,                     
        .speed_mode         = LEDC_HIGH_SPEED_MODE,
        .timer_num          = LEDC_TIMER_0       
    };

    ledc_timer_config(&ledc_timer);

    ledc_channel.channel       = 0;
    ledc_channel.duty          = 8191;
    ledc_channel.gpio_num      = GPIO_NUM_19;
    ledc_channel.speed_mode    = ledc_timer.speed_mode;
    ledc_channel.timer_sel     = ledc_timer.timer_num;

    ledc_channel_config(&ledc_channel);
    ledc_fade_func_install(0);

    ESP_LOGI(TAG, "before");
    ledc_set_duty_and_update(ledc_channel.speed_mode, ledc_channel.channel, 0, 8191);
    ESP_LOGI(TAG, "after");

}

```
Log "after" is never displayed

koobest, 2019-11-19:
Hi, @jilchab 
There is a bug here, we will fix it soon.
[image]

thanks !!

physicsboy0709, 2019-11-26:
The ledc_set_duty_and_update function also reference to an internal function _ledc_fade_start(speed_mode, channel, LEDC_FADE_WAIT_DONE), which may also related to this.
I cannot use ledc_set_duty and ledc_update_duty to get rid of this. Still block the user tasks

koobest, 2020-01-08:
Hi, @physicsboy0709 
Can you provide your IDF version?

Alvin1Zhang, 2020-05-27:
@jilchab Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2020-05-27 by Alvin1Zhang