espressif/esp-idf #4363: app doesn't start

mzimmers, 2019-11-18:
## Environment

- Development Kit:     proprietary board
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0-beta2-70-g0a03a55c1
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
Application doesn't start. It's as though app_main() isn't called. Bootloader does start.

This problem began after I updated my IDF. It occurs both with v3.3 and v4.0 (program currently set up for 4.0).

### Expected Behavior
App_main() should run and many log messages should appear.

### Actual Behavior
I get the bootloader output, then the program seems to halt.

### Steps to reproduce
1. build
2. flash monitor

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

I'm attaching a .7z of my project

## Debug Logs

I (39) boot: ESP-IDF v4.0-beta2-70-g0a03a55c1 2nd stage bootloader
I (39) boot: compile time 07:23:02
I (67) boot: Enabling RNG early entropy source...
I (67) boot: SPI Speed      : 40MHz
I (67) boot: SPI Mode       : DIO
I (69) boot: SPI Flash Size : 4MB
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (91) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (99) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (106) boot:  3 factory          factory app      00 00 00010000 00130000
I (114) boot:  4 ota_0            OTA app          00 10 00140000 00130000
I (121) boot:  5 ota_1            OTA app          00 11 00270000 00130000
I (129) boot: End of partition table
I (133) boot: Defaulting to factory image
I (138) boot_comm: chip revision: 1, min. application chip revision: 0
I (145) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3b804 (243716) map
I (248) esp_image: segment 1: paddr=0x0004b82c vaddr=0x3ffb0000 size=0x03240 ( 12864) load
I (253) esp_image: segment 2: paddr=0x0004ea74 vaddr=0x40080000 size=0x00400 (  1024) load
I (255) esp_image: segment 3: paddr=0x0004ee7c vaddr=0x40080400 size=0x01194 (  4500) load
I (265) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xdbc18 (900120) map
I (618) esp_image: segment 5: paddr=0x0012bc38 vaddr=0x40081594 size=0x10044 ( 65604) load
I (646) esp_image: segment 6: paddr=0x0013bc84 vaddr=0x400c0000 size=0x00064 (   100) load
I (657) boot: Loaded app from partition at offset 0x10000
I (657) boot: Disabling RNG early entropy source...
I (658) cpu_start: Pro cpu up.
I (662) cpu_start: Application information:
I (667) cpu_start: Project name:     wifibutton
I (672) cpu_start: App version:      V0.1-24-g54a0f12-dirty
I (678) cpu_start: Compile time:     Nov 18 2019 07:23:03
I (684) cpu_start: ELF file SHA256:  7be83e3c44922c36...
I (690) cpu_start: ESP-IDF:          v4.0-beta2-70-g0a03a55c1
I (696) cpu_start: Starting app cpu, entry point is 0x400813d4
I (0) cpu_start: App cpu up.
I (707) heap_init: Initializing. RAM available for dynamic allocation:
I (714) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (720) heap_init: At 3FFCC6A0 len 00013960 (78 KiB): DRAM
I (726) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (732) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (739) heap_init: At 400915D8 len 0000EA28 (58 KiB): IRAM
I (745) cpu_start: Pro cpu start user code
I (784) spi_flash: detected chip: generic
I (785) spi_flash: flash io: dio
I (785) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (39) boot: ESP-IDF v4.0-beta2-70-g0a03a55c1 2nd stage bootloader
I (39) boot: compile time 07:29:16
I (67) boot: Enabling RNG early entropy source...
I (67) boot: SPI Speed      : 40MHz
I (67) boot: SPI Mode       : DIO
I (69) boot: SPI Flash Size : 4MB
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (91) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (99) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (106) boot:  3 factory          factory app      00 00 00010000 00130000
I (114) boot:  4 ota_0            OTA app          00 10 00140000 00130000
I (121) boot:  5 ota_1            OTA app          00 11 00270000 00130000
I (129) boot: End of partition table
I (133) boot: Defaulting to factory image
I (138) boot_comm: chip revision: 1, min. application chip revision: 0
I (145) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3b804 (243716) map
I (248) esp_image: segment 1: paddr=0x0004b82c vaddr=0x3ffb0000 size=0x0323c ( 12860) load
I (253) esp_image: segment 2: paddr=0x0004ea70 vaddr=0x40080000 size=0x00400 (  1024) load
I (255) esp_image: segment 3: paddr=0x0004ee78 vaddr=0x40080400 size=0x01198 (  4504) load
I (265) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xdbc18 (900120) map
I (618) esp_image: segment 5: paddr=0x0012bc38 vaddr=0x40081598 size=0x10040 ( 65600) load
I (646) esp_image: segment 6: paddr=0x0013bc80 vaddr=0x400c0000 size=0x00064 (   100) load
I (657) boot: Loaded app from partition at offset 0x10000
I (658) boot: Disabling RNG early entropy source...
I (658) cpu_start: Pro cpu up.
I (662) cpu_start: Application information:
I (667) cpu_start: Project name:     wifibutton
I (672) cpu_start: App version:      1
I (676) cpu_start: Compile time:     Nov 18 2019 07:28:59
I (682) cpu_start: ELF file SHA256:  0f0a62f07385e273...
I (688) cpu_start: ESP-IDF:          v4.0-beta2-70-g0a03a55c1
I (695) cpu_start: Starting app cpu, entry point is 0x400813d4
I (0) cpu_start: App cpu up.
I (705) heap_init: Initializing. RAM available for dynamic allocation:
I (712) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (718) heap_init: At 3FFCC6A0 len 00013960 (78 KiB): DRAM
I (724) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (731) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (737) heap_init: At 400915D8 len 0000EA28 (58 KiB): IRAM
I (743) cpu_start: Pro cpu start user code
I (782) spi_flash: detected chip: generic
I (783) spi_flash: flash io: dio
I (783) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

## Other items if possible
sdkconfig in .zip file; no coredump.
[wb_stripped.zip](https://github.com/espressif/esp-idf/files/3859382/wb_stripped.zip)

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2019-11-18:
```
extern "C" void app_main()
{
    vTaskDelay(portMAX_DELAY);
```

mzimmers, 2019-11-18:
I am a f'n idiot. Feel free to shoot me at a time of your choosing.

closed 2019-11-18 by mzimmers

Magnatto, 2022-02-02:
Just to clarifiy since i'm a f'n idiot too that also deserves a shot, for those less enlightened like me, there was a delay holding the entire program, watch out for your delays!