espressif/esp-idf #4364: Problem Downloading and Installing esp-idf tools when running esp-idf/install.sh

FourFUN, 2019-11-18:
## Environment

- Development Kit:    none or n/a
- IDF version:    ``v4.1-dev-1067-gfb3d648e0``
- Build System:    not there yet :)
- Compiler version:    trying to get this to install :)
- Operating System:     Linux Mint Linux Mint 19.1 Tessa
- Relevant Environment Variables: 
    - ``http_proxy`` and ``https_proxy`` are set and are working.
    - ``IDF_TOOLS_PATH=~/esp/esp-tools``
    - In a previous installation I deleted so I could upgrade, I had ``IDF_PATH=~/esp/esp-idf``. In the process of debugging I have this unset (although 'install.sh' seems to export it)

## Problem Description

I am attempting to update a previous installation of the ``esp-idf`` to the newest version. As per the recommendation at the bottom of the Get Started guide I deleted my ``esp-idf`` and ``xtensa-elf`` folders and started fresh.

I begin by following the instructions outlined here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#

I am able to install all the prerequisites and get the ``esp-idf`` downloaded (Step 1 and Step 2 from the guide).

Unfortunately  when trying to run the install script in ``esp-idf/install.sh`` I get the following output/error:
```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz to /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Done
WARNING: file size mismatch for /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp, expected 85459985, got 2219
WARNING: Failed to download file /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz to /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Done
WARNING: file size mismatch for /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp, expected 85459985, got 2219
WARNING: Failed to download file /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz to /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Done
WARNING: file size mismatch for /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp, expected 85459985, got 2219
WARNING: Failed to download file /home/FourFUN/esp/esp-tools/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/home/FourFUN/esp/esp-idf/tools/idf_tools.py", line 1349, in <module>
    main(sys.argv[1:])
  File "/home/FourFUN/esp/esp-idf/tools/idf_tools.py", line 1345, in main
    action_func(args)
  File "/home/FourFUN/esp/esp-idf/tools/idf_tools.py", line 1134, in action_install
    tool_obj.download(tool_version)
  File "/home/FourFUN/esp/esp-idf/tools/idf_tools.py", line 579, in download
    raise DownloadError()
__main__.DownloadError
```
What seems to be happening is that either the download is failing so I am getting some fragment of the full file or the expected sizes have not been updated.

I could "fix" this issue by heavily editing ``idf_tools.py`` to ignore the file sizes but that sounds more like covering up the problem rather than fixing it.

I could also download the tools from the link in ```esp_idf/tools/tools.json``: https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz but I am not sure how to then point the python script to the manually downloaded file.

Lastly I could build from scratch (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html). I will proceed to do this if I can't resolve this soon but regardless I thought I should point it out.

In any case thank you in advance for your help!

igrr, 2019-11-19:
The files are downloaded into /home/username/.espressif/dist (by default, if you haven't changed IDF_TOOLS_PATH). Please check the contents of the downloaded .tmp file, does it look like a truncated gzip archive, or is it an HTML 404 error page?
In any case, you can download the archive manually and place it in this dist directory, then run install.sh again.

FourFUN, 2019-11-20:
Hey @igrr! I have chosen to instead install the LTS v3.3 of the esp-idf for now since we will be doing production work. That said I did some investigating based on your suggestions and some of my own and observed the following.

I have indeed set the ``IDF_TOOLS_PATH`` to a different location (/home/username/esp/esp-tools/). I checked the contents of the .tmp file in esp-tools/dist and it is a file containing an HTML document with a 500 error page. The document mentions an SSL handshake failing but it seems it is auto-generated by my organization when a request does not go through our proxy.

Since I am behind my organizations proxy, what I believe is happening, is that the python2 ``urllib`` is not respecting my system proxy settings. Based on the documentation here (https://docs.python.org/2/library/urllib.html) I should be able to get everything to work if I have ``http_proxy`` and ``https_proxy`` environment variables set before calling ``urlopen``. Browsing through your install script though, it doesn't seem to use ``urlopen`` but only ``urlretrieve`` which doesn't seem to obey the proxy settings when used on its own.

Based on some light research, I have identified 2 solutions: 

The first is to is to instead use  ``urllib.urlopen``, an example of which can be seen here https://stackoverflow.com/questions/28370820/use-urllib-urlretrieve-and-ignore-proxy (note that here proxy settings are explicitly ignored but the principal is the same).

The second is to use ``urllib.FancyURLOpener`` and then either the ``read`` or perhaps even the ``retrieve`` methods of the class. An example using the first method but completely ignoring proxy settings can be seen d in the examples at the bottom of the documentation. This might be the preferred method for you since you make extensive use of the reporthooks in ``urllib.urlretrieve``.

Hopefully this is all useful to you guys. If I have some time I can try my recommended fix locally and report back. I can let you decide if you wish to close this issue or leave it open for you to track.

Thank you for your help!

igrr, 2019-11-20:
Thanks, if one of the options works for you, please open a PR and we will merge it. The only constraint is that we need to support both python 2.7 and >=3.5.

Alvin1Zhang, 2020-01-08:
@FourFUN Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

FourFUN, 2020-01-29:
Hello everyone, Really sorry for the late reply. I have unfortunately been busy with other work stuff and have not gotten back to this.

That said I got back to this today and tried the following. In ``install.sh`` I change all the commands that call python to include a ``python3`` in front of them. Specifically my ``install.sh`` looks like this:
```
#!/usr/bin/env bash

set -e
set -u

export IDF_PATH=$(cd $(dirname $0); pwd)

echo "Installing ESP-IDF tools"
python3 ${IDF_PATH}/tools/idf_tools.py install

echo "Installing Python environment and packages"
python3 ${IDF_PATH}/tools/idf_tools.py install-python-env

basedir="$(dirname $0)"
echo "All done! You can now run:"
echo ""
echo "  . ${basedir}/export.sh"
echo ""
```

By making this change it seems the Download problem at least resolves itself and I am now downloading files happily! By my estimation the proxy issues go away when using ``python3`` because it properly pulls the ``http_proxy`` environment variable (as the documentation even for ``python2`` said it should).

All that said I am not sure what the "long-term" effects of using ``python3`` are. The documentation prerequisites all seem to suggest ``python2`` (Note: The Python people urge me to suggest you guys move to ``python3`` :) ) so I am hoping nothing breaks. The installed python environment seems to name itself ``idf4.1_py3.6_env`` so perhaps all this is moot.

As a selfish request I would like to leave this issue open a little longer just so I can make sure everything runs and works and I can close it myself probably next week.

igrr, 2020-01-29:
Good to know that you got it working. It should be no problem to use Python 3: IDF is compatible with it and we test our scripts with both python 2 and 3. The issue is partially in letting install.sh know that the default 'python' you wish to use is Python 3. On Windows, there is a canonical way of enumerating the existing Python installations, so the users of IDF Tools Installer for Windows can choose the interpreter they prefer. On Linux, we have to depend on `python` pointing to the interpreter you wish to use.

If you can't change the interpreter which `python` symlink points to (it may break some other software), one option is to use a tool such as direnv. In the directories where you make use of ESP-IDF, create an .envrc file which will add the path to Python 3 as the first entry in PATH. This way `python` will point to python 3 whenever you are working inside these directories.

Unfortunately we can't hardcode `python3` in install.sh since e.g. on macOS only Python 2 is shipped by default. Using `python` and relying on it to point to the interpreter of choice seems to be a more portable solution.

keelung-yang, 2020-05-11:
@igrr No need to hard code python3: Use python to determine interpreter 2 or 3 (python --> python3 on Arch/Manjaro Linux) and OS firstly, then use python3 as default if it exists.

liuziangexit, 2020-06-09:
> The files are downloaded into /home/username/.espressif/dist (by default, if you haven't changed IDF_TOOLS_PATH). Please check the contents of the downloaded .tmp file, does it look like a truncated [...]

Thx, works for me. 
For some unknown reason, the install.sh script can not download those files in my mac, the size of downloaded file is always 0...

carlsa, 2021-03-03:
In Ubuntu (and Mint) you can install the `python-is-python3` package that will make the default python interpreter be python3.

tom-borcin, 2021-03-26:
Hello @FourFUN, we are closing this issue. URLretreive was updated and there is also an external tool for download.

If you want to keep this issue open, please let me know. Thank you!

closed 2021-03-26 by tom-borcin