espressif/esp-idf #4365: idf.py not working on Windows 10-64 bit [CMAKE] since new toolchain release

hmcosentini, 2019-11-19:
## Environment

- Development Kit:        none - propietary design
- Kit version:                  none
- Module or chip used:  |ESP32-WROOM-32U|
- IDF version:                  v4.1-dev-1067-gfb3d648e0
- Build System:              [CMake]
- Compiler version:       xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows 10 64bit]
- Power Supply:         [external 5V]

## Problem Description

1.-The last time I build a project was the 11/11. Today tried to build a project and I got an error, sadly I didn't save the log.

2- Then I read that the toolchain was updated so I synced the esp-idf repository then executed install.bat and export.bat
I always had a problem with this to tools to get the windows **env variable path** updated, so I manually change the path 
_C:\Users\hmcos\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin_
_C:\Users\hmcos\.espressif\tools\xtensa-esp32s2-elf\esp-2019r2-8.2.0\xtensa-esp32s2-elf\bin_

3-Then I executed the **Open ESP-IDF Command Prompt (cmd.exe)**

I got the following log:
**Using Python in C:\Users\hmcos\AppData\Local\Programs\Python\Python36\
Python 3.6.8
Using Git in C:\Program Files\Git\cmd\
git version 2.23.0.windows.1
Setting IDF_PATH: E:\__SOFT_FIRMWARE\esp-idf

Adding ESP-IDF tools to PATH...
    E:\__SOFT_FIRMWARE\esp-idf\tools

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "E:\__SOFT_FIRMWARE\esp-idf\install.bat"**

4- I ran again install.bat

**E:\__SOFT_FIRMWARE\esp-idf>install
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Installing Python packages from E:\__SOFT_FIRMWARE\esp-idf\requirements.txt
Processing e:\__soft_firmware\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 4)) (41.6.0)
Requirement already satisfied: click>=5.0 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from -r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 13)) (0.25)
Requirement already satisfied: windows-curses in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from esp-windows-curses==0.1->-r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 18)) (2.1.0)
Requirement already satisfied: six>=1.4.1 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from cryptography>=2.1.4->-r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 11)) (1.13.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from cryptography>=2.1.4->-r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 11)) (1.13.2)
Requirement already satisfied: pycparser in c:\users\hmcos\.espressif\python_env\idf4.1_py3.6_env\lib\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r E:\__SOFT_FIRMWARE\esp-idf\requirements.txt (line 11)) (2.19)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py) ... done
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-cp36-none-any.whl size=1177 sha256=5d5177f0fcec1ac278614e8a4094dd2d038b542e7376638f2adcf3a5352738b5
  Stored in directory: C:\Users\hmcos\AppData\Local\pip\Cache\wheels\a6\91\3f\12b7948e8a810a7f8e2467f1daa4693ffea988fb322ff10258
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1
All done! You can now run:
   export.bat**

5- I ran export.bat

**E:\__SOFT_FIRMWARE\esp-idf>export.bat
Setting IDF_PATH: E:\__SOFT_FIRMWARE\esp-idf

Adding ESP-IDF tools to PATH...
No directories added to PATH:

E:\__SOFT_FIRMWARE\esp-idf\tools;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\STMicroelectronics\st_toolset\asm;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Files\cpp\bin\Intel64;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\PuTTY\;C:\Program Files (x86)\GNU Tools ARM Embedded\8 2019-q3-update\bin;C:\Users\hmcos\AppData\Local\Programs\Python\Python36\Scripts\;C:\Users\hmcos\AppData\Local\Programs\Python\Python36\;C:\Users\hmcos\AppData\Local\Microsoft\WindowsApps;C:\Users\hmcos\AppData\Local\GitHubDesktop\bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Users\hmcos\AppData\Local\Microsoft\WindowsApps;C:\Users\hmcos\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin;C:\Users\hmcos\.espressif\tools\xtensa-esp32s2-elf\esp-2019r2-8.2.0\xtensa-esp32s2-elf\bin;C:\Users\hmcos\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Users\hmcos\.espressif\tools\cmake\3.13.4\bin;C:\Users\hmcos\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190708\openocd-esp32\bin;C:\Users\hmcos\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;C:\Users\hmcos\.espressif\tools\ninja\1.9.0\;C:\Users\hmcos\.espressif\tools\idf-exe\1.0.1\;C:\Users\hmcos\.espressif\tools\ccache\3.7\;C:\Users\hmcos\.espressif\python_env\idf4.1_py3.6_env\Scripts;C:\Users\hmcos\AppData\Local\Programs\Microsoft VS Code\bin;

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "E:\__SOFT_FIRMWARE\esp-idf\install.bat"**

I synced the esp-idf local branch.

I believe the new toolchain is expecting to be in a 32bit environment but that's an error because I changed to the CMAKE toolchain because the 64bit was supported.

I hope you can help me to find the problem, 
Thanks

hmcosentini, 2019-11-19:
@dobairoland Hi I saw you are the one working on the **requirements.txt** and the  **tools/check_python_dependencies.py**
I installed MSYS2 when I was working with the legacy building toolchain. I never uninstalled it and I started working with the CMAKE which for me is a great improvement. I realize that I was having a problem with the requirement so I tried uninstalling MYSYS2 to see if I can run the **idf.py** But I still got the error message. 
As a provisional solution, I commented the line **#18 file://${IDF_PATH}/tools/kconfig_new/esp-windows-curses; sys_platform == 'win32'** in requirements.txt
And I was able to buil , flash execute menuconfig, monitor etc...  Let me know If I need to change semething else.

igrr, 2019-11-19:
> I always had a problem with this to tools to get the windows env variable path updated, so I manually change the path

I think this is actually the root cause of the error.
You are adding IDF tools into global PATH environment variable, placing the path of the Python environment somewhere close to the end of the list. As a result, some other python.exe (not the one from the IDF environment) gets executed when you run idf.py. However the packages have been installed into the IDF environment.

I would suggest to backtrack a bit, remove IDF related paths from global PATH variable, and provide us with the relevant log for the "problem with this to tools to get the windows env variable path" issue.

dobairoland, 2019-11-19:
I agree with the previous comment. You are probably invoking a Python from MSYS2 which was installed globally (with Windows installer) and that is why it is trying to install `esp-windows-curses` which is not required for MSYS2.

`E:__SOFT_FIRMWARE` seems to missing a backslash after the drive letter. I think that is the main reason of the failure in CMD. The root cause: `Setting IDF_PATH: E:__SOFT_FIRMWARE\esp-idf`

hmcosentini, 2019-11-19:
Thanks, guys for the prompt answer!!! I will clean install python and the toolchain. I have another question I used **esp-idf-tools-setup-2.1.exe** to install the toolchain but it seems is having a problem to set the **env variables**. I have the .spressif/ in C: and the esp-idf en e: drive. DO that .exe run install.bat and export.bat at the end? Because I can't get a log from the window.

Thanks again for you support

igrr, 2019-11-19:
> I used esp-idf-tools-setup-2.1.exe to install the toolchain but it seems is having a problem to set the env variables.

Can you please describe, how does the problem manifest itself? Are you getting an error message, so some other unexpected behavior?

hmcosentini, 2019-11-19:
Well, I proceed as follow:
1. uninstall msys2
2. uninstall idf tools
3. uninstall python
4. Cleaned register/deleted env var
5. Installed using idf2.1

Got installation failed error code 1

See the error log:
**2019-11-19 17:18:38.797   Log opened. (Time zone: UTC-03:00)
2019-11-19 17:18:38.797   Setup version: Inno Setup version 6.0.2 (u)
2019-11-19 17:18:38.797   Original Setup EXE: E:\__Instaladores\__ELECTRONICA-MECANICA\ESP32\esp-idf-tools-setup-2.1.exe
2019-11-19 17:18:38.797   Setup command line: /SL5="$80664,206263708,721408,E:\__Instaladores\__ELECTRONICA-MECANICA\ESP32\esp-idf-tools-setup-2.1.exe" 
2019-11-19 17:18:38.797   Windows version: 10.0.17763  (NT platform: Yes)
2019-11-19 17:18:38.797   64-bit Windows: Yes
2019-11-19 17:18:38.797   Processor architecture: x64
2019-11-19 17:18:38.797   User privileges: Administrative
2019-11-19 17:18:38.797   Administrative install mode: No
2019-11-19 17:18:38.797   Install mode root key: HKEY_CURRENT_USER
2019-11-19 17:18:38.797   64-bit install mode: Yes
2019-11-19 17:18:38.797   Created temporary directory: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp
2019-11-19 17:18:38.813   -- DLL function import --
2019-11-19 17:18:38.813   Function name: idpAddFile
2019-11-19 17:18:38.813   DLL name: files:idp.dll
2019-11-19 17:18:38.813   Extracting temporary file: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.813   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.813   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpAddFileComp
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpAddMirror
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpAddFtpDir
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpAddFtpDirComp
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpClearFiles
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpFilesCount
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
2019-11-19 17:18:38.844   Successfully imported the DLL function. Delay loaded? No
2019-11-19 17:18:38.844   -- DLL function import --
2019-11-19 17:18:38.844   Function name: idpFtpDirsCount
2019-11-19 17:18:38.844   DLL name: files:idp.dll
2019-11-19 17:18:38.844   Dest DLL name: C:\Users\hmcos\AppData\Local\Temp\is-63L17.tmp\idp.dll
2019-11-19 17:18:38.844   Importing the DLL function.
[... 365 lines cut ...]
                          ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '\\\\${IDF_PATH}\\tools\\kconfig_new\\esp-windows-curses'
                          
                          Creating a new Python environment in C:\Users\hmcos\.espressif\python_env\idf4.1_py3.7_env
                          Installing virtualenv
                          Installing Python packages from E:\__SOFT_FIRMWARE\esp-idf\requirements.txt
                          Traceback (most recent call last):
                            File "E:\__SOFT_FIRMWARE\esp-idf\tools\idf_tools.py", line 1349, in <module>
                              main(sys.argv[1:])
                            File "E:\__SOFT_FIRMWARE\esp-idf\tools\idf_tools.py", line 1345, in main
                              action_func(args)
                            File "E:\__SOFT_FIRMWARE\esp-idf\tools\idf_tools.py", line 1163, in action_install_python_env
                              subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
                            File "C:\Users\hmcos\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 347, in check_call
[... cut ...]
2019-11-19 17:23:18.848   --------
2019-11-19 17:23:18.848   Message box (OK):
                          Installation has failed with exit code 1
2019-11-19 17:23:54.443   User chose OK.
2019-11-19 17:23:54.449   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2019-11-19 17:24:00.175   User chose Yes.
2019-11-19 17:24:00.222   CurStepChanged raised an exception.**

igrr, 2019-11-19:
> No such file or directory: '\\${IDF_PATH}\tools\kconfig_new\esp-windows-curses

This is the same issue as reported in https://github.com/espressif/esp-idf/issues/4341. The fix for it will be merged soon. In the meantime you can use IDF release/v4.0 branch, it doesn't have this issue.

hmcosentini, 2019-11-19:
OK, thanks I changed to Branch **release/v4.0**  and is working.
Thanks

igrr, 2019-11-20:
Okay, I will close this issue as duplicate. You can click "Subscribe" button in the linked issue, to get notified when the bug is fixed.

closed 2019-11-20 by igrr

hmcosentini, 2019-11-20:
Okay thanks

mazorvanunu, 2020-09-13:
did anybody from espressif ever even tried to read/follow the instructions for "installing" esp idf? have you ever tried it with multiple versions? does anybody from espressif really feels the "how to's" are worth anything? why is this organization so broken??? how come you get to be there while other good honest script kiddies/web developers would like to work where you sit today (and would probably do a much better job)?? ALL RUBISH!, mongose and arduino works much better (and they rely on much older versions than current idf version, but implement everything on their own obviously.. since idf cannot be trusted), is this about money or anything like that? 

why wont you just direct users to use some other platform other than saying IDF has "everything"???.... maybe every bug conceivable to the human mind...

also, all this cmake and make and what not - you took it to a level i really did not thought was possible, such a poor usage for this tools! despicable.

versamodule, 2020-09-14:
I know this is old, but I am having the same issue on WIn10, using the release/4.1.

ryan4volts, 2020-09-15:
Commenting out that line in the requirements.txt file to bring it back to how v4.0 worked appears to compile fine.