espressif/esp-idf #4366: 蓝牙模块初始化失败 failed to load RF calibration data (0x1102)

Tflgit, 2019-11-19:
我正在使用一块ESP32-LyraT测试蓝牙功能，我烧写并运行“examples/bluetooth/bluedroid/ble/ble_compatibility_test”项目，代码分支是"release/v4.0"分支。
运行出现问题，主要log如下

I (30) boot: ESP-IDF v4.0-beta1-180-ga21eb04 2nd stage bootloader
I (30) boot: compile time 15:18:23
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed : 40MHz
I (41) boot: SPI Mode : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label Usage Type ST Offset Length
I (60) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (67) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (74) boot: 2 factory factory app 00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x303a0 (197536) map
I (164) esp_image: segment 1: paddr=0x000403c8 vaddr=0x3ffbdb60 size=0x02e3c ( 11836) load
I (169) esp_image: segment 2: paddr=0x0004320c vaddr=0x40080000 size=0x00400 ( 1024) load
0x40080000: _WindowOverflow4 at /home/tfl/workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (172) esp_image: segment 3: paddr=0x00043614 vaddr=0x40080400 size=0x0c9fc ( 51708) load
I (202) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x7ca88 (510600) map
0x400d0018: _flash_cache_start at ??:?

I (381) esp_image: segment 5: paddr=0x000ccaa8 vaddr=0x4008cdfc size=0x05374 ( 21364) load
0x4008cdfc: multi_heap_internal_unlock at /home/tfl/workspace/esp/esp-idf/components/heap/multi_heap.c:380
(inlined by) multi_heap_malloc_impl at /home/tfl/workspace/esp/esp-idf/components/heap/multi_heap.c:446

I (401) boot: Loaded app from partition at offset 0x10000
I (401) boot: Disabling RNG early entropy source...
I (401) cpu_start: Pro cpu up.
I (405) cpu_start: Application information:
I (410) cpu_start: Project name: ble_compatibility_test
I (416) cpu_start: App version: v4.0-beta1-180-ga21eb04
I (422) cpu_start: Compile time: Nov 13 2019 15:18:27
I (429) cpu_start: ELF file SHA256: 70a1568df114c7a0...
I (434) cpu_start: ESP-IDF: v4.0-beta1-180-ga21eb04
I (441) cpu_start: Starting app cpu, entry point is 0x400812f0
0x400812f0: call_start_cpu1 at /home/tfl/workspace/esp/esp-idf/components/esp32/cpu_start.c:272

I (0) cpu_start: App cpu up.
0x40080000: _WindowOverflow4 at /home/tfl/workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

D (521) memory_layout: Reserved memory range 0x40080000 - 0x4009216f
0x40080000: _WindowOverflow4 at /home/tfl/workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (725) heap_init: Initializing. RAM available for dynamic allocation:
D (732) heap_init: New heap initialised at 0x3ffaff10
I (737) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (743) heap_init: New heap initialised at 0x3ffb6388
I (748) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
D (754) heap_init: New heap initialised at 0x3ffb9a20
I (759) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (765) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
D (772) heap_init: New heap initialised at 0x3ffc8c38
I (777) heap_init: At 3FFC8C38 len 000173C8 (92 KiB): DRAM
I (783) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (789) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (796) heap_init: New heap initialised at 0x40092170
I (801) heap_init: At 40092170 len 0000DE90 (55 KiB): IRAM
I (807) cpu_start: Pro cpu start user code
D (819) clk: RTC_SLOW_CLK calibration value: 3277286
D (829) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (829) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (830) intr_alloc: Connected src 24 to int 9 (cpu 0)
D (834) FLASH_HAL: extra_dummy: 1
D (838) spi_flash: trying chip: issi
D (841) spi_flash: trying chip: generic
I (845) spi_flash: detected chip: generic
I (850) spi_flash: flash io: dio
D (854) efuse: coding scheme 0
D (857) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (864) efuse: coding scheme 0
D (867) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
I (874) cpu_start: Chip Revision: 1
I (879) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (894) heap_init: New heap initialised at 0x3ffe0440
D (904) heap_init: New heap initialised at 0x3ffe4350
D (914) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (914) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (1234) BTDM_INIT: Release DRAM [0x3ffb2730] - [0x3ffb6388]
I (1234) BTDM_INIT: BT controller compile version [62aac70]
D (1234) BTDM_INIT: .data initialise [0x3ffae6e0] <== [0x4000d890]
D (1244) BTDM_INIT: .bss initialise [0x3ffb0000] - [0x3ffb09a8]
D (1254) BTDM_INIT: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]
D (1254) BTDM_INIT: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]
D (1264) BTDM_INIT: .bss initialise [0x3ffb8000] - [0x3ffb9a20]
D (1264) BTDM_INIT: .bss initialise [0x3ffbdb28] - [0x3ffbdb5c]
I (1274) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (1284) efuse: coding scheme 0
D (1284) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (1294) efuse: coding scheme 0
D (1294) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (1304) efuse: coding scheme 0
D (1304) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1314) efuse: coding scheme 0
D (1314) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1324) efuse: coding scheme 0
D (1324) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1334) efuse: coding scheme 0
D (1334) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1344) efuse: coding scheme 0
D (1344) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
**D (1354) phy_init: loading PHY init data from application binary
D (1364) nvs: nvs_open_from_partition phy 0
D (1364) phy_init: esp_phy_load_cal_data_from_nvs: failed to open NVS namespace (0x1102)
W (1374) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration**
D (1384) efuse: coding scheme 0
D (1384) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (1394) efuse: coding scheme 0
D (1394) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (1404) efuse: coding scheme 0
D (1404) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1414) efuse: coding scheme 0
D (1414) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1424) efuse: coding scheme 0
D (1424) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1434) efuse: coding scheme 0
D (1434) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1444) efuse: coding scheme 0
D (1444) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
ets Jun 8 2016 00:22:57

请问这个如何解决。

Tflgit, 2019-11-20:
OK，Thanks！

closed 2019-11-20 by Tflgit

alessandroAmedei, 2019-12-29:
You said this is a duplicate of another post, but in the other post they said that post is a duplicate of this one. For this, the issue is not resolved.
I am getting this at `EEPROM.begin();`
`[E][EEPROM.cpp:73] begin(): Unable to open NVS namespace: 4353`

gengyuchao, 2019-12-30:
As mentioned above, this is not a bug, please check your nvs_partition are correct. 
You can refer: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/RF_calibration.html