espressif/esp-idf #4367: How to call ESP_GATTC_NOTIFY_EVT in "spp_client_demo" example

ROYMURLI, 2019-11-19:
OS- Windows (Developemnt in msys32)
IDF- v3.3-beta1-136-g97eecfa1b

I am working on BLE to send and recieve data. I used "BLE_SPP_CLIENT" example for this. I connected this to a BLE 5.0 device which is HC-42 and also edited the "ESP_GATT_SPP_SERVICE_UUID" with its service UUID. I have written a program in HC-42 based system in which it can receive any input through BLE (of ESP32) and then transmit data accordingly to ESP32 via BLE.

Now I am able to connect with BLE (HC-42) by edited "static const char device_name[] = "HC-42";" this and then also able to write character to HC-42 from ESP32. But I am unable to recieve any data from HC-42 system to ESP32. For confirmation I used android BLE scanner app to send and get data from HC-42 system which is working fine. 

Do I need to add NOTIFY_CHAR UUID of HC-42 for incoming data. As I sew and tried the code of "gattc_demo.c" of "gatt_client" example. In this I am able to recieve the data but not able to write it, just vice versa of previous. Please let me know how can I read or recive data in BLE_SPP_CLIENT or write data in gattc_demo.c example. Both the example with little changes are attached below.
[spp_client_demo - Copy.docx](https://github.com/espressif/esp-idf/files/3862526/spp_client_demo.-.Copy.docx)
[gattc_demo - Copy.docx](https://github.com/espressif/esp-idf/files/3862769/gattc_demo.-.Copy.docx)

Please change the .docx to .c

ROYMURLI, 2019-11-21:
Any help on this @blueMoodBHD

gengyuchao, 2019-11-26:
Hello, do you mean you want to use the function GATT_NOTIFY to transmit data? Recently I find a problem with the GATT_NOTIFY function that may cause your case. 

You can try to use this patch to solve the problem. And the available branch of this path is release/v3.3 .

[fix_pointer_type_3.3.zip](https://github.com/espressif/esp-idf/files/3891693/fix_pointer_type_3.3.zip)
(First, you need to unzip the attachment file to your computer. )

You can use this command
`git apply you_file_location/fix_pointer_type_3.3.patch`

ROYMURLI, 2019-11-26:
Hi, Thanks for your reply. I want to use GATT_NOTIFY to recieve data. 
Let me explain you what I am doing. I am using HC-42 module as a server with a MCU to recieve and transmit data. If HC-42 will recieve "S" then it should start transmitting sensor data to client. Now for client I am using ESP32 to write "S" to HM-42 and then receive data accordingly. To check my server part I used a android BLE scanner app to send the "S" and then recieve data accodingly. 
Now, To implement it with ESP32 (as a client). I used "BLE_SPP_CLIENT" example and edited the "ESP_GATT_SPP_SERVICE_UUID" with HC-42 (server BLE) service UUID. 

 I am able to connect with HC-42 by edited "static const char device_name[] = "HC-42";" and then also able to write character to HC-42 from ESP32. But I am unable to recieve any data from HC-42 to ESP32 as it is transmitting but there is no event to recieve data in ESP-32. Do I need to add NOTIFY_CHAR UUID of HC-42 for incoming data, but this variable is not defined in this example.

As I saw and also tried the code of "gattc_demo.c" of "gatt_client" example. In this I am able to recieve the data but not able to write it, just vice versa of previous. In this NOTIFY_EVT called after connected to HC-42 and resposible for receiving data in ESP-32 from HC-42 module. But in this I am unable to send/write "S" to server (HC-42). 
Please let me know if you need anything more in this. Meanwhile I will try your patch.

gengyuchao, 2019-11-26:
I have understood your issue and I will try to reproduce it on my environment.

gengyuchao, 2019-11-28:
Hello, I found you need to register the notify function for the attribute. 

If you want to receive the notify messages, you need to call the following API to register the notify function.
```c
esp_err_t esp_ble_gattc_register_for_notify (esp_gatt_if_t gattc_if,
                                             esp_bd_addr_t server_bda, uint16_t handle)
```

For the spp_client_demo, the notify function will be registered by ESP_GATTC_CFG_MTU_EVT when connecting with the spp_server_demo.  But in this demo, the client and server need to match. If the client checks that the server service does not match the preset protocol, the client cannot register the notify function successfully.

Here is an example I set a small change on spp_client_demo to make it can register the notify function with the other device (rather than the spp_server).

Note: only the characteristic attribute can be notified.

And for more details,  you can find in the tutorials of the Gatt_Client. This is the document link.

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_client/tutorial/Gatt_Client_Example_Walkthrough.md#registering-for-notifications

ROYMURLI, 2019-11-29:
Hi, Thanks for your reply. I tried your code but still it is not receiving data from HC-42 (server). Screenshot is attached here. I checked GATT_CLIENT example in this I am able to recieve the data from HC-42 (Server) but not able to write any data to server as I mentioned earlier. 
I think we have to define NOTIFY_UUID of "server" in "ble_spp_cleint" example same as "gatt_client" example and then have to  call notify function. I tried to implement the notify function and defined NOTIFY_UUID in "ble_spp_client" to receive data but that is not working and crashing. 
[image]

gengyuchao, 2019-11-29:
You don't need to set a special uuid for notify, because notify is a method that the server can send the attribute data to master. 
From you log I found that the nottify has been registed. 
If you want to send data form client to server, please use esp_ble_gattc_write_char function. If you want to send data from server to client, please use notify. 

And I write a server that can send notify to the master when it got 's' from spp_client used gattc_write_char function. 
If you need I can show you the server code to see what's the different with our server action.
Screenshot is attached here.

[image]

ROYMURLI, 2019-11-29:
I again tried with your "ble_spp_client.c" example. But, I did not received any data from server. I can't change the server code. It is pre defined with some command to get data through BLE. As I mentioned earlier It is working fine as I checked it through any BLE android application. I am unable to call ESP_GATTC_NOTIFY_EVT in this "ble_spp_client" for receiving data. In the "gatt_client.c" example I am unable to call "esp_ble_gattc_write_char" to write data. The connection is established successfully in the example but the problem is still there.

For this I used a small arduino code and connected HC-42 BLE module with arduino:-
The code is :-

#include <SoftwareSerial.h>
SoftwareSerial mySerial(8,9);
void setup()
{
mySerial.begin(115200);   
Serial.begin(115200);   
delay(100);
}
void loop()
{
if (Serial.available()>0)
mySerial.write(Serial.read());
if (mySerial.available()>0)
Serial.write(mySerial.read());
}

Now I am using above code for server and connect the Arduino with a separate PC and start observing the Data through serial monitor. On the other side I used same ESP32 setup with both the example one by one. In ble_spp_client example it is successfully sending the data and I can see the data on serial monitor of my Server (Arduino + HC-42 setup). But when I tried to send data from server same problem exist. In Gatts_Client.c example when I send from Server side I am able to  recieve in client part and can observe the data in NOTIFY_EVT in esp32.

gengyuchao, 2019-11-29:
Sorry, I'm not familiar with how HC-42 works. Can you provide a log of the Gatts_Client running with HC-42? Thank you.

chegewara, 2019-11-29:
Maybe its just logging issue, but i see problem on this screenshot:
https://user-images.githubusercontent.com/20817125/69847390-f9996800-129c-11ea-9c1e-bd3fec896e70.jpg

You have only 1 handle for characteristic (handle 13) and 2 handles for descriptor, which is wrong, because characteristic should have 2 handles and descriptor 1 handle, but this can be not related.

Could you check with nRF connect on android smartphone if characteristic you are trying to write is with WRITE or WRITE WITH NO RESPONSE properties? Because on this screen last log shows that you are trying to write attribute with handle 13 with status 0 (ESP_OK).

Also for testing purpose you can clone HC-42 in nRF connect, then connect with esp32 client and see logs on android smartphone.

ROYMURLI, 2019-11-30:
Hi, @gengyuchao 
The screenshot of GATT_CLIENT example with HC-42 is attached here. In this I am able to receive data through NOTIFY_EVT but not able to write to server (just vice-versa of ble_spp_client example). 
[image]

ROYMURLI, 2019-11-30:
Hi, @chegewara 
The screenshot which you have sharep of me is not with the original "ble_spp_client" example. Original one is attached here. Also i am using the nRF connect app for checking purpose. The screenshot is also attached here. 
[image]
[image]

chegewara, 2019-11-30:
You see, just like i suspected. Your HC-42 characteristic property is WRITE NO RESPONSE. You have to set parameter `write_type` for it or esp32 will await response:
https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/bluetooth/esp_gattc.html#_CPPv424esp_ble_gattc_write_char13esp_gatt_if_t8uint16_t8uint16_t8uint16_tP7uint8_t21esp_gatt_write_type_t19esp_gatt_auth_req_t

ROYMURLI, 2019-11-30:
What parameter should I use in attribute "write_type". Currently, I am using ESP_GATT_WRITE_TYPE_RSP by default. esp_ble_gattc_write_char( spp_gattc_if, spp_conn_id, db+SPP_IDX_SPP_DATA_RECV_VAL)->attribute_handle, event.size, temp,ESP_GATT_WRITE_TYPE_RSP, ESP_GATT_AUTH_REQ_NONE);

chegewara, 2019-11-30:
https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/bluetooth/esp_gatt_defs.html#_CPPv421esp_gatt_write_type_t

ROYMURLI, 2019-11-30:
I changed the "write_type" to ESP_GATT_WRITE_TYPE_NO_RSP but still it is not receiving data from server.

chegewara, 2019-11-30:
I am confused now. 
One time you write you cant receive data from server, other time you getting notifications but cant write to server.

ROYMURLI, 2019-11-30:
Hey, I am trying with two example. I mentioned that with "**ble_spp_client**" (spp_client_demo.c) example I am able to write to server but can't receive any attribute from server AND when I used "gatt_client" example (gattc_demo.c) I am unable to write character but able to receive it just vice-versa of ble_spp_client. For both the example screenshot is attached here. Please let me know if you need more for this. I can upload a video of all the demonstration.
[image]
[image]

chegewara, 2019-11-30:
From this screenshot i can see you can write characteristic with success and receive notifications too:
https://user-images.githubusercontent.com/20817125/69900307-87f21480-1397-11ea-9de4-5782ee367a36.jpg

From this screeshot you can see you have issue with handlers DB:
https://user-images.githubusercontent.com/20817125/69900308-87f21480-1397-11ea-9408-f98e1a685f4b.jpg
Look at third line from the bottom.

ROYMURLI, 2019-12-01:
Yes, if you see this one:-
https://user-images.githubusercontent.com/20817125/69900307-87f21480-1397-11ea-9de4-5782ee367a36.jpg
ESP_GATTC_WRITE_CHAR_EVT event is called automatically after connection with server and the "write char success" printed. But after connection once this event is over I am trying to write character through serial monitor but unable to call ESP_GATTC_WRITE_CHAR_EVT and unable to write it to server.  Yes, you are correct I am receiving notification as I mentioned earlier. Code is attached here. 
[gatt_demo1.zip](https://github.com/espressif/esp-idf/files/3908494/gatt_demo1.zip)

And for "ble_spp_client" I am trying to solve handlers DB issue which you have mentioned above.

ROYMURLI, 2019-12-03:
Any Solutions? @chegewara  @gengyuchao

gengyuchao, 2019-12-04:
There are no discoveries yet, I am waiting for my HC-42 arrive. I will try your issue when I receive my new HC-42. : )

gengyuchao, 2020-01-02:
Hi, @ROYMURLI I finally got HC-42 and tried to connect with ESP32 using HC-42. After some effort, the SPP function can be used normally. Because HC-42 and ESP32 have different service definitions, they cannot be used directly. The following modifications need to be made.
1.Modify the service UUID searched by ESP32 Client to the service provided by HC-42，or search all service
2. Modify the handle of the registered notify.
3. Modify the handle of write char of Enable Notify.

I will send you my code later。

ROYMURLI, 2020-01-03:
Thanks @gengyuchao  It worked. Will open again if I will face any problem.

closed 2020-01-03 by ROYMURLI