espressif/esp-idf #4368: Capture classical bluetooth RSSI in the station always returns the same value

riccardobrue, 2019-11-19:
Hi, I am trying to read the RSSI of a connected device to the ESP32-WROOM every 5 seconds but the read RSSI from the ESP32 is always the same, it changes just after 1/2 minutes.

What I have done is to start this task:
`void rssi_task(void *pvParameter){
	while(1) {
		esp_bt_gap_read_rssi_delta(connectedDeviceAddr);
		vTaskDelay(5000 / portTICK_PERIOD_MS);
	}
}`

And then I am going to read the result in the ESP_BT_GAP_READ_RSSI_DELTA_EVT event. What am I missing? Why is it so slow to refresh the RSSI? Thank you.

blueMoodBHD, 2019-12-27:
rssi delta value range -128 ~127, The value zero indicates that the RSSI is inside the Golden Receive Power Range, the Golden Receive Power Range is from ESP_BT_GAP_RSSI_LOW_THRLD to ESP_BT_GAP_RSSI_HIGH_THRLD

Alvin1Zhang, 2020-01-08:
@riccardobrue Thanks for reporting, would you please help share if any updates or further questions on this issue? Thanks.

Alvin1Zhang, 2020-05-27:
@riccardobrue Thanks for reporting, will close due to short of feedback, feel free to reopen if any more updates. Thanks.

closed 2020-05-27 by Alvin1Zhang