espressif/esp-idf #4369: mdns does not work in AP mode

AxelLin, 2019-11-19:
Test on IDF-4.0-beta2.

Test with getting_started/station/softAP example code.
Adding below changes to softap_example_main.c for the test.
Using avahi-discover command on my Linux laptop to test mdns response.
It does not work.

Also note, it works if I add the same test code to getting_started/station example.
It only does not work in AP mode.

`diff --git a/examples/wifi/getting_started/softAP/main/softap_example_main.c b/examples/wifi/getting_started/softAP/main/softap_example_main.c
index 010289554524..4e2401595762 100644
--- a/examples/wifi/getting_started/softAP/main/softap_example_main.c
+++ b/examples/wifi/getting_started/softAP/main/softap_example_main.c
@@ -14,6 +14,7 @@
 #include "esp_event.h"
 #include "esp_log.h"
 #include "nvs_flash.h"
+#include "mdns.h"
 
 #include "lwip/err.h"
 #include "lwip/sys.h"
@@ -26,9 +27,36 @@
 #define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
 #define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
 #define EXAMPLE_MAX_STA_CONN       CONFIG_ESP_MAX_STA_CONN
+#define EXAMPLE_MDNS_INSTANCE      "mdns_test_instance"
 
 static const char *TAG = "wifi softAP";
 
+static void initialise_mdns(void)
+{
+    char* hostname = "mdns_test";
+    //initialize mDNS
+    ESP_ERROR_CHECK( mdns_init() );
+    //set mDNS hostname (required if you want to advertise services)
+    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
+    ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);
+    //set default mDNS instance name
+    ESP_ERROR_CHECK( mdns_instance_name_set(EXAMPLE_MDNS_INSTANCE) );
+
+    //structure with TXT records
+    mdns_txt_item_t serviceTxtData[3] = {
+        {"board","esp32"},
+        {"u","user"},
+        {"p","password"}
+    };
+
+    //initialize service
+    ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3) );
+    //add another TXT item
+    ESP_ERROR_CHECK( mdns_service_txt_item_set("_http", "_tcp", "path", "/foobar") );
+    //change TXT item value
+    ESP_ERROR_CHECK( mdns_service_txt_item_set("_http", "_tcp", "u", "admin") );
+}
+
 static void wifi_event_handler(void* arg, esp_event_base_t event_base,
                                     int32_t event_id, void* event_data)
 {
@@ -86,4 +114,5 @@ void app_main()
 
     ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
     wifi_init_softap();
+    initialise_mdns();
 }
`

david-cermak, 2019-11-20:
Hi @AxelLin 

The issue is that the mdns has to be initialised before `wifi_init_softap()` in order to register on and receive the AP events. Please modify the initialisation order to:
```
void app_main()
{
    ESP_ERROR_CHECK(nvs_flash_init());
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    initialise_mdns();

    wifi_init_softap();
}
```

(Plus remove the initialisation of network stack, events and nvs from softAP init)

AxelLin, 2019-11-20:
Hi @david-cermak 
I tried exactly the same modification as you did, but it still does not work in my test.

Your modification is basically the same as moving initialise_mdns() into wifi_init_softap() function, put initialise_mdns() after esp_event_loop_create_default() call.

david-cermak, 2019-11-22:
@AxelLin That is strange, have just tested the 4.0-beta2 with avahi-browse and see no problem here. Is it just the service that cannot be resolved or a hostname too? does it work with any other  mdns app, such as a ServiceBrowser on the phone?

Could you please check the multicast packets are received by the board? Might you uncomment [this line](https://github.com/espressif/esp-idf/blob/master/components/mdns/private_include/mdns_private.h#L20) and check if the packets are printed to the output?

AxelLin, 2019-11-22:
Hi @david-cermak 

I tried again with enable MDNS_ENABLE_DEBUG.
It still shows nothing on avahi-discover and ServiceBrowser.

BTW, 
It has below log for 3 times *before* I join the AP.

TX[1][0]: To: 224.0.0.251:5353, Packet[2460]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [18] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [24] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [23] 0 0 80 mdns_test.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: mdns_test.local. A IN FLUSH 120 [4] 192.168.4.1

Below is the complete log:
I tried avahi-discover first, then tried ServiceBrowser on phone.

I (0) cpu_start: App cpu up.
I (412) heap_init: Initializing. RAM available for dynamic allocation:
I (419) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (425) heap_init: At 3FFB9910 len 000266F0 (153 KiB): DRAM
I (431) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (437) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (444) heap_init: At 40090730 len 0000F8D0 (62 KiB): IRAM
I (450) cpu_start: Pro cpu start user code
I (468) spi_flash: detected chip: generic
I (469) spi_flash: flash io: dio
W (469) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (479) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (608) wifi softAP: ESP_WIFI_MODE_AP
I (608) wifi softAP: mdns hostname set to: [mdns_test]
I (628) wifi: wifi driver task: 3ffba4f4, prio:23, stack:3584, core=0
I (628) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (628) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (658) wifi: wifi firmware version: 7b0c0f3
I (658) wifi: config NVS flash: enabled
I (658) wifi: config nano formating: disabled
I (658) wifi: Init dynamic tx buffer num: 32
I (658) wifi: Init data frame dynamic rx buffer num: 32
I (668) wifi: Init management frame dynamic rx buffer num: 32
I (668) wifi: Init management short buffer num: 32
I (678) wifi: Init static rx buffer size: 1600
I (678) wifi: Init static rx buffer num: 10
I (688) wifi: Init dynamic rx buffer num: 32
I (1798) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (1798) wifi: mode : softAP (30:ae:a4:09:75:2d)
I (1798) wifi: Total power save buffer number: 16
I (1798) wifi: Init max length of beacon: 752/752
I (1808) wifi: Init max length of beacon: 752/752
I (1808) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword

TX[1][0]: To: 224.0.0.251:5353, Packet[1560]: 
    Q: *U* ESP32-WebServer._http._tcp.local. ANY IN
    Q: *U* mdns_test...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [23] 0 0 80 mdns_test.local.
    S: mdns_test.local. A IN 120 [4] 192.168.4.1

TX[1][0]: To: 224.0.0.251:5353, Packet[1860]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: mdns_test...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [18] 0 0 80 mdns_test.local.
    S: mdns_test.local. A IN 120 [4] 192.168.4.1

TX[1][0]: To: 224.0.0.251:5353, Packet[2160]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: mdns_test...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [18] 0 0 80 mdns_test.local.
    S: mdns_test.local. A IN 120 [4] 192.168.4.1
[... 260 lines cut ...]
    Q: _services._dns-sd._udp.local. PTR IN

TX[1][0]: To: 224.0.0.251:5353, Packet[160760]: AUTHORITATIVE
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.

RX[1][0]: From: 192.168.4.3:5353, To: 224.0.0.251, Packet[161560]: 
    Q: _services._dns-sd._udp.local. PTR IN

TX[1][0]: To: 224.0.0.251:5353, Packet[161660]: AUTHORITATIVE
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.

RX[1][0]: From: 192.168.4.3:5353, To: 224.0.0.251, Packet[164570]: 
    Q: _services._dns-sd._udp.local. PTR IN

TX[1][0]: To: 224.0.0.251:5353, Packet[164660]: AUTHORITATIVE
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.

RX[1][0]: From: 192.168.4.3:5353, To: 224.0.0.251, Packet[173590]: 
    Q: _services._dns-sd._udp.local. PTR IN

TX[1][0]: To: 224.0.0.251:5353, Packet[173760]: AUTHORITATIVE
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
I (186118) wifi: station: 4c:ed:fb:4f:69:55 leave, AID = 1, bss_flags is 134259, bss:0x3ffbae80
I (186118) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (186118) wifi softAP: station 4c:ed:fb:4f:69:55 leave, AID=1

AxelLin, 2019-11-22:
Hi @david-cermak 

It works if I move nitialise_mdns() into wifi_event_handler() as below.
But I'm not sure if it's correct or not because that means initialise_mdns() can be called multiple times.

static void wifi_event_handler(void* arg, esp_event_base_t event_base,
                                    int32_t event_id, void* event_data)
{
    if (event_id == WIFI_EVENT_AP_STACONNECTED) {
        wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
        ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
                 MAC2STR(event->mac), event->aid);
    **initialise_mdns();**
} else if (event_id == WIFI_EVENT_AP_STADISCONNECTED) {

david-cermak, 2019-11-22:
@AxelLin Thanks for testing with debug logs. This looks pretty ok to me, can see the queries and responses being transmitted. Can you please share the exact commit you're on? There was one fix (IPv6 size) which might've caused the mdns packet to be dropped due to AAAA record size (but this doesn't happen with my avahi nor with dns-sd)

Yes in theory, you can call `mdns_init()` multiple times, as it returns `ESP_OK` if it was initialised before. The registration of the service (inside the `initialise_mdns()`) would however return error if there already is the same service registered.

What I would suggest is having a `mdns_init()` called once in `app_main()` and then `mdns_service_add()` and `mdns_service_remove()` in the event handler for WIFI_EVENT_AP_STACONNECTED and DISCONNECTED respectively. (which would actually make sense as the service would get advertised only after a station joined the AP)

This behaviour should work however irrespective of the initial probing/advertizing packets and as seen from your log it does as apparently the board receives queries and responds to them. For some reason these responses do not show...

AxelLin, 2019-11-22:
Hi @david-cermak 

Below is the latest commit shows by git log:
commit 99fb9a3f7c28c5fa12b1bd4aa6fb7b622d841326 (HEAD -> b4.0, origin/release/v4.0)
Merge: 4e6c575fed7c d49077c13b1a
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Nov 19 06:52:07 2019 +0800

    Merge branch 'bugfix/miscellaneous_cmake_fixes' into 'release/v4.0'
    
    CMake bugfixes
    
    See merge request espressif/esp-idf!6654

david-cermak, 2019-11-22:
Alright, thanks for sharing, the IPv6 size record fix is included. So it's not the issue with the wrong AAAA record size then.

AxelLin, 2019-11-26:
Hi @david-cermak 

Even though it works if adding mdns_service_add() in WIFI_EVENT_AP_STACONNECTED
event handler, I still feel this seems not a correct thing.
An AP can be associated by multiple STAs, so calling mdns_service_add() in WIFI_EVENT_AP_STACONNECTED
event handler will add *the same* service multiple times.
(And also adding the same mdns_service_txt_item_set calls.)
This seems does not make sense.

david-cermak, 2019-11-26:
@AxelLin Yes, you're correct. This is not the right solution, just a workaround to get it running  your environment.

As said before, the mdns library apparently works as expected, since the avahi-service sends out queries when it joins the AP, these queries could be seen in the log you've posted as well as the responses to them. Not sure what else could go wrong. Might I yet suggest capturing the mdns packets using Wireshark on the station when joining the AP? For some reason your avahi-service cannot see those packets or doesn't show them after querying, but does show after probing/announcing (when add_service triggered on `WIFI_EVENT_AP_STACONNECTED` event).

Also, strange as it may seem, I cannot reproduce it using the same commit id, not even with `avahi-discover`. can clearly see that when joining the ESP AP, avahi restarts the queries and shows the advertised service.

AxelLin, 2019-11-27:
@david-cermak 

Ah, I didn't realize you still not able to reproduce the issue.
Though this seems nothing to do with environment because the ESP32 is an AP.
(And I also try with different computer and phone, all got the same result)

I attached my changes in examples/wifi/getting_started/softAP and the wireshark log.
Thanks.

[mdns_log.zip](https://github.com/espressif/esp-idf/files/3894936/mdns_log.zip)

david-cermak, 2019-11-27:
@AxelLin Thanks for providing the Wireshark log, really appreciated!
This however still points to the environment, I can see the response packet coming from the 192.168.4.1 (the ESP AP) with the relevant and valid service data. (there's a little IP layer issue, probably caused by linux-capture as I can see it too when doing Wireshark on linux).

When running the test on my end (testing on two platforms: Mac and Linux) the capture looks almost identical, except for the fact that your capture contains multiple responses from the ESP AP. Which again seems like the querying party doesn't accept the responses and resends the queries.

Would you the service appear once you refresh the `avahi-browser` or if you restart the avahi?

david-cermak, 2019-11-27:
One more update from me. Was able to reproduce the behaviour with Android Service Browser, not sure though where the issue lies, since the ESP mDNS code passes Bonjour Conformance test and ESP32 AP services are detectable on Apple tools.

Posting a quick-fix here, could you please check if this solves your issue? The actual solution would be different though. Please use this patch only to check the environment (or as a workaround) since it may break the mdns library behaviour and thus violates the conformity with apple certification.

[0001-mdns-sketch-of-sending-also-a-sdptr-answer-to-sdptr-.patch.txt](https://github.com/espressif/esp-idf/files/3897243/0001-mdns-sketch-of-sending-also-a-sdptr-answer-to-sdptr-.patch.txt)

AxelLin, 2019-11-27:
Hi @david-cermak 
Thanks for the prompt update.
It works with your patch.

AxelLin, 2019-11-27:
BTW, I notice in the previous (before applying your patch) wireshark the Time to live field is1. This seems strange.
see attached screenshot.
[image]

david-cermak, 2019-11-27:
@AxelLin Thanks for the quick update, as well!

Wrtt "TimeToLive", yes this is weird, but probably caused by linux capture (when the same packet captured on Mac, the TTL is correct) and nothing to do with mdns. Noted that in the previous comment as well:

>  (there's a little IP layer issue, probably caused by linux-capture as I can see it too when doing Wireshark on linux).

closed 2019-12-16 by espressif-bot

AxelLin, 2019-12-17:
Hi @david-cermak 
This issue was reported on IDF-v4.0-beta2, any change to apply the fix to v4.0 branch?
Thanks.

david-cermak, 2019-12-17:
@AxelLin Yes, this fix will be back ported to all major releases including the v4.0.
This is just the default workflow that the issue gets closed once the fix appears on master.