espressif/esp-idf #4371: Request compatibility "websocket client" for esp-idf v3.3 (used by esp-jumpstart)

diamond2nv, 2019-11-19:
I'm new to esp-idf SDK, so project "esp-jumpstart" helped a lot.

We would like using "wss://" in esp-jumpstart by esp-idf v3.3, but the official "websocket client" is for esp-idf v4.0 and not kowning how to fit...

 "websocket client" compatible for esp-idf v3.3 will help us a lot. Thanks !

diamond2nv, 2019-11-20:
Well, just found a example, hope it would run ok fro ESP-IDF v3.3: https://github.com/kakopappa/idf_esp32_platformio_websocket_example

david-cermak, 2019-11-20:
Hi @diamond2nv 

We generally do not backport features, so there won't be a Websocket-client components in v3.3.

What we could however do is to backport the updates on `tcp_transport` component so the 4.0 version of `esp_websocket_client` could be used with v3.3. Would this help to address your issue?

diamond2nv, 2019-11-21:
OK, I get it. Thanks a lot !

By the way, with Espressif's quick-start-level guide, the "esp_websocket_client" is build & run successfully in v3.3.

closed 2019-11-21 by diamond2nv