espressif/esp-idf #4375: ESP-WROOM-32: Failed to init external RAM!

Pro, 2019-11-20:
## Environment

- Development Kit:      Adafruit HUZZAH32
- Module or chip used:  ESP32-WROOM-32
- IDF version: master (93a8603c545fb8e54741d6685146e2f3b874378d) and v4.0-beta2 (abdf12dc9e288a43783668e93ec5e1db126e06e2)
- Build System:         CMake
- Compiler version:     xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I have the Adafruit HUZZAH32 Board (see https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview) with an ESP32-WROOM-32 chip.

I want to use the external SPI RAM for dynamic memory allocations (i.e. increase the heap size from approx 300kB internal RAM to around 4MB).

To test this, I used the `himem` example. Unfortunately after flashing it shows the error **E (203) cpu_start: Failed to init external RAM!**

(I know that this chip only has 4MB so the himem example will not make much sense, but it's a good example for reproducing the issue)

Questions:

1. v4.0-beta2 should fully support external SPI RAM of the ESP32-WROOM-32 chip?
2. How can I get the SPI RAM being set up?

### Expected Behavior

The external RAM should be initialized as per the README.md
https://github.com/espressif/esp-idf/tree/master/examples/system/himem

### Actual Behavior

I'm getting the error:
**E (203) cpu_start: Failed to init external RAM!**

### Steps to repropduce

1. `cd examples/system/himem`
2. `idf.py menuconfig` (just save, do not change anything. README says so)
3. `idf.py -p /dev/ttyUSB9 flash`
4. `idf.py -p /dev/ttyUSB9 monitor`

## Debug Logs

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6440
load:0x40078000,len:15120
ho 0 tail 12 room 4
load:0x40080400,len:4552
entry 0x400806e4
I (30) boot: ESP-IDF v4.0-beta2 2nd stage bootloader
I (31) boot: compile time 16:54:27
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 2MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
[... 713 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
```

chegewara, 2019-11-20:
You can't init something that board does not have. WROOM modules does not have external RAM.

Pro, 2019-11-20:
> You can't init something that board does not have. WROOM modules does not have external RAM.

Thanks for your answer.

According to the official manual of the ESP-WROOM-32:

> ESP32-WROOM-32 integrates a 4 MB SPI flash, which is connected to GPIO6, GPIO7, GPIO8, GPIO9, GPIO10 and GPIO11. These six pins cannot be used as regular GPIOs.

I do not want to initialize external RAM, I want to use this integrated 4 MB SPI flash.

Isn't this what the himem example is showing?

Pro, 2019-11-20:
I did some more debugging into the spi psram initialization.

The line which fails is this one:

https://github.com/espressif/esp-idf/blob/abdf12dc9e288a43783668e93ec5e1db126e06e2/components/esp32/spiram_psram.c#L719

Do I need to change some sdkconfig settings to fix this?

chegewara, 2019-11-20:
SPI RAM is not the same as SPI flash. You dont have to initialize SPI flash. What you have to do now is to open `menuconfig` and disable this option `Component config ---> ESP32-specific ---> Support for external, SPI-connected RAM`

Pro, 2019-11-20:
Oh, right.... Damn it! The Adafruit HUZZAH32 does not have an external RAM.
I messed up the SPI Flash with SPI RAM....

Now I just need to find a way to work with the limited Flash, but that's another issue...

Thanks @chegewara

closed 2019-11-20 by Pro

Alvin1Zhang, 2019-11-21:
@chegewara Thanks for the notes and explanation.