espressif/esp-idf #4378: NVS Strings and Blobs disappear after reboot

mauriciofarina, 2019-11-20:
Hello,

I'm having some trouble with NVS. Basically, I write some variables to it and after that, I can successfully read them from the NVS partition. However, after rebooting the device, Strings and Blobs can't be found anymore. 

To make things easier, I have created a repository with a demo of the problem and all the description:

[https://github.com/mauriciofarina/ESP32_Bug_NVS](url)

What I have tried so far:

- Checked partition sizes and alignments
- Checked the NVS read and write process and it seems to be correct
- Tried Strings and Blobs with different lengths
- Removed from the project all code that is not related to the NVS

Alvin1Zhang, 2019-11-21:
@mauriciofarina Thanks for reporting.

0xjakob, 2019-11-21:
Hi @mauriciofarina I could reproduce your issue. But I couldn't without the middleware code. Is there an example which only reads/iterates and writes using IDF code?

0xjakob, 2019-11-21:
It seems like the iterator has problems iterating correctly over string/blob type entries when the iterator type is NVS_TYPE_I32 (and possibly other fixed-length entries, too). It then deletes part of the string/blob entry. I think it has nothing to do with erasing the flash before. 

I can reproduce the issue now by 1.) writing a string to nvs, 2.) iterating over the nvs with an iterator which looks for type NVS_TYPE_I32. This works without erasing the flash.

0xjakob, 2019-11-22:
@mauriciofarina Could you please verify whether this patch of the idf repo fixes the issue on your system?
[0001-bugfix-iterator-of-non-matching-type-works.patch.zip](https://github.com/espressif/esp-idf/files/3877955/0001-bugfix-iterator-of-non-matching-type-works.patch.zip)

mauriciofarina, 2019-11-28:
Hello @0xjakob !

The problem was indeed the iterator. After removing it from the functions that were using it everything worked just fine.

I also applied the patch and was able to use the iterator successfully!

You can close the issue.

Thank You!

0xjakob, 2019-11-29:
Good to hear! The patch will go to through internal review and will then be deployed on github.

Alvin1Zhang, 2020-01-08:
@0xjakob Thanks for the support. @mauriciofarina Thanks for reporting, the fix has been available on GitHub, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-08 by Alvin1Zhang