espressif/esp-idf #4380: Some mistakes in BT components

xiruilin, 2019-11-21:
File->\‪esp-idf\components\bt\common\osi\fixed_queue.c:
In line 157:
```
    if (osi_sem_take(queue->dequeue_sem, timeout) != 0) {
```
Should be changed to:
```
    if (osi_sem_take(&queue->dequeue_sem, timeout) != 0) {
```
In line 211:
```
    osi_sem_take(queue->dequeue_sem, 0) == 0) {
```
Should be changed to:
```
    osi_sem_take(&queue->dequeue_sem, 0) == 0) {
```
In line 218:
```
    osi_sem_give(queue->enqueue_sem);
```
Should be changed to:
```
    osi_sem_give(&queue->enqueue_sem);
```

File->\esp-idf\components\bt\host\bluedroid\btc\profile\std\a2dp\btc_av.c:
In line 1468:
```
void btc_a2dp_cb_handler(btc_msg_t *msg)
{
    btc_sm_dispatch(btc_av_cb.sm_handle, msg->act, (void *)(msg->arg));
    btc_av_event_free_data(msg->act, msg->arg);
}
```
Should be changed to:
```
void btc_a2dp_cb_handler(btc_msg_t *msg)
{
#if A2D_DYNAMIC_MEMORY == TRUE
    if (btc_av_cb_ptr != NULL)
#endif
    btc_sm_dispatch(btc_av_cb.sm_handle, msg->act, (void *)(msg->arg));
    btc_av_event_free_data(msg->act, msg->arg);
}
```

AbnerFederer, 2019-11-26:
Hi,@xiruilin,

Thanks very much for your codes.

1. The arg of `queue` in `osi_xxx()` function should be fixed to `&queue`.

2. The `btc_av_cb_ptr` has been define as `btc_av_cb` in the very front of source file `/esp-idf/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_av.c`. And we have some confused code in the same file, we will fix them as soon as possible.

enricop, 2019-11-26:
hi @AbnerFederer ,
please push, if possible, to `release/v4.0` branch as well

AbnerFederer, 2019-11-26:
> hi @AbnerFederer ,
> please push, if possible, to `release/v4.0` branch as well

sure，maybe available next week.

Than you.

Alvin1Zhang, 2020-01-08:
@AbnerFederer Thanks for the notes. @xiruilin Thanks for reporting, the fix has been available, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-08 by Alvin1Zhang