espressif/esp-idf #4381: BT connection issue

reshma26p, 2019-11-21:
hello..
i am usinf esp wroom32 board i want to connect esp to pc via bluetooth and send character from pc to bluetooth..
i flash the code of bt_spp_initiator code and i get esp bluetooth i  paird it with pc but as i want to send some txt file of small size it not send gives response as check bluetooth connection
and also com port could not assigned to bluetooth of esp

reshma26p, 2019-11-21:
please some one give me response

reshma26p, 2019-11-21:
please give me response its urgent for me

xiruilin, 2019-11-23:
@reshma26p 
I'm sorry to read the message so late.
I haven't talked about transferring files through Bluetooth to PC.
I'll try to see if I can work it out.

xiruilin, 2019-11-24:
@reshma26p 

If you want to send character from PC to esp32, "bt_spp_acceptor" is required to implement.
Follow the steps: (PC running Windows 10 system)
1. Run “bt_spp_acceptor” in esp32
2. PC->Setting->Device->Bluetooth->More Bluetooth options->
    COM port->Add...->Output->ESP_SPP_ACCEPTOR
    A virtual serial port will be created for esp32.
3. Use serial tools to send character from PC to esp32.
    (May be prompted to pair with "esp_spp_acceptor" device)

xiruilin, 2019-11-24:
@reshma26p 

If you want to send character from esp32 to PC, "bt_spp_initiator" is required to implement.
Follow the steps: (PC running Windows 10 system)
1. PC->Setting->Device->Bluetooth->More Bluetooth options->
   COM port->Add...->Input
   A virtual serial port will be created for esp32.
2. Use serial tools open virtual serial port.
3. In line 45 of "example_spp_initiator_demo.c", change the "remote_device_name[]" from
   "ESP_SPP_ACCEPTOR" to you PC bluetooth name, run "bt_spp_initiator" in esp32
   (May be prompted to pair with "bt_spp_initiator" device in PC)

AbnerFederer, 2019-11-28:
Hi, @reshma26p 

Any problem right now?

Alvin1Zhang, 2019-12-12:
@xiruilin Thanks for the notes. @reshma26p Thanks for reporting, will close this issue due to short of feedback. Feel free to reopen if the issue still happens. Thanks.

closed 2019-12-12 by Alvin1Zhang