espressif/esp-idf #4385: CMake Static library's dependency can't be found

drony, 2019-11-21:
MySDK uses my own MyLib and this uses nvs_flash lib. when i export mylib as a static library and when i used another application as libMylib.a. mylib's give error undefined reference to nvs_flash_init() etc. what i missing?

[image]

renzbagaporo, 2019-11-27:
You need to add `nvs_flash` to the dependencies of the imported library as well.

drony, 2019-11-29:
> 
> 
> You need to add `nvs_flash` to the dependencies of the imported library as well.

thanks it's resolved bu this time another function required.

` undefined reference to `vsnprintf`

but stdio.h included everywhere but give this error

closed 2019-12-09 by drony