espressif/esp-idf #4387: Default menuconfig colors incompatible with KDE's Konsole default color scheme

KaeLL, 2019-11-21:
Almost impossible to see the footnote instructions.
[image]

dobairoland, 2019-11-21:
`tools/cmake/kconfig.cmake` and `make/project_config.mk` contains `MENUCONFIG_STYLE` variable set to `aquatic`. You could try to rename `aquatic` to `default` or `monochrome`. You can customize it further as described here: https://github.com/espressif/esp-idf/blob/master/tools/kconfig_new/menuconfig.py#L67

Please let us know which setup works for you the best.

I plan to make this customizable.

drtyhlpr, 2019-11-21:
Hi!,

menuconfig: some recent changes seem to cause (me) problems (on a default debian10 vm). latest git, ./install and ./export checked. a simple `idf.py menuconfig` crashes with:

```
    curses.init_pair(len(color_attribs) + 1, fg_color, bg_color)
_curses.error: init_pair() returned ERR
```

any ideas? thanks!

dobairoland, 2019-11-21:
@drtyhlpr You have to set the TERM environment variable for your terminal. Something like `xterm-256color` should work.

Note: There will be a warning message about this in `master` soon.

drtyhlpr, 2019-11-21:
thanks for your fast solution!

ulfalizer, 2019-11-23:
@drtyhlpr 
What was your `TERM` setting when you got that error by the way?

FWIW, I recommend the default theme in a cross-platform environment, because I've made sure it's legible on Windows. I know it's a bit yellow, but it seemed like the least ugly option that avoids color-clashing with just ANSI colors (try de-yellowing it and you might see what I mean). :)

ulfalizer, 2019-11-23:
@dobairoland 
I see what's going on now. With e.g. `TERM=xterm`, `curses.COLORS` ends up as 8, but the `aquatic` theme uses one 9 color pairs.

Switching to the default theme would indirectly solve that, because it only uses 6 color pairs.

I wonder if I should have the color pairs wrap on gimped terminals. Might break the display in random ways depending on theme, but it might be better than crashing still. Another option might be to use monochrome on those.

Maybe aquatic could be tweaked to use less color pairs too...

ulfalizer, 2019-11-23:
Oh, wait, I had misremembered what `curses.COLORS` means. The problem is actually that `aquatic` uses a bright white color that's not part of the 8 basic ANSI colors.

This patch fixes it, for e.g. `TERM=xterm MENUCONFIG_STYLE=aquatic`:

```diff
diff --git a/menuconfig.py b/menuconfig.py
index 1f2e9fa..3a56d52 100755
--- a/menuconfig.py
+++ b/menuconfig.py
@@ -318,7 +318,7 @@ _STYLES = {
     separator=fg:white,bg:cyan,bold
     help=path
     frame=fg:white,bg:cyan,bold
-    body=fg:brightwhite,bg:blue
+    body=fg:white,bg:blue
     edit=fg:black,bg:white
     """
 }
```

Will add some version of that upstream tomorrow.

ulfalizer, 2019-11-23:
Fixed the code to avoid the crash @drtyhlpr saw, in https://github.com/ulfalizer/Kconfiglib/commit/7a98bc7944f1b390e5a82e40c62d0ce7540befdc.

Also changed the aquatic theme to not include 'brightwhite', in https://github.com/ulfalizer/Kconfiglib/commit/1ad02f526c3c0e2488f54c6028e998370839c7a5.

Re. the original issue, check if the default theme might be a better default than aquatic. I'm happy to take in any theme tweaks upstream too.

closed 2019-12-16 by espressif-bot

ulfalizer, 2019-12-19:
FYI, I just updated the aquatic theme to avoid cyan-on-blue for the path row and help text (and removed bold text, because it looks a bit cleaner IMO). It's white-on-blue now instead.