espressif/esp-idf #4389: Changes from tcpip_adapter to esp_netif broke C++ code

skandalfo, 2019-11-22:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1086-g93a8603c5
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0

- Operating System:     Linux
- Power Supply:         USB

## Problem Description

Changes ffe043b1a81a0f9e1cc2cfa8873e21318ec89143 and after that replaced tcpip_adapter.h with esp_netif.h.

The issue is not about the infrastructure change itself, but because the new code seems to not have had C++ usage in consideration.

A bunch of old guards for C++ (#ifdef __cplusplus extern "C" {) were removed and the new files don't seem to have any. This results in linkage errors when C++ mangled symbols are searched for.

### Expected Behavior

The headers can be directly used from C++ code.

### Actual Behavior

Compilation breaks unless I change the code to include first esp_netif.h and esp_wifi.h within an extern "C" block in my code.

### Steps to repropduce

Rename some example code that calls esp_netif_init() to be .cpp instead of .c. Linking will fail.

suda-morris, 2019-11-22:
Well spotted, actually we have fixed this issue in our internal task queue, will push to GitHub in the near future.

skandalfo, 2019-12-17:
I think this should be reopened.

I updated esp-idf to v4.1-dev-1543-g431066f16, and I'm still seeing code that doesn't work right when used from C++:

```
FAILED: esp-idf/esp_cpp_network/CMakeFiles/__idf_esp_cpp_network.dir/network.cpp.obj 
/home/skandalfo/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/include -I/home/skandalfo/esp/esp-idf/components/newlib/platform_include -I/home/skandalfo/esp/esp-idf/components/freertos/include -I/home/skandalfo/esp/esp-idf/components/heap/include -I/home/skandalfo/esp/esp-idf/components/log/include -I/home/skandalfo/esp/esp-idf/components/soc/esp32/include -I/home/skandalfo/esp/esp-idf/components/soc/include -I/home/skandalfo/esp/esp-idf/components/esp_rom/include -I/home/skandalfo/esp/esp-idf/components/esp_common/include -I/home/skandalfo/esp/esp-idf/components/xtensa/include -I/home/skandalfo/esp/esp-idf/components/xtensa/esp32/include -I/home/skandalfo/esp/esp-idf/components/esp32/include -I/home/skandalfo/esp/esp-idf/components/driver/include -I/home/skandalfo/esp/esp-idf/components/driver/esp32/include -I/home/skandalfo/esp/esp-idf/components/esp_ringbuf/include -I/home/skandalfo/esp/esp-idf/components/esp_event/include -I/home/skandalfo/esp/esp-idf/components/esp_netif/include -I/home/skandalfo/esp/esp-idf/components/lwip/include/apps -I/home/skandalfo/esp/esp-idf/components/lwip/include/apps/sntp -I/home/skandalfo/esp/esp-idf/components/lwip/lwip/src/include -I/home/skandalfo/esp/esp-idf/components/lwip/port/esp32/include -I/home/skandalfo/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/skandalfo/esp/esp-idf/components/vfs/include -I/home/skandalfo/esp/esp-idf/components/esp_wifi/include -I/home/skandalfo/esp/esp-idf/components/esp_wifi/esp32/include -I/home/skandalfo/esp/esp-idf/components/esp_eth/include -I/home/skandalfo/esp/esp-idf/components/tcpip_adapter/include -I/home/skandalfo/esp/esp-idf/components/efuse/include -I/home/skandalfo/esp/esp-idf/components/efuse/esp32/include -I/home/skandalfo/esp/esp-idf/components/app_trace/include -I/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_support/include -I/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_nvs/include -I/home/skandalfo/esp/esp-idf/components/nvs_flash/include -I/home/skandalfo/esp/esp-idf/components/spi_flash/include -I/home/skandalfo/esp/esp-idf/components/mbedtls/port/include -I/home/skandalfo/esp/esp-idf/components/mbedtls/mbedtls/include -mlongcalls -Wno-frame-address -g   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fstack-protector -std=gnu++11 -fexceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.1-dev-1543-g431066f16\" -DESP_PLATFORM -std=c++17 -MD -MT esp-idf/esp_cpp_network/CMakeFiles/__idf_esp_cpp_network.dir/network.cpp.obj -MF esp-idf/esp_cpp_network/CMakeFiles/__idf_esp_cpp_network.dir/network.cpp.obj.d -o esp-idf/esp_cpp_network/CMakeFiles/__idf_esp_cpp_network.dir/network.cpp.obj -c /home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp
In file included from /home/skandalfo/esp/esp-idf/components/esp_netif/include/esp_netif.h:23,
                 from /home/skandalfo/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/skandalfo/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp:8:
/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp: In function 'void network::start_networking()':
/home/skandalfo/esp/esp-idf/components/esp_netif/include/esp_netif_defaults.h:54:5: warning: missing initializer for member 'esp_netif_config::driver' [-Wmissing-field-initializers]
     }
     ^
/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp:170:37: note: in expansion of macro 'ESP_NETIF_DEFAULT_WIFI_STA'
   esp_netif_config_t netif_config = ESP_NETIF_DEFAULT_WIFI_STA();
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/skandalfo/esp/esp-idf/components/esp_netif/include/esp_netif_defaults.h:54:5: error: designator order for field 'esp_netif_config::driver' does not match declaration order in 'esp_netif_config_t' {aka 'esp_netif_config'}
     }
     ^
/home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp:170:37: note: in expansion of macro 'ESP_NETIF_DEFAULT_WIFI_STA'
   esp_netif_config_t netif_config = ESP_NETIF_DEFAULT_WIFI_STA();
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

The syntax with designated initializers in use in ´esp_netif_defaults.h´ is problematic with C++, which [doesn't officially support them until C++20](https://en.cppreference.com/w/cpp/language/aggregate_initialization#Designated_initializers). Even then, the C++ standard has extra requirements for the initializers to come in the same order as the struct field declarations, and partial initialization leaving no holes between initialized fields.

While this initialization would work in C99 code, these macros defined in the header are eventually expanded in a C++ compilation unit, then failing. Reordering the initializers seems to be enough to make this compile with the supplied toolchain even with -std=c++17, but depending on this behavior or C++20 looks like an unreasonable restriction.

What's the recommended approach to use these default configurations from C++ code?

skandalfo, 2019-12-17:
Even trying to workaround the above by copying the initialization of individual fields into my code, there are still headers missing the extern "C" marking; it looks like `esp_wifi/include/esp_wifi_default.h` is to blame:

```
FAILED: ledder.elf 
: && /home/skandalfo/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address -g   CMakeFiles/ledder.elf.dir/project_elf_src.c.obj  -o ledder.elf  esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_event/libesp_event.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/driver/libdriver.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/perfmon/libperfmon.a esp-idf/esp32/libesp32.a esp-idf/esp_common/libesp_common.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/esp_cpp_support/libesp_cpp_support.a esp-idf/esp_cpp_nvs/libesp_cpp_nvs.a esp-idf/esp_cpp_network/libesp_cpp_network.a esp-idf/esp_cpp_ota/libesp_cpp_ota.a esp-idf/esp_cpp_i2c/libesp_cpp_i2c.a esp-idf/esp_cpp_spi/libesp_cpp_spi.a esp-idf/esp_cpp_locality/libesp_cpp_locality.a esp-idf/esp_cpp_status_display/libesp_cpp_status_display.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=/home/skandalfo/esp/projects/esp_cpp/build/Debug/ledder.map -fno-rtti -fno-lto esp-idf/sdmmc/libsdmmc.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/protocomm/libprotocomm.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/esp_cpp_status_display/libesp_cpp_status_display.a esp-idf/esp_cpp_ota/libesp_cpp_ota.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_cpp_locality/libesp_cpp_locality.a esp-idf/esp_cpp_network/libesp_cpp_network.a esp-idf/esp_cpp_nvs/libesp_cpp_nvs.a esp-idf/esp_cpp_i2c/libesp_cpp_i2c.a esp-idf/esp_cpp_spi/libesp_cpp_spi.a esp-idf/esp_cpp_support/libesp_cpp_support.a esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_event/libesp_event.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/driver/libdriver.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/perfmon/libperfmon.a esp-idf/esp32/libesp32.a esp-idf/esp_common/libesp_common.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/ulp/libulp.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libespnow.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libmesh.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/librtc.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a /home/skandalfo/esp/esp-idf/components/esp_wifi/lib/esp32/libphy.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_event/libesp_event.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter [... cut ...]
/home/skandalfo/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/skandalfo/esp/projects/esp_cpp/components/esp_cpp_network/network.cpp:180: undefined reference to `esp_wifi_set_default_wifi_sta_handlers()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
"/usr/local/bin/ninja -j 8 all" terminated with exit code 1. Build might be incomplete.
```

Would it make sense to do a batch pass on public headers that define functions to verify that they are all guarded for C++?

igrr, 2019-12-17:
Reopening until the initializers order in the macro is fixed.

The other part related to esp_wifi_default.h is tracked in https://github.com/espressif/esp-idf/issues/4497; please follow the link and click "Subscribe" to be notified when that is fixed.

closed 2020-01-07 by espressif-bot

rebeliousconformist, 2020-01-08:
After update it throws the following error
error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]

sebastianbasierski, 2020-03-09:
I have also encountered such problem after updating to latest master (f7d1ed5f329c14c356a710c4c10ba50cea277b55)
`
/opt/esp-idf/components/heap/heap_caps_init.c: In function 'heap_caps_add_region_with_caps':
/opt/esp-idf/components/heap/heap_caps_init.c:230:44: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_LOCK(&registered_heaps_write_lock);
                                            ^~
/opt/esp-idf/components/heap/heap_caps_init.c:232:45: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_UNLOCK(&registered_heaps_write_lock);
      `

igrr, 2020-03-09:
@sebastianbasierski @rebeliousconformist Please check that the compiler version you are using is esp-2019r2 (check using `xtensa-esp32-elf-gcc --version`).

sebastianbasierski, 2020-03-11:
I can confirm i'm using given version

xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE

igrr, 2020-03-11:
@sebastianbasierski okay, could you please open a new issue for this, fill in the issue template and attach your sdkconfig file?

Alvin1Zhang, 2020-03-12:
@sebastianbasierski https://github.com/espressif/esp-idf/issues/new/choose, for your reference, thanks.