espressif/esp-idf #4390: TLS issues when pinning multiple certificates

tijnkooijmans, 2019-11-22:
## Environment

- Using our own custom hardware
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.2.3
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: macOS
- Power Supply: 3.3V LDO

## Problem Description

We use esp-mqtt to communicate with our server that runs on Amazon AWS. As recommended by Amazon we perform certificate verification on all 5 root certificates listed here: https://www.amazontrust.com/repository/. 

After setting up the mqtt connection we have about 70k of heap free, but when sending a lot of data over the line (+/- 1kb every second), we get a `esp-tls: write error` (see log below) and the connection doesn't recover until we reboot the ESP. 

When verifying only the currently active root certificate, so 1 instead of 5, we never see this issue. 

### Steps to repropduce

1. Setup a TLS MQTT connection using esp-mqtt lib
2. Have around 70k of heap free after
3. Send a lot of data to the broker (+/- 1kb every second)

### Code to reproduce this issue

Please find the certificate pem constant that we use for certificate verification here: https://gist.github.com/tijnkooijmans/605bd78f18163acc757b976259916446

## Debug Logs

```
.[0;31mE (106077) esp-tls: write error :-78:.[0m
.[0;31mE (106077) TRANS_SSL: esp_tls_conn_write error, errno=Software caused connection abort.[0m
.[0;31mE (106077) MQTT_CLIENT: Error write data or timeout, written len = -78, errno=113.[0m
```

david-cermak, 2019-11-25:
Hi @tijnkooijmans 

The error your are receiving, `-78` (https://github.com/espressif/mbedtls/blob/19eb57f7f7b2f4312d497ddbcb5f104fc6877c70/include/mbedtls/net_sockets.h#L60) means that the underlying socket cannot transmit data.
I would also suspect your're getting some network related errors too, such as shown below, imediatelly before or after the reported errors:
```
I (16197) wifi: bcn_timout,ap_probe_send_start
I (18707) wifi: ap_probe_send over, resett wifi status to disassoc
I (18707) wifi: state: run -> init (c800)
I (18707) wifi: pm stop, total sleep time: 5933068 us / 16612961 us

I (18707) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (18717) mbedtls: ssl_tls.c:2775 ssl->f_send() returned -78 (-0x004e)

W (18727) mbedtls: ssl_tls.c:3466 mbedtls_ssl_flush_output() returned -78 (-0x004e)

W (18727) mbedtls: ssl_tls.c:8566 mbedtls_ssl_write_record() returned -78 (-0x004e)

I (18737) mbedtls: ssl_tls.c:8647 <= write

E (18747) esp-tls: write error :-78
E (18747) TRANS_SSL: esp_tls_conn_write error, errno=Software caused connection abort
E (18757) MQTT_CLIENT: Error write data or timeout, written len = -78, errno=113
I (18767) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
W (18777) MQTT_CLIENT: Publish: Loosing qos0 data when client not connected
W (18777) MQTT_CLIENT: Publish: Loosing qos0 data when client not connected
I (18787) MQTTS_EXAMPLE: Sending id 0
I (18837) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (18837) wifi: state: init -> auth (b0)
I (18837) wifi: state: auth -> assoc (0)
I (18847) wifi: state: assoc -> run (10)
```

If that is the case it seems that the connection aborts and should reconnect automtically (in about 10s).
Do you still keep publishing data when the client disconnects? I'm asking because when publishing data with qos > 0, messages gets queued 
and with your data rate of > 1kb/s the outbox could easily overflow.

tijnkooijmans, 2019-11-26:
@david-cermak thanks for looking into this. Indeed my WIFI connection disconnects directly after and automatically reconnects. The MQTT connection never reconnects though. This automatic-reconnect works fine in other cases like when the WIFI goes offline. 

We're using QOS 0 for all messages except LWT and never send any message when the client is not connected.

Finally, do you understand why the above behaviour happens when 1 verify for all 5 AWS root certificates and doesn't happen when I only verify 1? I would expect the verification to only happen when establishing the MQTT connection and further not affecting the communication. Clearly it currently does affect the communication after the connection is established.

david-cermak, 2019-11-27:
@tijnkooijmans The only thing I could think of is the free heap. Could you please check the free heap sizes after connecting to the broker for these two scenarios (with only 1 certificate vs. with all 5 certs). Are these number roughy the same?

That could make a difference especially on slower networks, where allocating is more demanding when allocating buffers to be written to sockets not always writable...

tijnkooijmans, 2019-12-09:
@david-cermak I tested and when using 1 certificate we have 75972 bytes of free heap after the connection is established and all topics are subscribed, while with 5 certificates we have 69824 bytes left. Shouldn't that be enough? Also, I don't understand why the heap memory for verifying the certificates is not released after this verification is done. Am I not correct that the verification only needs to happen when establishing the MQTT connection?

david-cermak, 2019-12-17:
@tijnkooijmans Thanks for the update. Yes in most cases the server certificates are not needed and I think more recent versions of mbedtls support freeing the certificates after the handshake. (see mbedtls-2.17).

Yes, in general 60k seems to be quite enough, but as said it really depends on the network load (more allocation would be needed when consistently writing to buffers which couldn't be immediately sent to the network).
Could you please also test the scenario of using just 1 certificate and preallocate some dummy data to get those 60k free heap? In order to confirm that the issues your are seeing are really caused by the heap space? Thanks!

Alvin1Zhang, 2020-05-29:
@tijnkooijmans Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2020-05-29 by Alvin1Zhang