espressif/esp-idf #4391: how to configure two gpio pins as deep sleep

manojsai17, 2019-11-22:
how to configure two gpio pins as deep sleep . 
 I configured two pins as deep sleep   but deep sleep is  not enabling on two pins . only one pin is supporting .

how to configure deep sleep on two gpio pins?

chegewara, 2019-11-22:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#_CPPv428esp_sleep_enable_ext1_wakeup8uint64_t28esp_sleep_ext1_wakeup_mode_t

manojsai17, 2019-11-24:
I want to configure , Gpio 32 and Gpio 35 for deep sleep. 
what is the mask value to be configure in ext1_wakeup function ?
i kept the value (1<<gpio_32)| (1<gpio 35 )in fiunction . but wake up is not setting .

negativekelvin, 2019-11-24:
Use 64-bit value 1ULL<<32

stickbreaker, 2019-11-26:
@manojsai17 as @negativekelvin says, the bit shift command needs to use 64bit values.  The compiler defaults to 32bit when it sees a constant without a type override. so this means `(1 <<32)` would set a 32bit int to 1 then shift it left 32 places, with would cause the 1 to shift past the 31st bit position. so the result would be `0`.
```c++
uint64_t mask = ( 1ULL <<32 ) |  ( 1ULL << 35 );
```

Chuck.

Alvin1Zhang, 2020-03-02:
@manojsai17 Has your issue been resolved? Thanks.

closed 2022-04-07 by Alvin1Zhang