espressif/esp-idf #4392: BLE Notification doesn't work when "Stack Smashing Protection Mode" is enabled

joehui, 2019-11-23:
This is what I noticed, when "Stack Smashing Protection Mode" is "None", my BLE notification (ESP32 as being a peripheral) is working fine.  Once it is set to any one of the other 3 modes, BLE notification stops working.

Is this expected behavior?   Or is something wrong with my code?

chegewara, 2019-11-23:
If code does not work with stack smashing it means you have problem with tasks stack size.

joehui, 2019-11-24:
I had already increased the Timer Task Stack Size to 3072 before.

Should I try increase Idle Task Stack Size and ISR Stack Size?  How do I know which one to increase?  Typically, what would be a  good value?

joehui, 2019-11-24:
BTW, I significantly increased the 3 stack sizes and BLE notification still doesn't work. (Once I disable BLE stack smashing, things work again.)

 Timer Task Stack Size -> 8000
  Idle Task Stack Size -> 4096
  ISR Stack Size -> 4096

joehui, 2019-11-24:
I also increased these values through make menuconfig, still no effect:

Event Loop Stack Size --> 8000
Main Task Stack Size --> 8000
IPC Task Stack Size --> 2048
High Res Timer Stack Size --> 8000

gengyuchao, 2019-11-25:
Thank you for you report. Could you provide the debug log?

joehui, 2019-11-25:
There is nothing unusual logged.  There is no crash.  Notification registration is successful.  But when the central sends notifications, the ESP32 (which acts as peripheral) doesn't receive any event at all.

gengyuchao, 2019-11-25:
> There is nothing unusual logged. There is no crash. Notification registration is successful. But when the central sends notifications, the ESP32 (which acts as peripheral) doesn't receive any event at all.

Yes, I find the root cause. The problem doesn't come from the "Stack Smashing Protection Mode" option. It is a problem with pointer type conversion. And we will fix it as soon as possible.

gengyuchao, 2019-11-25:
You can use this command to solve it before we publish it in github.

`git apply you_file_location/fix_pointer_type.patch`

[fix_pointer_type.zip](https://github.com/espressif/esp-idf/files/3886619/fix_pointer_type.zip)

joehui, 2019-11-25:
Great.  Thanks.
What version of IDF can I apply this patch to?

gengyuchao, 2019-11-25:
You can use it in the master branch.

Alvin1Zhang, 2020-01-07:
@joehui Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

joehui, 2020-03-19:
Sorry for the late response.  I just confirmed that with the patched code, notification is working even when stack smashing is turned on.

Alvin1Zhang, 2020-03-20:
@joehui Thanks for updates, glad that your issue gets resolved. Feel free to reopen if the issue still happens. Thanks.

closed 2020-03-20 by Alvin1Zhang