espressif/esp-idf #4393: [BLE Mesh]IRAM0 segment data does not fit

isDerek, 2019-11-25:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.1-dev-1086-g93a8603c5
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

init the PSRAM
```
ESP32-specific --> Support for external,SPI-connected RAM --> Try to allocate memories of Wifi and LWIP...
Bluetooth --> Bluedriod Enable --> BT/BLE will first malloc the memory from the PSRAM
Bluetooth --> Bluedriod Enable --> Use dynamic memory allocation in BT/BLE stack.
Bluetooth --> Blutooth controller --> BLE full scan feature supported.
```

### Expected Behavior

success

### Actual Behavior

```
/Users/zhujunkun/Desktop/ayla-espressif/espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/zhujunkun/Desktop/ayla-espressif/pda-http-2.4-esp32/examples/ayla_demo/blemesh-gateway/build/blemesh_gateway.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/zhujunkun/Desktop/ayla-espressif/espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/Users/zhujunkun/Desktop/ayla-espressif/espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 21116 bytes
collect2: error: ld returned 1 exit status
make: *** [/Users/zhujunkun/Desktop/ayla-espressif/pda-http-2.4-esp32/examples/ayla_demo/blemesh-gateway/build/blemesh_gateway.elf] Error 1
```

0xjakob, 2019-11-25:
Hi @isDerek, could you please elaborate on the problem description? Just "init the PSRAM" and some seemingly menuconfig settings are a bit vague. It would as well be helpful to describe what you're trying to achieve, including an example of your code which has this issue. Thanks.

isDerek, 2019-11-25:
i want to add the ble mesh module to my wifi demo(the demo is a wifi agent for our company's cloud).  when i never change the sdkconfig ( ble & wifi coexist sdkconfig), the log tell me 

```
I (985) wifi: wifi firmware version: a44d1c6
I (986) wifi: config NVS flash: enabled
I (986) wifi: config nano formating: disabled
I (987) wifi: Init dynamic tx buffer num: 32
I (997) wifi: Init data frame dynamic rx buffer num: 64
I (998) wifi: Init management frame dynamic rx buffer num: 64
I (1009) wifi: Init management short buffer num: 32
I (1009) wifi: Init static tx buffer num: 16
I (1010) wifi: Init static rx buffer size: 1600
W (1021) wifi: malloc buffer fail
I (1022) wifi: Init static rx buffer num: 14
I (1022) wifi: Init dynamic rx buffer num: 64
E (1033) wifi: Expected to init 16 rx buffer, actual is 14
I (1033) wifi: Deinit lldesc rx mblock:0
I (1034) wifi: Deinit lldesc rx mblock:0
I (1045) wifi: Deinit lldesc rx mblock:0
I (1045) wifi: Deinit lldesc rx mblock:0

```
so i think maybe i need init the PSRAM, when i modify the menuconfig, the log occured

```
region `iram0_0_seg' overflowed by 21116 bytes
```

negativekelvin, 2019-11-25:
iram is instruction ram and unrelated to psram which is a type of data ram.  This means too much code is placed in iram.

isDerek, 2019-11-25:
oh~ how could i tracer the iram space, have some methods?

negativekelvin, 2019-11-25:
https://github.com/espressif/esp-idf/issues/4234

isDerek, 2019-11-25:
i modify the esp32.ld file, i change the iram0_0_seg size from 0x20000 to 0x30000
```
 iram0_0_seg (RX) :                 org = 0x40080000, len = 0x30000
```
does it not the allowed action? the code can't run completely, the code run stopped
```
I (1896) wifi: wifi firmware version: a44d1c6
I (1897) wifi: config NVS flash: enabled
I (1897) wifi: config nano formating: disabled
I (1898) wifi: Init dynamic tx buffer num: 32
I (1909) wifi: Init data frame dynamic rx buffer num: 64
I (1909) wifi: Init management frame dynamic rx buffer num: 64
I (1920) wifi: Init management short buffer num: 32
I (1921) wifi: Init static tx buffer num: 16
I (1922) wifi: Init static rx buffer size: 1600
I (1933) wifi: Ini
```

negativekelvin, 2019-11-25:
You can't just change iram length because it will overlap with other memory regions that are in use

isDerek, 2019-11-25:
could i decrease the size (bt) in the iram0_0_seg ? menuconfig?

isDerek, 2019-11-25:
i cancel the WiFi IRAM speed optimization and WiFi RX IRAM speed optimization（menuconfig）. finally, the iram size could be accepted.

closed 2019-11-25 by isDerek