espressif/esp-idf #4394: advanced https ota failed oftenly

huazhenjiang, 2019-11-25:
### Environment:
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): v2
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-released
Operating System: Windows 10
Power Supply: USB

###Problem Description
I try the advanced https ota example to test ota . And I observer the ota procedure get low success rate. The ota get success sometimes, but worse oftenly. The following show the log :

###Expected Behavior
ota successful oftenly
###Actual Behavior
ota failed oftenly
###Steps to repropduce
Just copy a pem of https server and run the advanced https ota example
###Code to reproduce this issue
none.

###Debug Logs
I (1990) wifi: new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (2960) wifi: state: init -> auth (b0)
I (2970) wifi: state: auth -> assoc (0)
I (2970) wifi: state: assoc -> run (10)
I (2980) wifi: connected with test, channel 11, 40D, bssid = 78:44:76:f8:b4:44
I (2990) wifi: pm start, type: 1

I (5110) tcpip_adapter: sta ip: 192.168.200.18, mask: 255.255.255.0, gw: 192.168.200.1
I (5110) example_connect: Connected to test
I (5110) example_connect: IPv4 address: 192.168.200.18
I (5120) example_connect: IPv6 address: fe80:0000:0000:0000:266f:28ff:fed7:a8f4
I (5130) advanced_https_ota_example: Starting Advanced OTA example

I (12030) esp_https_ota: Starting OTA...
I (12030) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
I (15450) advanced_https_ota_example: Running firmware version: 1.1.5-3-gebd9eab-dirty
E (23230) esp-tls: read error :-76:
E (23230) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (213020) esp-tls: read error :-76:
E (213020) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (259580) esp-tls: read error :-76:
E (259580) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (264660) esp-tls: read error :-76:
E (264660) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (319000) esp-tls: read error :-76:
E (319000) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (336510) esp-tls: read error :-76:
E (336510) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
I (341510) esp_https_ota: Connection closed, all data received
I (341510) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x2c480 (181376) map
I (341650) esp_image: segment 1: paddr=0x0013c4a8 vaddr=0x3ffb0000 size=0x03540 ( 13632) 
I (341660) esp_image: segment 2: paddr=0x0013f9f0 vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /home/huazhen/Work/mmwave-esp32/esp-idf/components/freertos/xtensa_vectors.S:1778

I (341660) esp_image: segment 3: paddr=0x0013fdf8 vaddr=0x40080400 size=0x00218 (   536) 
I (341670) esp_image: segment 4: paddr=0x00140018 vaddr=0x400d0018 size=0xb96fc (759548) map
0x400d0018: _stext at ??:?

E (342250) esp_image: invalid segment length 0xffffffff
E (342260) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed...
I (343260) wifi: state: run -> init (0)
I (343260) wifi: pm stop, total sleep time: 299923894 us / 340271911 us

I (343260) wifi: new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (343270) example_connect: Wi-Fi disconnected, trying to reconnect...
E (343270) wifi: esp_wifi_connect 1151 wifi not start
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED)I (343280) wifi: flush txq
I (343280) wifi: stop sw txq
I (343290) wifi: lmac stop hw txq
I (12) boot: ESP-IDF 1.1.5-3-gebd9eab-dirty 2nd stage bootloader
I (12) boot: compile time 10:00:50
I (12) boot: Enabling RNG early entropy source...
I (17) boot: SPI Speed      : 40MHz
I (21) boot: SPI Mode       : DIO
I (25) boot: SPI Flash Size : 4MB
I (29) boot: Partition Table:
I (33) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (47) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (55) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (62) boot:  3 factory          factory app      00 00 00010000 00100000
I (70) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (77) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (85) boot: End of partition table
I (89) boot: Defaulting to factory image
I (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1dafc (121596) map
I (146) esp_image: segment 1: paddr=0x0002db24 vaddr=0x3ffb0000 size=0x024ec (  9452) load
I (150) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x8eb04 (584452) map
0x400d0018: _stext at ??:?

I (362) esp_image: segment 3: paddr=0x000beb24 vaddr=0x3ffb24ec size=0x00d60 (  3424) load
I (363) esp_image: segment 4: paddr=0x000bf88c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/huazhen/Work/mmwave-esp32/esp-idf/components/freertos/xtensa_vectors.S:1778

I (369) esp_image: segment 5: paddr=0x000bfc94 vaddr=0x40080400 size=0x1036c ( 66412) load
I (415) boot: Loaded app from partition at offset 0x10000
I (416) boot: Disabling RNG early entropy source...

mahavirj, 2019-11-26:
@huazhenjiang 

 - How often do you see this issue? 
 - Can you please share more details on server setup? Is it based on `openssl`, if not then does it support `chunked-encoding`?
 - Is it possible to try with IDF `master` branch and see if it has similar behavior?
 - Can you please share log after enabling `debug` from logging component config?

huazhenjiang, 2019-11-27:
@mahavirj 

I try 101 times and 22 of that failed. The target server is on amazon s3, I do not know that did it support chunked-encoding actually.

The following log show more detail. It looks like the ota procedure read a part of data and after that can not read anymore.
 
D (66873) esp_https_ota: Written image length 277570
D (66873) advanced_https_ota_example: Image bytes read: 277570
D (66883) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (66883) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=0
D (66893) HTTP_CLIENT: http_on_body 512
D (66903) esp_https_ota: Written image length 278082
D (66903) advanced_https_ota_example: Image bytes read: 278082
D (66903) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (66913) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=360, ridx=0
D (66923) HTTP_CLIENT: http_on_body 360
D (66923) HTTP_CLIENT: is_data_remain=1, is_chunked=0
W (81523) mbedtls: ssl_tls.c:5031 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (81523) mbedtls: ssl_tls.c:4350 ssl_get_next_record() returned -26624 (-0x6800)

W (81523) mbedtls: ssl_tls.c:8341 mbedtls_ssl_read_record() returned -26624 (-0x6800)

E (81533) esp-tls: read error :-26624:
E (81543) TRANS_SSL: esp_tls_conn_read error, errno=Success
D (81543) HTTP_CLIENT: need_read=152, byte_to_read=152, rlen=-26624, ridx=360
D (81563) esp_https_ota: Written image length 278442
D (81563) advanced_https_ota_example: Image bytes read: 278442
D (81563) HTTP_CLIENT: is_data_remain=1, is_chunked=0
W (93193) mbedtls: ssl_tls.c:5031 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (93193) mbedtls: ssl_tls.c:4350 ssl_get_next_record() returned -26624 (-0x6800)

 

W (93193) mbedtls: ssl_tls.c:8341 mbedtls_ssl_read_record() returned -26624 (-0x6800)

E (93203) esp-tls: read error :-26624:
E (93213) TRANS_SSL: esp_tls_conn_read error, errno=Success
D (93213) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=-26624, ridx=0
I (93233) esp_https_ota: Connection closed, all data received
D (93233) esp_image: reading image header @ 0x110000
D (93233) esp_image: image header: 0xe9 0x06 0x02 0x02 4008138c
V (93243) esp_image: loading segment header 0 at offset 0x110018
V (93243) esp_image: segment data length 0x2c480 data starts 0x110020
V (93253) esp_image: segment 0 map_segment 1 segment_data_offs 0x110020 load_addr 0x3f400020
I (93263) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x2c480 (181376) map
D (93273) esp_image: free data page_count 0x0000003d
V (93413) esp_image: loading segment header 1 at offset 0x13c4a0
V (93413) esp_image: segment data length 0x3540 data starts 0x13c4a8
V (93413) esp_image: segment 1 map_segment 0 segment_data_offs 0x13c4a8 load_addr 0x3ffb0000
I (93423) esp_image: segment 1: paddr=0x0013c4a8 vaddr=0x3ffb0000 size=0x03540 ( 13632)
D (93433) esp_image: free data page_count 0x0000003d
V (93443) esp_image: loading segment header 2 at offset 0x13f9e8
V (93443) esp_image: segment data length 0x400 data starts 0x13f9f0
V (93453) esp_image: segment 2 map_segment 0 segment_data_offs 0x13f9f0 load_addr 0x40080000
0x40080000: _WindowOverflow4 at /home/peter/Work/ESP32/Bitbucket/mmwave-esp32/esp-idf/components/freertos/xtensa_vectors.S:1778

enricop, 2019-11-27:
which toolchain are you using? tls issues should be solved in 2019r2

huazhenjiang, 2019-11-27:
@enricop 

how to check the toolchain version?
I download from here.(https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz)

enricop, 2019-11-27:
that toolchain is the old one.
use this: 2019**r2**
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/linux-setup.html#toolchain-setup

huazhenjiang, 2019-11-28:
@enricop 

ok, I wiil try this. So is that the root cause which causing the ota failed? the old toolchain version.

mahavirj, 2019-11-28:
@enricop Any specific reason why do you think this could be toolchain version related issue?

@huazhenjiang Error code from earlier log `0x6800` points to timeout, so one thing you could try is to increase timeout and check (can be done by modifying `timeout_ms` field of `esp_http_client_config_t` in example code). Other question I have is, if you download same file from any host tool (like `curl` or `wget`) do you observe any such issue in multiple iteration run?

huazhenjiang, 2019-11-28:
@mahavirj 

I have been tried to add 10 seconds timeout for testing this issue, but it looks not work. 
Also I try to download by *wget* with 30 times, it get 100% success rate.

mahavirj, 2019-11-29:
@huazhenjiang Is it possible to share reference code (this will need your S3 url to be public), I will try to reproduce at my end?

One more suggestion that could be tried out is to modify application code per https://github.com/espressif/esp-idf/blob/master/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c#L137 (not sure it will help or not but worth trying)

huazhenjiang, 2019-12-02:
@mahavirj 

OK, I could share all of the testing files including the FW bin, s3 public url, log and the sniffing log with wireshark ... etc.
I packed them with zip, please see the 'readme' file first.

[advanced_https_ota_example.zip](https://github.com/espressif/esp-idf/files/3909295/advanced_https_ota_example.zip)

hehao9051, 2019-12-02:
@mahavirj I have the same question. IDF version is V3.2 286202c。I not set the timeout_ms(actually if i set timeout_ms=5000, the error still appear) see the log:
lwip_select(57, 0x3ffcc340, 0x0, 0x0, tvsec=5 tvusec=0)
lwip_selscan: fd=56 ready for reading
lwip_select: nready=1
lwip_recvfrom(56, 0x3ffec634, 5, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x3ffdec54
lwip_recvfrom: buflen=1200 len=5 off=0 sock->lastoffset=813
lwip_recvfrom(56): addr=58.215.145.137 port=443 len=5
lwip_recvfrom: lastdata now netbuf=0x3ffdec54
lwip_recvfrom(56, 0x3ffec639, 16408, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x3ffdec54
lwip_recvfrom: buflen=1200 len=16408 off=0 sock->lastoffset=818
lwip_recvfrom: deleting netbuf=0x3ffdec54
lwip_recvfrom: top while sock->lastdata=0x0
lwip_recvfrom: netconn_recv err=0, netbuf=0x3ffcf07c
lwip_recvfrom: buflen=1200 len=16026 off=382 sock->lastoffset=0
lwip_recvfrom: deleting netbuf=0x3ffcf07c
lwip_recvfrom: top while sock->lastdata=0x0
lwip_recvfrom: netconn_recv err=0, netbuf=0x3ffd0198
lwip_recvfrom: buflen=1200 len=14826 off=1582 sock->lastoffset=0
lwip_recvfrom: deleting netbuf=0x3ffd0198
lwip_recvfrom: top while sock->lastdata=0x0
lwip_recvfrom: netconn_recv err=0, netbuf=0x3ffd01b4
lwip_recvfrom: buflen=1200 len=13626 off=2782 sock->lastoffset=0
lwip_recvfrom: deleting netbuf=0x3ffd01b4
lwip_recvfrom: top while sock->lastdata=0x0
lwip_recvfrom: netconn_recv err=0, netbuf=0x3ffdc54c
lwip_recvfrom: buflen=1200 len=12426 off=3982 sock->lastoffset=0
lwip_recvfrom(56): addr=58.215.145.137 port=443 len=5182
lwip_recvfrom: deleting netbuf=0x3ffdc54c
lwip_recvfrom(56, 0x3ffeda77, 11226, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x0
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select(56, 0x0, 0x3ffaf7a4, 0x0, tvsec=0 tvusec=200000)
lwip_selscan: fd=55 ready for writing
lwip_select: nready=1
lwip_sendto(55, data=0x3ffdc730, short_size=363, flags=0x0 to=255.255.255.255 port=5683
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select(56, 0x0, 0x3ffaf7a4, 0x0, tvsec=0 tvusec=200000)
lwip_selscan: fd=55 ready for writing
lwip_select: nready=1
lwip_sendto(55, data=0x3ffdc730, short_size=363, flags=0x0 to=255.255.255.255 port=5683
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select(56, 0x0, 0x3ffaf7a4, 0x0, tvsec=0 tvusec=200000)
lwip_selscan: fd=55 ready for writing
lwip_select: nready=1
lwip_sendto(55, data=0x3ffdc730, short_size=363, flags=0x0 to=255.255.255.255 port=5683
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_select: timeout expired
lwip_select(56, 0x3ffe1b54, 0x0, 0x0, tvsec=0 tvusec=200000)
lwip_recvfrom: netconn_recv err=-3, netbuf=0x0
lwip_recvfrom(56): buf == NULL, error is "Timeout."!
lwip_getsockopt(56, SOL_SOCKET, SO_ERROR) = 11
E (28426) esp-tls: read error :-76:
E (28426) TRANS_SSL: esp_tls_conn_read error, errno=No more processes

shubhamkulkarni97, 2019-12-03:
@huazhenjiang I tried updating from the link that you have mentioned in readme.txt, with timeout set to 60000. The example worked fine for around 100 iterations. Can you please try setting timeout to 60000? This is a workaround, however we are trying to dig deeper to check is any bug exists. I have attached a patch for setting the timeout, for your reference.
[advanced_https_ota.patch.zip](https://github.com/espressif/esp-idf/files/3914854/advanced_https_ota.patch.zip)

hehao9051, 2019-12-04:
@shubhamkulkarni97
I have the same question. IDF version is V3.2 286202c。actually even i set timeout_ms=10000, the error still appear

shubhamkulkarni97, 2019-12-05:
@huazhenjiang @hehao9051 I have attached a patch to fix advanced_https_ota example. It has two changes,

- Call to `fcntl()` is not required in function `net_would_block`, as it is already handled in other layer. Due to this, `net_would_block` used to return 0 for any condition. This change fixes `esp-tls: read error :-76` error.

- Errno check is added in function `esp_https_ota_perform()` to check whether OTA data is completely received or `esp_http_client_read()` returned 0 due to an error. This will fix the bug of connection being closed even if the entire file is not received.

Can you please try running the example after applying this patch? Feel free to get-back if you face any errors.
[bugfix_advanced_https_ota.patch.zip](https://github.com/espressif/esp-idf/files/3926552/bugfix_advanced_https_ota.patch.zip)

huazhenjiang, 2019-12-07:
@shubhamkulkarni97 

I have been tried this patch and it works. The success rate of ota is improved to 97 %.
So should I close this topic or not ? until you verified and merged .

hehao9051, 2019-12-07:
@shubhamkulkarni97 thanks your work, but the error is still appear, when data_read == 0 && errno == EAGAIN, the next esp_https_ota_perform will return data_read == 0 && errno != EAGAIN.

mahavirj, 2019-12-09:
@huazhenjiang Issue will get closed once fix is officially merged in github repository. Final fix will depend on some additional tests and applicability across different ESP-IDF release (v3.x/v4.x).

@hehao9051 We are trying some additional tests, will keep you posted on this.

shubhamkulkarni97, 2019-12-11:
> @shubhamkulkarni97 thanks your work, but the error is still appear, when data_read == 0 && errno == EAGAIN, the next esp_https_ota_perform will return data_read == 0 && errno != EAGAIN.

@hehao9051 In which scenario do you observe this problem? Can you please explain in detail?

shubhamkulkarni97, 2019-12-12:
@huazhenjiang @hehao9051  I have attached a patch, it should fix the issues with timeout in OTA Update. We have added an option in `menuconfig` to set receive timeout, by default it is set to `5000ms`.

Can you please try running the example after applying the patch?
[bugfix_for_ota.patch.zip](https://github.com/espressif/esp-idf/files/3954635/bugfix_for_ota.patch.zip)

Arthedian, 2019-12-19:
@shubhamkulkarni97

I am trying to apply your patch, but I am not successful. I am on newest commit of release/v.4.0  - 20e6d46d4421414aa017e451dc6041b5de5b221e. How can i apply it?

git apply ../../Downloads/bugfix_for_ota.patch/bugfix_for_ota.patch
../../Downloads/bugfix_for_ota.patch/bugfix_for_ota.patch:81: trailing whitespace.
 * 
error: patch failed: components/mbedtls/port/net_sockets.c:198
error: components/mbedtls/port/net_sockets.c: patch does not apply

shubhamkulkarni97, 2019-12-20:
@Arthedian 
As some changes from internal repository are synced with public GitHub repository, the patch fails to apply.

I have attached another patch, which should be applied successfully.
Please try the patch and let me know if you face any issues.
[ota_bugfix.patch.zip](https://github.com/espressif/esp-idf/files/3987661/ota_bugfix.patch.zip)

shubhamkulkarni97, 2019-12-30:
> @huazhenjiang @hehao9051 I have attached a patch, it should fix the issues related to timeout in OTA Update. We have added an option in `menuconfig`, to set receive timeout, by default it is set to `5000ms`.
> 
> Can you please try running the example after applying this patch?
> [bugfix_for_ota.patch.zip](https://github.com/espressif/esp-idf/files/3954635/bugfix_for_ota.patch.zip)
@huazhenjiang @hehao9051 Did the patch work for you?

Arthedian, 2019-12-30:
@shubhamkulkarni97 I tried the patch which you entered 10 days ago and it worked on WiFi for me, so I want to thank you for it. 

I am trying to use it with SIM800L but it is taking too long and still saying “HW FIFO Overflow”. I probably know where is the problem, and I already tried to tweak some uart setting but without any luck. Do you know how to fix it?

closed 2020-01-06 by espressif-bot

shubhamkulkarni97, 2020-01-10:
@Arthedian OTA Update may take some time depending on data rates. For HW FIFO Overflow issue, can you please provide logs?

Arthedian, 2020-01-10:
@shubhamkulkarni97 I created a new issue for this problem [here](https://github.com/espressif/esp-idf/issues/4582 )